espressif/esp-idf #7718: RTC_NOINIT_ATTR not working

baldhead69, 2021-10-18:
Hi,

I'm trying to preserve a structure with 2 variables bool and 1 variable ​​uint8_t between resets using RTC_NOINIT_ATTR, but the data are not preserved.
I am testing reseting by a reset button on a ESP32-DevKitC-VB.
I dont know if reseting by en pin preserve or not preserve rtc_data memory.

h file:
```c
typedef struct 
{
    bool a;
    bool b;
    uint8_t thing;   
}someState_t;    

extern serverState_t state;
```

c file:
```c
RTC_NOINIT_ATTR someState_t state;

static void init_someState_struct()
{
    esp_reset_reason_t reset_reason = esp_reset_reason();
    
    if( reset_reason == ESP_RST_POWERON )
    {
        state.a = false;   
        state.b = false;    
        state.thing = 100;    
    }
}
```

baldhead69, 2021-10-18:
Pressing the reset button is characterized as power on reset ?

I don't remove the power from the circuit.

RichFalk, 2021-10-18:
The restart reason you showed was ESP_RST_POWERON which is a power-on reset.  A software reset is SW_CPU_RESET which would retain NOINIT section data in RTC RAM.

The EN pin (that I presume the reset button is connected to in order to pull it to GND) is for a hardware reset equivalent to a power-on reset.  See [ESP32-DevKitC V4 Getting Started Guide - Functional Description](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html#functional-description).

If you want to save information over power cycles or hardware reset, you can use the non-volatile storage (NVS) system that stores data in Flash (there is no EEPROM in the ESP32).

baldhead69, 2021-10-18:
Hi, 

I only want to test the functionality of RTC_NOINIT_ATTR.

I assumed that hardware reset did not reset rtc ram, only the power cycle would reset the rtc ram memory, for this reason my if no reset reason.  I want to preserve the data structure in a possible system crash/reset.

Thank's.

baldhead69, 2021-10-31:
Thank`s @RichFalk 

it's working now with "esp_restart();".

My goal is to restore the system to the state it was in before the crash, so that the system user has a good impression of it if the system crashes(sporadic crash).

closed 2021-10-31 by baldhead69

rob040, 2025-08-04:
@RichFalk: Although this is an old issue, it appears still to be true.

Above, you state 'Power on reset does not preserve RTC RAM'. That appears to be true, but I like to understand why? What makes a power-on reset (HW reset pulling the EN pin low) different from an SW restart ? 
In the __NOINIT_ATTR and RTC_NOINIT_ATTR locations I read some pseudo random data after HW reset. Is it some bootloader that uses that memory? Is it because it is MMU mapped?
What memory (-segment) can be expected to be undisturbed when HW reset occurs and device remains powered?

Background: I am making a double reset detector, in order to enter a configuration mode, or in general, enter an alternative startup mode. In my opinion, it is unwise to use NVS that involves flash writes in a moment of time when you can expect HW resets.