espressif/esp-idf #7731: v4.3.1 long time running crashed

achao1104, 2021-10-19:
Hello,

idf version is v4.3.1 and we used esp32s2 modules.
we have made up 3 devices, and keep them running for a long time, after around 30 hours running, one of them crashed.
we have checked the dump content, it looks it is not due to our applicaiton codes, please help.

D:\project\esp32\project\solardevice>python D:\project\esp32\v4.3.1\components\espcoredump\espcoredump.py info_corefile  -t raw -c D:\project\esp32\temp\a.dump D:\project\esp32\project\solardevice\build\solardevice.elf
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ff9f36c, name: 'main', GDB name: 'process 1073345388'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x40089320
pc             0x4002edb9       0x4002edb9 <panic_abort+21>
lbeg           0x0      0
lend           0x0      0
lcount         0x0      0
sar            0x8      8
ps             0x60c20  396320
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8002f58c       -2147289716
a1             0x3ffdacb0       1073589424
a2             0x3ffdacf0       1073589488
a3             0x3ffdad3d       1073589565
a4             0xa      10
a5             0x3f00a435       1057006645
a6             0x40000aa0       1073744544
a7             0x0      0
a8             0x0      0
a9             0x1      1
a10            0x3ffdad3d       1073589565
a11            0x3ffdad3d       1073589565
a12            0xa      10
a13            0xa      10
a14            0x3ffdac99       1073589401
a15            0xff000000       -16777216

==================== CURRENT THREAD STACK =====================
#0  0x4002edb9 in panic_abort (details=0x3ffdacf0 \"abort() was called at PC 0x40019fbe on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/panic.c:379
#1  0x4002f58c in esp_system_abort (details=0x3ffdacf0 \"abort() was called at PC 0x40019fbe on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/system_api.c:112
[... 114 lines cut ...]

===================== ESP32 CORE DUMP END =====================
===============================================================
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1238, in _readerthread
    buffer.append(fh.read())
OSError: [Errno 22] Invalid argument

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1238, in _readerthread
    buffer.append(fh.read())
OSError: [Errno 22] Invalid argument

WARNING: Attempt to terminate the GDB process failed, because it is already terminated. Skip
Done!

AxelLin, 2021-11-14:
@o-marshmallow 
Any finding?

AxelLin, 2022-01-16:
Hi @achao1104 
Does this issue still happen on v4.3.2?

achao1104, 2022-01-16:
we  are using 4.3.1,not sure about 4.3.2

o-marshmallow, 2022-01-17:
Hi @achao1104 @AxelLin ,
To me this sounds like a memory/heap corruption, can you try cherry-picking the commit `d300a9cfe3bf09f83448b9c9075c2adccebac7d2` and try again?

AxelLin, 2022-02-11:
@achao1104  Any update regarding this issue? (Did you try with https://github.com/espressif/esp-idf/issues/7731#issuecomment-1014089279 ?)

o-marshmallow, 2022-03-16:
I am closing this issue, if you still have this issue, feel free to re-open

closed 2022-03-16 by o-marshmallow