espressif/esp-idf #7734: ESP32 c3 wdt reboot fail

u-tec-com, 2021-10-20:
use C3 in battery/low power product. The wdt reboot system in some situation. But reboot fail

u-tec-com, 2021-10-20:
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalÿESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c0dc
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f

Alvin1Zhang, 2021-11-19:
Thanks for reporting, would you please help share if more details for the issue? Thanks.

AxelLin, 2022-02-13:
@Alvin1Zhang 
What could be the reason to cause "invalid header: 0xffffff7f"?

I had similar issue on a esp32-wroom-32e module:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
invalid header: 0xaacc9cbb
invalid header: 0xaacc9cbb

I couldn't resolve the issue (cannot reprogram the module any more) and I cannot reproduce the issue on other devices, then I just throw away that module.
But I'm curious what can cause the "invalid header"?

bbinet, 2022-04-04:
I face the exact same issue as @u-tec-com, with same symptoms:
```
...
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid heade�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
invalid header: 0xffffff7f
...

```
We also power our ESP32 C3 on battery while trying to minimize consumption by user the Power Management module with the following configuration:
```
    esp_pm_config_esp32c3_t pm_config = {
        .max_freq_mhz = 160,
        .min_freq_mhz = 10,
        .light_sleep_enable = true,
    };
```
I don't know what has initially triggered the watchdog, but after running normally during about a day, the ESP32 C3 is now is a reboot (RTCWDT_RTC_RESET) loop. If I manually reset the ESP32 C3, it restarts normally though.

As a workaround, is there a way to configure the watchdog to act as a normal reset (like if we reset the ESP32 C3 manually), waiting for this problem to be solved?

FYI, we use a ESP32-C3-WROOM-02 module, with ESP-IDF master branch.

bbinet, 2022-04-14:
The issue has just occurred a second time on the same esp32c3 after a few days where it was working just fine.
@Alvin1Zhang do you know what does mean "invalid header: 0xffffff7f" ?

esp-wzh, 2022-05-19:
Hi, @bbinet , when [first-stage-bootloader](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/startup.html#first-stage-bootloader) finishes executing and then loads data Second stage bootloader from the flash, it will verify Second stage bootloader's header, and if it fails, it will prompt “invalid header: ...”, as you have encountered.

This error is usually caused by reading the wrong data from the flash, based on the above information this is difficult to reproduce, I suspect the problem is on the flash side, can you reproduce the above failure with USB powered?
Another suggestion is that you can configure the flash wait time after power on through eFuse, you can search for `EFUSE_FLASH_TPUW` in the technical reference manual for more information.

For software, `RTCWDT_RTC_RESET` is already the highest level of reset, it will reset the entire system and RTC domain.

I am curious about if you have been stuck in the reset caused by "invalid header: 0xffffff7f", the reset reason should be `TG0WDT_SYS_RST`, how do you get `RTCWDT_RTC_RESET`. Can you show more logs?

esp-wzh, 2022-05-19:
One more question, have you enabled flash encryption ? @bbinet

bbinet, 2022-05-20:
Thank you @esp-wzh for your support.

No, flash encryption is not enabled.

I don't know why I get `RTCWDT_RTC_RESET` in the logs and not `TG0WDT_SYS_RST`, and I don't really have more logs to show as my esp32c3 is stuck showing `invalid header: 0xffffff7f`, and reset regularly by the watchdog with RTCWDT_RTC_RESET, but the watchdog does not fix the issue:
```
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid heade�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
invalid header: 0xffffff7f
[many more invalid header: 0xffffff7f]
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid heade�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
invalid header: 0xffffff7f
[many more invalid header: 0xffffff7f]
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid heade�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
invalid header: 0xffffff7f
[many more invalid header: 0xffffff7f]
```
The esp32c3 restarts normally only when I manually reset the esp32c3 through the EN pin: how do the RTCWDT_RTC_RESET compares to a manual EN pin reset (what are the differences)?

Regarding configuring EFUSE_FLASH_TPUW in eFuse, which value should I set? And could you explain how to proceed to write to the eFuse?

Also, I will try to power my esp32c3 board from USB during a few days, and see if I can reproduce the issue.

esp-wzh, 2022-05-20:
@bbinet thanks for the reply!
&emsp;&emsp;A difference between RTCWDT_RTC_RESET and POWERON_RESET is that  when you click the reset button it wil re-power the FLASH. So I think this problem is most likely caused by FLASH entering some unknown error state. Have you only had this problem with one module so far? Or encountered it on multiple modules?

(Considering these I don't think burn EFUSE_FLASH_TPUW is effective for this problem, please ignore this suggestion)

bbinet, 2022-05-20:
@esp-wzh Ok so if POWERON_RESET re-power the FLASH and fixes the issue, is there a way to configure RTCWDT_RTC_RESET to do the same and re-power the FLASH too?

For the moment we encounter this issue with a single esp32c3 (but we are in a prototype phase so we have tested only 2 of them).
Also @u-tec-com, on how many modules do you face this issue?

esp-wzh, 2022-05-20:
@bbinet, Sorry it doesn't support re-power FLASH by watchdog because FLASH is independent of esp32c3 chip.

Mosher23, 2022-05-21:
I am experiencing same issue on my M5Stack Stamp-C3 board.
After flashing with ESPHome Web tool i receive following logs.
[image]

esp-wzh, 2022-05-21:
Hi, @Mosher23 , Thanks for your feedback!
As far as I know ESPHome does not currently support esp32c3, how did you flash it?

Mosher23, 2022-05-21:
Hello @esp-wzh, 
i flashed it using ESP Web and compiled on my ESPHome instance 2022.05. I can choose C3 from board selection.

esp-wzh, 2022-05-21:
Judging from your log, it is because a bin that does not meet the requirements of the [ESP Firmware Image Format](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/firmware-image-format.html#firmware-image-format), or it is downloaded to the wrong location in FLASH.
Maybe you can erase the flash and try it again, If it still doesn't work, you can create  issue in the [esphome](https://github.com/esphome/esphome) project!

Mosher23, 2022-05-21:
> Judging from your log, it is because a bin that does not meet the requirements of the [ESP Firmware Image [...]

Could you give some hints on how to erase it?

esp-wzh, 2022-05-21:
@Mosher23 , You can erase flash with esptool. [Install](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/installation.html#installation) it and run 
```
esptool.py erase_flash
```

Mosher23, 2022-05-21:
Thank you!
I have just tried it and it doesn't work. 
I get a fatal errror

`esptool.py erase_flash
esptool.py v4.0
Found 2 serial ports
Serial port /dev/cu.usbmodem51850299911
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:a8:d5:68
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)`

esp-wzh, 2022-05-21:
But it works fine on my side... It's so weird, maybe there is a problem with the boot mode switching circuit of your dev-board, do you have any additional boards encountering these problems? @Mosher23

Mosher23, 2022-05-21:
@esp-wzh I have zero problems with other boards. But it is also my first C3 board.

HiFiPhile, 2022-06-17:
I've met 2 times this issue. 

- IDF version: v5.0-dev-3055-g179ea878c7
- Build System: idf.py
- Compiler version: gcc version 8.4.0 (crosstool-NG esp-2021r2-patch3)
- Operating System: Windows

## 1st time
- Module or chip used: Customized PCB ESP32-C3-MINI-1
- Power Supply: Energy harvesting
- No flash encryption

## 2nd time
- Module or chip used: Customized PCB ESP32-C3-WROOM
- Power Supply: USB 5V
- Flash Encryption enabled

esp-wzh, 2022-06-20:
Hi~@HiFiPhile,

&emsp;I think the first thing to do is to confirm whether the data in the flash is still correct

&emsp;For the first times, whether the boot is successful after powering on again? If it always prompts invalid header error. You can try to use the `read_flash` command of `esptool` to dump the data in the flash and compare it with the bin you downloaded to check whether the data in the flash is damaged.

&emsp;You can also use the same method to check the chip with flash encryption enabled, but you need to use the `encrypt_flash_data` command of `espsecure` to encrypt the plaintext bin and then compare it with the dumped bin

HiFiPhile, 2022-06-20:
>  I think the first thing to do is to confirm whether the data in the flash is still correct

The flash data is correct. The first time I recovered it with a manual reset.

Yesterday it happened again with my 2nd board, this time a reset with idf_monitor recovered it. Unfortunately I didn't activate log to file so I don't know the cause of reset.

I'm going to open the shield of ESP32-C3-WROOM and hook a logic analyzer into SPI flash and wait next failure, hope I don't need to wait 2 weeks:
- If the timing is wrong, I suspect sleep gpio workaround.
- if the timing is right, it could be flash MMU & cache.

HiFiPhile, 2022-06-22:
Hi @u-tec-com @bbinet ,

Have you seen anymore this issue ? 

I've set up logic analyzer & monitor log to wait the issue to trigger. I've only seen 3 times this issue in 4 months, I hope this time it won't take too much time.

bbinet, 2022-06-22:
@HiFiPhile  Since I've changed my code to deep sleep most of the time, I've not seen the issue anymore.

bbinet, 2022-06-27:
I've finally reproduced the same issue on another esp32-c3 which was also in deep sleep most of the time.
This new esp32-c3 was powered from an AC-DC converter (220V to 5V).

@esp-wzh So I have reproduced the issue on 2 different esp32-c3, either battery powered or powered from the mains...

niwantha33, 2022-06-30:
### Try manually 

### check revision (Chip is ESP32-C3 (revision 3) )

```esptool.py --chip auto --port /dev/ttyUSB0 chip_id```

#### Detected flash size: 4MB

``` esptool.py --chip auto --port /dev/ttyUSB0 flash_id```

#### Flash 

```esptool.py --chip esp32c3 --port /dev/ttyUSB0 write_flash -z 0x0 [firmware]```

HiFiPhile, 2022-06-30:
@niwantha33 hi, did you meet this issues ?
What you posted seems off-topic.

esp-wzh, 2022-07-08:
@HiFiPhile , Yes this is a possible reason if you power down the flash during sleep. Usually Flash power down has timing requirements, take XM25QH32C datasheet as an example,
[image]
You must at least make the Flash voltage drop below 0.6V and keep it for at least 100us before the Flash can be reset normally.

It is also applicable to other scenarios where the flash is powered off before the chip starts.

bbinet, 2022-07-08:
> @esp-wzh I saw [d3c6fff](https://github.com/espressif/esp-idf/commit/d3c6fff25aed52b9082286ad8621054ed39f7cd4), is it related to this issue ?
> 
> I've flash power down enabled, how about you @bbinet ?

Yes, me too: `CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=y`

bbinet, 2022-07-08:
I'm going to set `CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=n` and see if I still reproduce the issue.

@esp-wzh CONFIG_ESP_SLEEP_POWER_DOWN_FLASH is only used for light sleep, not deep sleep, right?

esp-wzh, 2022-07-11:
Yes, this option is only valid for lightsleep, it will perform strict checks before lightsleep to ensure that the sleep time is longer than the power-down time required by Flash, otherwise the device will still not power-down. And deepsleep IDF will definitely power down the Flash, it is necessary to manually evaluate the risk that the sleep time is less than the time required for the Flash to power down.([d3c6fff](https://github.com/espressif/esp-idf/commit/d3c6fff25aed52b9082286ad8621054ed39f7cd4)  The ``` `esp_deep_sleep_start` will not power down flash``` described in this commit is wrong, it will be corrected later)

HiFiPhile, 2022-07-11:
I captured the voltage of VDD_SPI of ESP32-C3-WROOM module on my board, it turns out it violate heavily the timing of XM25QH32C.

Even though I still can't confirm it's the cause of this issue, this timing is problematic.

[image]

Lowest voltage is about 680mV so a chip reset is not guaranteed because the 1uF capacitor.

@esp-wzh As not every user knows the capacitor of VDD_SPI, I suggest add the words of "DO NOT enable this option on ESP32-C3-WROOM or ESP32-C3-MINI-1" on flash power down option.

HiFiPhile, 2022-07-25:
I've to stop my test, for a month I couldn't make it to happen.

Now I stopped using CONFIG_ESP_SLEEP_POWER_DOWN_FLASH and hope it was indeed the issue.

@esp-wzh Do you have any comment regarding my oscilloscope capture ? Does Espressif knows VDD_SPI timing of ESP32C3 modules violate Flash specification with sleep power down ?

esp-zlq, 2022-07-27:
> I captured the voltage of VDD_SPI of ESP32-C3-WROOM module on my board, it turns out it violate heavily the timing of XM25QH32C.
>
> Even though I still can't confirm it's the cause of this issue, this [...]

@HiFiPhile Could you provide the test code, so that we can debug the waveform for more information?

HiFiPhile, 2022-07-27:
@esp-zlq 
[VDD_SPI.zip](https://github.com/espressif/esp-idf/files/9197395/VDD_SPI.zip)
It was tested under 179ea878c7, but I can't reproduce on HEAD 36f49f361c001b49c538364056bc5d2d04c6f321.
Once wifi is initiated VDD_SPI is always on regardless CONFIG_ESP_SLEEP_POWER_DOWN_FLASH.

esp-wzh, 2023-03-08:
We think that this problem is most likely caused by insufficient flash power-off time during lightsleep and we have updated the design guidelines on whether Flash is necessary to power down during sleep in the document. 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#power-down-of-flash

https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/Kconfig#L51

If there are no more problems and reports we will close this issue.

bbinet, 2023-03-08:
on my side, this issue is fixed since I've set CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=n

Alvin1Zhang, 2023-03-09:
Thanks for sharing the updates, feel free to reopen.

closed 2023-03-09 by Alvin1Zhang

vanwykc, 2023-04-20:
I've experienced similar boot fail behavior, BUT it happened at random power-up events.  The product is mains powered. Flipped the switch a number of times and then boot failed with:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c0d4
invalid header: 0xffffff7f
```

The above happened quite repeatably (and very frustratingly) with 1 fail every 10 to 15 power-cycles. 

After trying various things (including the above suggestion of `CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=n` , I changed `CONFIG_ESPTOOLPY_FLASHMODE` from "qio" to "dio".  
Built and flashed the bin.
Flipped the switch over 100 times without any boot failures.  

Interestingly though:

I noticed in my sdkconfig file:
Under # Serial flasher config:
`CONFIG_ESPTOOLPY_FLASHMODE_QIO=y`
`CONFIG_ESPTOOLPY_FLASHMODE="dio"`

AND

Under # Deprecated options for backward compatibility:
`CONFIG_FLASHMODE_QIO=y`

From previous bad experiences, I don't edit the sdkconfig directly any more, but stick to `idf.py menuconfig` rather.  I went through the settings, but couldn't find where I could change those remaining QIO references to "n" ... 

I'm hoping that the "dio" mode overrides the QIO=y options.

AxelLin, 2023-06-16:
@esp-wzh 

I just observed "invalid header: 0xffffff7f" once.
This time it seems boot ok as below boot log, and I cannot reproduce it again.
Is it safe to ignore it?

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d2c0
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x12dc
load:0x403cc710,len:0x9d8
load:0x403ce710,len:0x2c00
entry 0x403cc710
I (286) cpu_start: Unicore app
I (286) cpu_start: Pro cpu up.
I (470) cpu_start: Pro cpu start user code
I (470) cpu_start: cpu freq: 160000000 Hz
...

ordonezt, 2024-09-05:
@esp-wzh Hello, I have some confusions about the `EFUSE_FLASH_TPUW` EFuse.

Image taken from the [ESP32-C3 datasheet page 119](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#efuse):
[image]

> If the value is less than 15, the waiting time is the configurable value. Otherwise, the waiting time is always 30 ms

1. How can be more than 15 if the efuse is only 4 bits (maximum value is 15)

> Otherwise, the waiting time is always 30 ms

2. Is 30ms the default time? So if I change this EFuse I can only reduce the flash waiting time, not increase it, rigth?

ordonezt, 2024-09-05:
I'm having the same `invalid header: 0xffffff7f` issue.
I'm using an ESP32-C3FH4 module with [ESP32-C3-MINI-1-AT-V2.4.2.0](https://dl.espressif.com/esp-at/firmwares/esp32c3/ESP32-C3-MINI-1-AT-V2.4.2.0.zip) firmware (also tried with v3.3.0.0)

To reproduce the issue I just need to restart the esp32 module a couple of times (sometimes 4, sometimes more than 20) using the `CHIP_EN` pin.
[image]

Once the issue start I can't exit from that state until a power off / power on cycle, the `CHIP_EN` reset doesn't help.

Here I have some logs of the issue:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffff7f (Repeated 528 times over 1,28 seconds)
...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffff7f
...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT) (Every 3 seconds)
invalid header: 0xffffff7f
...
```
Should I try to program the `EFUSE_FLASH_TPUW` efuse?

esp-wzh, 2024-09-06:
Hi~ @ordonezt 

>  How can be more than 15 if the efuse is only 4 bits (maximum value is 15)

If and only if the configuration value is equal to 15, the delay time is 30ms.

>  Is 30ms the default time? So if I change this EFuse I can only reduce the flash waiting time, not increase it, rigth?

Default time is zero, the default value is 0, which means no waiting, because tsome software initialization has already taken up  about 5ms.

[image]

Both tPWD and tVSL violations can trigger this problem. On C3, I am more worried about tPWD violation, because as discussed in this thread, the reference design has a capacitor on VDD_SPI, which may cause the Flash chip power not drop below Vpwd voltage quickly when the chip is reset. 

When the chip needs to be reset, the low level time of CHIP_EN can be increased to avoid this problem.

esp-wzh, 2024-09-09:
> Do you have any estimated value on the minimum CHIP_EN low level time?

Theoretically, no configuration is required, but if your application does not mind the startup time, the larger the better (the maximum is 30ms)

> In the first set of pictures you can see the moment were the error ocurrs. The low level time of CHIP_EN is about 1500 ms

Does it mean that even if CHIP_EN lasts for 1.5 seconds, the error may still occur? This doesn't make sense.

[image]

Can you try to capture the voltage waveform on the VDD_SPI pin (with CHIP_EN pin) when the error occurs?

ordonezt, 2024-09-09:
> Does it mean that even if CHIP_EN lasts for 1.5 seconds, the error may still occur? This doesn't make sense.

Yes

> Can you try to capture the voltage waveform on the VDD_SPI pin (with CHIP_EN pin) when the error occurs?

Sure, here is a capture of `CHIP_EN` (yellow) and `VDD_SPI` (blue) when the error occurs, the total time of CHIP_EN on low level state is about 980 ms.
[image]

And here is the logic analyzer capture of the same pulse
[image]

And here I have a capture of a `CHIP_EN` 50ms pulse during the error state. `CHIP_EN` (yellow) and `VDD_SPI` (blue)
[image]

[image]

esp-wzh, 2024-09-10:
Does your hardware design have a power supply VDD_SPI PIN? Or is the VDD_SPI pin used as a GPIO? It seems that the FLASH chip is not powered off and up when the chip is reset. ESP32-C3FH4 has a built-in Flash, and when the chip is reset, it will also re-power on VDD_SPI to reset the Flash chip (SPI FLASH chip does not have a reset pin or SPI command for reset, and can only be reset by re-powering on. Usually, when the Flash chip is accidentally interrupted during a write operation, it needs to be reset to continue accessing)

ordonezt, 2024-09-13:
Thanks @esp-wzh! I can't reproduce the problem anymore after removing R220 🙌

Unfortunately we have several boards already produced and it would be very difficult to do the rework on all of them.

> Usually, when the Flash chip is accidentally interrupted during a write operation, it needs to be reset to continue accessing

Is there anything we can do to avoid this accidental interrupt? I'm thinking of putting the ESP32-C3 in [sleep mode](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Set/Basic_AT_Commands.html#at-sleep-set-the-sleep-mode) before applying a CHIP_EN reset.

esp-wzh, 2024-09-14:
@ordonezt  Yeah, setting it to sleep mode is one way to ensure that there is no ongoing Flash access. It is more recommended to use [deepsleep](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Set/Basic_AT_Commands.html#at-gslp-enter-deep-sleep-mode) mode.

ordonezt, 2024-09-16:
@esp-wzh deep sleep approach is not working, I'm still able to reproduce the invalid header problem.
Do you know how much should I wait to set CHIP_EN to low after the deep sleep command?

[image]

This is my procedure:

1. Set CHIP_EN to high and wait for ready AT command from ESP32
2. Configure BLE and start advertising
3. Wait 700ms
4. Send deep sleep for 10 seconds AT command (AT+GSLP=10000)
5. Wait around 2 seconds
6. Set CHIP_EN to low
7. Wait around 1.4 seconds
8. Set CHIP_EN to high

This is the log:
```
AT Sent: AT+GSLP=10000
AT Received: OK

UART0 ESP32: wifi deinited
UART0 ESP32: ble deinited

*RESET PULSE HERE*

UART0 ESP32: 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffff7f
invalid header: 0xffffff7f
...
```

esp-wzh, 2024-09-18:
> > Does it mean that even if CHIP_EN lasts for 1.5 seconds, the error may still occur? This doesn't make sense.
> 
> Yes
> 
> > Can you try to capture the voltage waveform on the VDD_SPI pin (with CHIP_EN pin) when the error occurs?
> 
> Sure, here is a capture of `CHIP_EN` (yellow) and `VDD_SPI` (blue) when the error occurs, the total time of CHIP_EN on low level state is about 980 ms. ![dso_invalid_header_vdd](https://private-user-images.githubusercontent.com/23056645/365683609-1e16e827-7565-4b78-8e67-1442f7b4b518.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MjczNjUsIm5iZiI6MTcyNjYyNzA2NSwicGF0aCI6Ii8yMzA1NjY0NS8zNjU2ODM2MDktMWUxNmU4MjctNzU2NS00Yjc4LThlNjctMTQ0MmY3YjRiNTE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDAyMzc0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkMzA4OTI2ZjVjYjE4NWNiZmI5OGNiZGZhZWE4MmNlOGM0NWMxNWY1MGMxODg3NTUxODc2OWE2YjgzNDQ5YmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Aozuq2Bg-hTVMdDAN3Mp4z_AoER080ZKByFwi38Q3NM)
> 
> And here is the logic analyzer capture of the same pulse ![logic_invalid_header_vdd](https://private-user-images.githubusercontent.com/23056645/365683501-c3be7cc6-dc49-45f9-aa4f-f98d45a07cc3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MjczNjUsIm5iZiI6MTcyNjYyNzA2NSwicGF0aCI6Ii8yMzA1NjY0NS8zNjU2ODM1MDEtYzNiZTdjYzYtZGM0OS00NWY5LWFhNGYtZjk4ZDQ1YTA3Y2MzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDAyMzc0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3ZDNhYTE1ODAxODJjOWY0NjllYjhkOGViMTQxYjJhMGNlYTRhMDliYmM0Y2IwZGI3MmQ4ZjI2ZGI2NzFkYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.cQWhTu8_nUatC7E_p5UbgCTaOd_5xLb5D5FVe8Ekpv0)
> 
> And here I have a capture of a `CHIP_EN` 50ms pulse during the error state. `CHIP_EN` (yellow) and `VDD_SPI` (blue) ![dso_reset_50ms](https://private-user-images.githubusercontent.com/23056645/365680434-0d8ef31f-06f7-4476-86d9-d0947b2febf3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MjczNjUsIm5iZiI6MTcyNjYyNzA2NSwicGF0aCI6Ii8yMzA1NjY0NS8zNjU2ODA0MzQtMGQ4ZWYzMWYtMDZmNy00NDc2LTg2ZDktZDA5NDdiMmZlYmYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDAyMzc0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFkYjcwZmU5Y2VlNDI3ZjI2ODVkNjg1NTVjMTQ0MjVlYjYyZmM0ZGIxMjg1NTBkYWFlYTYyZGJlYWNjMGMxNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pCYiEeMfHVzEkvdAgmWr0_imC2fl8lVX62471jxOx_E)
> 
> ![image](https://private-user-images.githubusercontent.com/23056645/365680758-7db78588-0f5c-4ed2-9305-64e0bec10cee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MjczNjUsIm5iZiI6MTcyNjYyNzA2NSwicGF0aCI6Ii8yMzA1NjY0NS8zNjU2ODA3NTgtN2RiNzg1ODgtMGY1Yy00ZWQyLTkzMDUtNjRlMGJlYzEwY2VlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDAyMzc0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxYTVlZTZjOTVlYTUwZmZlMzIyYjg0NmI1Mzg0NWViN2I2M2ZiMjI5NDA5YTYzYjQ5Y2YyNzJhYzExNjZkNzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pjYy9had8gY4Dy0_UNWjcJZzBfI2RI-KRB3KVD54wpI)

@ordonezt  In this figure, notice that the external power supply voltage of VDD_SPI is 3.8V, which exceeds the **Operating Range** limits in the datasheet. I suspect this problem may be related to the voltage. Can the voltage be lowered to within 3.6V for testing?

[image]