espressif/esp-idf #7746: Consistent crash in block_trim_free (heap_tlsf.c : 377)

ammaree, 2021-10-21:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3540-g4e03a9c34c
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp-2021r1) 8.4.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description
With latest version of IDF FW crash with error log as below. 

### Expected Behavior
Firmware to start and run as per usual

### Actual Behavior
Firmware crashes

### Steps to reproduce
Cannot share the application but nothing was changed from before IDF upgrade to latest master

## Debug Logs
```
assert failed: block_trim_free heap_tlsf.c:377 (block_is_free(block) && "block must be free")

Backtrace:0x40081f85:0x3ffcfae00x400906cd:0x3ffcfb00 0x400957f5:0x3ffcfb20 0x400949af:0x3ffcfc50 0x4009541f:0x3ffcfc70 0x400821e1:0x3ffcfc90 0x40082299:0x3ffcfcb0 0x401666c0:0x3ffcfd00 0x4017036f:0x3ffcfd20 0x4016bc01:0x3ffcfd40 0x4016ed21:0x3ffcfd70 0x4016f481:0x3ffcfd90 0x401656b1:0x3ffcfdb0 0x40165709:0x3ffcfdd0 
0x40081f85: panic_abort at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/esp_system/panic.c:402

0x400906cd: esp_system_abort at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/esp_system/esp_system.c:129

0x400957f5: __assert_func at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/newlib/assert.c:85

0x400949af: block_trim_free at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/heap_tlsf.c:377
 (inlined by) block_prepare_used at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/heap_tlsf.c:460
 (inlined by) tlsf_malloc at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/heap_tlsf.c:758

0x4009541f: multi_heap_malloc_impl at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/multi_heap.c:197

0x400821e1: heap_caps_malloc at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/heap_caps.c:145

0x40082299: heap_caps_malloc_prefer at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/heap/heap_caps.c:226

0x401666c0: mem_malloc at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x4017036f: handle_dhcp at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:983

0x4016bc01: udp_input at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/core/udp.c:404

0x4016ed21: ip4_input at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:778

0x4016f481: ethernet_input at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186

0x401656b1: tcpip_thread_handle_msg at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/tcpip.c:180

0x40165709: tcpip_thread at /Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
```

## Other items if possible

- [Y ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7388486/sdkconfig.txt)

AxelLin, 2021-10-22:
@ESP-YJM 
FYI, this looks like another regression report of https://github.com/espressif/esp-idf/issues/7742.

ESP-YJM, 2021-10-22:
@AxelLin Yes, Thanks for providing such important information.

AxelLin, 2021-11-15:
This should be fixed by 8987393

Alvin1Zhang, 2021-11-22:
Thanks for reporting, the fix is available, feel free to reopen.

closed 2021-11-22 by Alvin1Zhang