espressif/esp-idf #7750: Support other address filtering modes for KSZ8851SNL Ethernet driver, e.g., "Perfect with Multicast address passed"

p-ranav, 2021-10-21:
**Is your feature request related to a problem? Please describe.**

This feature request is related to using multicast send/receive on the KSZ8851SNL ethernet driver. Currently, we're able to set the address filtering to either PROMISCUOUS or HASH-PERFECT using:

```cpp
  {
    // Enable promiscuous mode
    // All Rx frames are passed without any conditions.
    auto eth_set_promiscuous = true;
    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_PROMISCUOUS, &eth_set_promiscuous));
  }
```

The driver supports a number of other modes which I am unable to set using the current driver API:

[image]

In order to support multicast without any MAC address filtering conditions, the driver implementation needs to support the address filtering mode "Perfect with multicast address passed" (Item 9 in the table above). 

**Describe the solution you'd like**

Add to the KSZ8851SNL driver API, providing the means to set the register values for `KSZ8851_RXCR1` register and enable the additional modes that the chip supports. 

**Describe alternatives you've considered**

The current solution for working with UDP multicast without MAC address filtering is to set the mode to promiscuous and receive all Rx frames without any conditions. This works for now but would be better if it can filter on the multicast group.

**Additional context**

* [KSZ8851SNL Datasheet](http://ww1.microchip.com/downloads/en/devicedoc/00002381a.pdf)
   - See `Section 3.3.6 Address Filtering Function` for more information on supported address filtering modes
* Current implementation allows to set promiscuous mode with [emac_ksz8851_set_promiscuous(...)](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_ksz8851snl.c#L537).

kostaond, 2021-11-03:
hi @p-ranav could you please have a look and let me know if proposed solution is acceptable for you? If so, I would start working on it.

p-ranav, 2021-11-03:
Thanks @kostaond for your response. I agree with your points. 

Yes! This solution works.

closed 2021-11-08 by espressif-bot