espressif/esp-idf #7799: [usb host] HUB: Configuration descriptor too small

chegewara, 2021-10-29:
Crash when read broken device descriptor. Im not sure the descriptor is broken, because if its true then problem is bigger, because in this case my test device is esp32 S3 rev0 in download mode.
commit e5f5eb3cbb5d6cf65e25dd8d64f1cced36f2ca49
board: esp32 S2 saola
Code basically works with other devices, just crash with esp32 S3 as device so far.

Here is code with additional logs:
```
    usb_config_desc_t *config_desc = (usb_config_desc_t *)(enum_urb->transfer.data_buffer + sizeof(usb_setup_packet_t));
    if (enum_urb->transfer.actual_num_bytes < sizeof(usb_setup_packet_t) + sizeof(usb_config_desc_t)) {
        ESP_LOGE(HUB_DRIVER_TAG, "Configuration descriptor too small");
        return ENUM_STAGE_CLEANUP_FAILED;
    }
```
and logs:
```
E (12562) HUB: Root port reset failed
E (12934) HUB: Configuration descriptor too small, actual_num_bytes: 8, expected: 17
I (12934) : 80 06 00 02 00 00 00 01 
```
Here is full crash log:
```
assert failed: usbh_hal_chan_request_halt IDF/components/hal/usbh_hal.c:294 (chan_obj->flags.active && !chan_obj->flags.error_pending)

Backtrace:0x400239e3:0x3ffd4c400x4002bcf1:0x3ffd4c70 0x4003479a:0x3ffd4ca0 0x400973c9:0x3ffd4dc0 0x4008c961:0x3ffd4df0 0x4008dfce:0x3ffd4e20 0x4008ec0e:0x3ffd4e50 0x4008eda7:0x3ffd4e80 0x4008f09e:0x3ffd4eb0 0x40089e85:0x3ffd4ee0 0x40089178:0x3ffd4f10 0x4002f44e:0x3ffd4f40 
0x400239e3: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:402

0x4002bcf1: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:129

0x4003479a: __assert_func at /home/chegewara/esp/master/components/newlib/assert.c:85

0x400973c9: usbh_hal_chan_request_halt at /home/chegewara/esp/master/components/hal/usbh_hal.c:294 (discriminator 1)

0x4008c961: _pipe_cmd_halt at /home/chegewara/esp/master/components/usb/hcd.c:1726

0x4008dfce: hcd_pipe_command at /home/chegewara/esp/master/components/usb/hcd.c:2023

0x4008ec0e: enum_stage_cleanup_failed at /home/chegewara/esp/master/components/usb/hub.c:451

0x4008eda7: enum_handle_events at /home/chegewara/esp/master/components/usb/hub.c:516

0x4008f09e: hub_process at /home/chegewara/esp/master/components/usb/hub.c:653

0x40089e85: usb_host_lib_handle_events at /home/chegewara/esp/master/components/usb/usb_host.c:490

0x40089178: client_async_seq_task(void*) at /home/chegewara/demos/github/usb_host/esp32-usb-host/components/usb_host/host/usb_host.cpp:31

0x4002f44e: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/port/xtensa/port.c:130
```

chegewara, 2021-10-29:
Ok, ive found problem.
Default value in menuconfig for `CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE` is very unfortunate. Value is 256 and seems to be problem with edge case on esp32 S3 and probably esp32 C3 too. After changing to 250 it didnt crash anymore.

@Dazza0 Can we change that default value to 255 or something?

Thanks

Dazza0, 2021-11-16:
@chegewara Thanks for reporting this issue. I've confirmed that it IS a bug with the USB Serial/JTAG Controller. Good news is that is should have been fixed on the mass production version of the ESP32-S3 (but that means the ESP32-C3 still has this bug). Basically, whenever a configuration descriptor request of exactly `256` bytes is sent by the Host, the USB Serial/JTAG Controller will just respond with a zero length data packet.

> @Dazza0 Can we change that default value to 255 or something?

We technically could (and that's what Windows does on enumeration). But I also want to support the cases where users shrink the size of the configuration descriptor. So I think I'll do split the configuration descriptor request into a pre-fetch then an actual request.

closed 2022-01-06 by espressif-bot