espressif/esp-idf #7802: esp_wifi_scan_start using blocking fires WIFI_EVENT_SCAN_DONE event

norventa, 2021-10-30:
The code below calls esp_wifi_scan_start with blocking = true and esp_wifi_scan_start with blocking = false.
Both cause WIFI_EVENT_SCAN_DONE to get fired.

This is contrary to: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html

`/*
Demonstrates esp_wifi_scan_start(&config, true) and esp_wifi_scan_start(&config, false)
both cause WIFI_EVENT_SCAN_DONE to get fired.

This is contrary to: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html

Program output:
status = 0, number = 4.
status = 0, number = 4.
*/

#include <string.h>
#include "nvs_flash.h"
#include "esp32/rom/ets_sys.h"
#include "esp_wifi.h"

//#include "freertos/FreeRTOS.h"

//#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
//#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_WIFI_SSID      "SSID"
#define EXAMPLE_ESP_WIFI_PASS      "PASSWORD"

/* Handler for all events */
static void all_event_handler(void* handler_args, esp_event_base_t base, int32_t id, void* event_data)
{
    if (base == WIFI_EVENT && id == WIFI_EVENT_SCAN_DONE)
    {
        const wifi_event_sta_scan_done_t *pdata = (const wifi_event_sta_scan_done_t *)event_data;

        if (pdata != 0)
            ets_printf("status = %u, number = %u.\n", pdata->status, (uint32_t)pdata->number);
    }
}

static void scan(bool block)
{
    wifi_scan_config_t config;

    config.ssid                 = 0;
    config.bssid                = 0;
    config.channel              = 0;
    config.show_hidden          = false;
    config.scan_type            = WIFI_SCAN_TYPE_ACTIVE;
    config.scan_time.active.min = 100;
    config.scan_time.active.max = 1500;
    ESP_ERROR_CHECK(esp_wifi_scan_start(&config, block));
}

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());

    // Create the default event loop
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    // Register the handler for all event. This will execute if either the timer events or the task iteration event
    // is posted to the default loop.
    ESP_ERROR_CHECK(esp_event_handler_instance_register(ESP_EVENT_ANY_BASE, ESP_EVENT_ANY_ID, all_event_handler, NULL, NULL));

    esp_netif_init();
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    wifi_config_t configSTA;
    memcpy(configSTA.sta.ssid, EXAMPLE_ESP_WIFI_SSID, sizeof EXAMPLE_ESP_WIFI_SSID);
    memcpy(configSTA.sta.password, EXAMPLE_ESP_WIFI_PASS, sizeof EXAMPLE_ESP_WIFI_PASS);
    configSTA.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;
    configSTA.sta.pmf_cfg.capable    = true;
    configSTA.sta.pmf_cfg.required   = false;

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &configSTA));
    ESP_ERROR_CHECK(esp_wifi_start());

    // block first so scan finishes.
    scan(true);
    scan(false);
}
`

mhdong, 2021-11-16:
hi @norventa  if block is true, this API will block the caller until the scan is done, otherwise it will return immediately. Both cause WIFI_EVENT_SCAN_DONE event is the desired result.
/home/muhaidong/Pictures/Screenshot from 2021-11-16 17-07-14.png

norventa, 2021-11-16:
Yes, I understand that that is desired, however the document:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html

says this:

WIFI_EVENT_SCAN_DONE
The scan-done event is triggered by esp_wifi_scan_start() and will arise in the following scenarios:
        The scan is completed, e.g., the target AP is found successfully, or all channels have been scanned.
        The scan is stopped by esp_wifi_scan_stop().
        The esp_wifi_scan_start() is called before the scan is completed. A new scan will override the current scan and a scan-done event will be generated.

**The scan-done event will not arise in the following scenarios**:
        **It is a blocked scan.**
        The scan is caused by esp_wifi_connect().

mhdong, 2021-11-16:
hi @norventa  It is a blocked scan mean something prevent the completion of scan, which has nothing to do with synchronous and asynchronous block.

Alvin1Zhang, 2021-11-24:
Thanks for reporting, feel free to reopen.

closed 2021-11-24 by Alvin1Zhang

mksafavi, 2024-06-13:
> hi @norventa It is a blocked scan mean something prevent the completion of scan, which has nothing to do with synchronous and asynchronous block.

the documentation is confusing on this. After reading the comments here, I understand that difference between a blocking scan  `esp_wifi_scan_start(NULL, true)` when `block=true` and a blocked scan that was aborted by `esp_wifi_connect`. but the terminology used in the docs is not consistent
Even the error log when you call esp_wifi_connect prints this: `wifi:sta_scan: STA is connecting, scan are not allowed!`

here I'm assuming the ` blocked scan` is used for an aborted on:

```
The scan-done event will not arise in the following scenarios:

        It is a blocked scan.

        The scan is caused by esp_wifi_connect()

```

But here the docs clearly uses `Blocked` for the blocking scan and says the event will not be emitted even though that's not how it's behaving:

[Scan in Blocked Mode](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#scan-in-blocked-mode)

If the block parameter of [esp_wifi_scan_start()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb) is true, then the scan is a blocked one, and the application task will be blocked until the scan is done. The blocked scan is similar to an unblocked one, except that no scan-done event will arise when the blocked scan is completed.