espressif/esp-idf #7803: PPPOS reconnection/disconnection faliuer after sometime.

AdiXPS, 2021-10-31:
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      Custom
- Module or chip used:  ESP32-PICO-D4
- IDF version 4.3
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: Yes | VS CODE
- Power Supply:         |Battery]

## Problem Description

I've tried ESP_cermak example, basiclly I found two main issues after leave it running for a couble of hours:

1-after awhile, I'am getting the famous "GGuru Mediation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled." One more thing here, I notice that its showing Core 1 while I dont have any task running on it..?

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d52e5  PS      : 0x00050031  A0      : 0x400826f8  A1      : 0x3ffb0dc0
0x400d52e5: uart_ll_get_intsts_mask at C:/esp4/esp-idf/components/hal/esp32/include/hal/uart_ll.h:170
 (inlined by) uart_rx_intr_handler_default at C:/esp4/esp-idf/components/driver/uart.c:746

0x400826f8: _xt_lowint1 at C:/esp4/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

A2      : 0x3ffb7760  A3      : 0x3ffb0aa0  A4      : 0x00000091  A5      : 0x8ca96391  
A6      : 0x00000910  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ffb0d50
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x80103e4e  A13     : 0x3ffb86c0  
A14     : 0x3ffb2f6c  A15     : 0x3ffb86e0  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400d52e2:0x3ffb0dc0 0x400826f5:0x3ffb0e10 0x40118563:0x3ffb8710 0x400d845b:0x3ffb8730 0x40088505:0x3ffb8750 0x4008a525:0x3ffb8770
0x400d52e2: uart_ll_get_intsts_mask at C:/esp4/esp-idf/components/hal/esp32/include/hal/uart_ll.h:170
 (inlined by) uart_rx_intr_handler_default at C:/esp4/esp-idf/components/driver/uart.c:746

0x400826f5: _xt_lowint1 at C:/esp4/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40118563: cpu_ll_waiti at C:/esp4/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/esp4/esp-idf/components/esp_pm/pm_impl.c:827

0x400d845b: esp_vApplicationIdleHook at C:/esp4/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40088505: prvIdleTask at C:/esp4/esp-idf/components/freertos/tasks.c:3839 (discriminator 1)

0x4008a525: vPortTaskWrapper at C:/esp4/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: d843f1542d225f73`

2-Mostly the crash is happening when stopping PPP only if there is existing connection. If for any reason the disconnection happen from host (ISP; no IP EVENT) its very stable.

`[0;32mI (511128) esp-modem: entering esp_modem_stop_ppp...[0m
[0;31mE (512058) esp-modem: esp_dte_handle_line(124): handle line failed[0m
[0;33mW (512058) pppos_example: Unknow line received: ~!E[0m
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`

Always getting Unknow line received: ~!E.

When I monitor the modem I am getting this transmitted:

`"10/30/2021 3:50:38 PM",CONNECT 150000000
"10/30/2021 3:50:38 PM",~ÿ}#À!}!¦} }9}"}&} } } } }#}%Â#}%}%}&Öcº÷}'}"}(}"£~~ÿ}#À!}"}!} }4}"}&} } } } }%}&M§T}$}'}"}(}"³ø~~ÿ}#À!}!§} }8}"}&} } } } }#}$À#}%}&Öcº÷}'}"}(}"í×~~ÿ}#À!}+¨} }(Öcº÷CC~~À#  ý0~~€!< jà~~€! 
"10/30/2021 3:50:39 PM", -ø0~~€!= ¶º~~€! 
"10/30/2021 3:50:39 PM",™ mTë7ƒTë9æÄG~~€! 
"10/30/2021 3:50:39 PM",™ mTë7ƒTë9æõÙ~~!E  ;å@ U­Ä
"10/30/2021 3:50:43 PM",E
[... 182 lines cut ...]
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400d52e2:0x3ffb0dc0 0x400826f5:0x3ffb0e10 0x40118563:0x3ffb8710 0x400d845b:0x3ffb8730 0x40088505:0x3ffb8750 0x4008a525:0x3ffb8770
0x400d52e2: uart_ll_get_intsts_mask at C:/esp4/esp-idf/components/hal/esp32/include/hal/uart_ll.h:170
 (inlined by) uart_rx_intr_handler_default at C:/esp4/esp-idf/components/driver/uart.c:746

0x400826f5: _xt_lowint1 at C:/esp4/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40118563: cpu_ll_waiti at C:/esp4/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/esp4/esp-idf/components/esp_pm/pm_impl.c:827

0x400d845b: esp_vApplicationIdleHook at C:/esp4/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40088505: prvIdleTask at C:/esp4/esp-idf/components/freertos/tasks.c:3839 (discriminator 1)

0x4008a525: vPortTaskWrapper at C:/esp4/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: d843f1542d225f73`
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

david-cermak, 2021-11-01:
@AdiXPS could you tell us which modem are you using? Also, are you seeing the first crash with just the code you've shared or was there anything else running on ESP32 at the same time?

As for the first issue, null dereference in the internal UART structures, I couldn't think of anything else but memory corruption. Is this happening randomly or could it be somehow related to the second issue? (like one occurs after another?)

As for the second problem, it might be possible that `+++` command won't exit the data mode, this is typically defined in the specification of the device used (and for our supported devices, the procedure is to repeat the sequence after one second)

AdiXPS, 2021-11-01:
Hi David,
Sure, I am using sim7070g, mostly like sim800. There is no other task is running other than connect and disconnect. 

`As for the first issue, null dereference in the internal UART structures, I couldn't think of anything else but memory corruption. Is this happening randomly or could it be somehow related to the second issue? (like one occurs after another?)`

Its randomly. I notice that its crashing on Core 1? not sure why though.. moreover, based on my observation, its not related as its rare..

`As for the second problem, it might be possible that +++ command won't exit the data mode, this is typically defined in the specification of the device used (and for our supported devices, the procedure is to repeat the sequence after one second)`

Its the same as sim800. i've added 1s delay before initiating ath command (based on last change you've proposed in other issue). But that did't prevent the issue from happening again..

Its happening randomly

david-cermak, 2021-11-02:
@AdiXPS 
> I notice one thing that modem in exiting PPP is still sometimes sending data by UART to esp even when confirming for +++..

This is weird, I've never seen anything like that with any device. Is the `OK` response surrounded by `<CR><LF>`? Could you please share binary hexdump?
Are you sure, you don't use any kind of virtual terminal feature/CMUX mode? Or isn't it that your device is somehow configured to multiplex terminals?

Anyway, I'll look into it and try to reproduce the issue, at least the first one. For the second one, we'd probably have to purchase your device `sim7070g`

AdiXPS, 2021-11-03:
> This is weird, I've never seen anything like that with any device. 
Yup, I totally agree with you.

>Is the `OK` response surrounded by `<CR><LF>`? 
Yes it is.

>Could you please share binary hexdump? 
I appreciate to get your advice regarding proper way to get the hexdump. 

>Are you sure, you don't use any kind of virtual terminal feature/CMUX mode? Or isn't it that your device is somehow configured to multiplex terminals?
I've double check, mux is not enabled. 

> Anyway, I'll look into it and try to reproduce the issue, at least the first one. For the second one, we'd probably have to purchase your device `sim7070g`
Thanks.. from what I can see that there are other issues raised here facing same issue, but no one actually tried to check the data coming from modem..

Alvin1Zhang, 2021-11-22:
@AdiXPS Thanks for reporting, would you please help share if any updates for the issue? Thanks.

AdiXPS, 2021-11-29:
@david-cermak @Alvin1Zhang I am still facing the same issue with different SIM7070G module. Different PCB also. I'll share the hexdump today.

AdiXPS, 2021-12-08:
@david-cermak , @Alvin1Zhang We'll, seems that the issue is the module 7070G/7080 have backdoor after full analysis. The modem keeps sending data even after exiting data mode, I have also requested to get the latest firmware from simcom (just to eliminate any M-in-M kind of issues) but its even worse now. Its getting certificates and all kind of alarming stuff:

`
"11/29/2021 10:49:10 PM",]߱-¹IW:~
"11/29/2021 10:49:11 PM",OK

"11/29/2021 10:49:15 PM",OK <<=== ATH confirmation..
"11/29/2021 10:49:15 PM",~!E xߴ@ 3ڴ¤Zi
"11/29/2021 10:49:15 PM",o»ʕس¼؈א <   J  F/· ʔi`󘩢=춒ͽ]½핉
"11/29/2021 10:49:15 PM",Ę À0      ÿ      #     h2] Y V û0÷0ߠ:B­G8ӅÀw¯ ª¹R0
"11/29/2021 10:49:15 PM",	*H÷
"11/29/2021 10:49:15 PM", 0Y10	UUS10U
"11/29/2021 10:49:15 PM",DigiCert Inc1301U*RapidSSL TLS DV RSA Mixed SHA256 2020 CA-10
"11/29/2021 10:49:15 PM",200810000000Z
"11/29/2021 10:49:15 PM",220811120000Z010U*.piojm.tech0"0
"11/29/2021 10:49:15 PM",	*H÷
"11/29/2021 10:49:15 PM",  0
"11/29/2021 10:49:15 PM", ­²qf6põ|5RB1<½;ވ¯䌌´¯r!ÀN$Vb✇Uڏ>ù^´­׫¤Zűü¯gͮ褔 ۟ݞøx&»񍒥}^p¸褼½ފ"11/29/2021 10:49:16 PM",+|8#5ɅώˮKհ
"11/29/2021 10:49:16 PM",^2¡»𽖀õ¤ûy¯W7kѨ¨ᔢ°e1u?H֌1.§(ٴک[¨2sJ䆤k5püU㣃��┸_5*2«גѷ?r¦(PA$¿٦¸ꌉO¹ùuO$ΌV$upŬP3讖¤»`?b_±¬9HĖ £û0÷0U#0¤徼y䰣m.)4­#Xܵ10U˶He�]𒯆ýmX噘0#U0*.piojm.tech
"11/29/2021 10:49:16 PM",piojm.tech0Uÿ 0U%0++0LU E0C07	`Hýl0*0(+https://www.digicert.com/CPS0g0+y0w0$+0http://ocsp.digicert.com0O+0Chttp://cacerts.digicert.com/RapidSSLTLSDVRSAMixedSHA2562020CA-1.crt0	U0 0}^
"11/29/2021 10:49:16 PM",+ֹnjh u )y¾𞹹!𖳟c¥w得}]`
"11/29/2021 10:49:16 PM",øùM]&\%]Ǆ  s׭üߠ  F0D (L8ö@²¹j߅K®¡A[Vþµڢ	Ժ¯ Q7´®QIY¹xµ髼E󌱓˦3趥l´䠷 "EEYU$V?¡/񷭆ࣦc­ÀK]ƃ\n⏂  s׭ý   H0F! Р¤;͠YþĞYM¹;'A®緌zȣjcOy! Rf䒄¹´<ùÀõ ͟&Bxӿ띠¯N^Ơv Q£°õýyVm¸7x¤z̛'˷B
"11/29/2021 10:49:16 PM",þԋ堠s׭ý[   G0E ={*é慦*%:䟙9��A-f/ПU¦4! ¢n³lt3꘠e�EZ)z»ù¶TIڈd0
"11/29/2021 10:49:17 PM",	*H÷
"11/29/2021 10:49:17 PM",þ5~~!E  4=+  7¡û%"11/29/2021 10:49:17 PM",o»˝µ󋚱ƕ
"11/29/2021 10:49:17 PM",º7  
"11/29/2021 10:49:17 PM",E½z'-¹IW@`

I am not interested in this aspect (backdoor or not), how can we ignore such coming data?

david-cermak, 2021-12-10:
Hi @AdiXPS 

The famous `~!..` is just an ascii representation of start of frame of PPP and an IP protocol id. This is what you've already pointed out above, that the device didn't exit the PPP session, but is still talking in PPP back. Couldn't be the backdoor data just public certs of some connection, e.g. forcefully retried after the hangup?
I couldn't reproduce the issue with the devices supported in IDF, which makes me think it has to be device specific, something to do with `7070G/7080` modems.

> how can we ignore such coming data?

I think you can try to reset the device, either using a reset command or pulling the reset line directly. Or you can try to use CMUX mode, but it depends on your usecase. What's the reason for switching to command mode? Do you need to use AT interface or close the network connection?

AdiXPS, 2021-12-12:
Hi @david-cermak 

>  Couldn't be the backdoor data just public certs of some connection, e.g. forcefully retried after the hangup?

AdiXPS, 2021-12-12:
Hi @david-cermak 

>  Couldn't be the backdoor data just public certs of some connection, e.g. forcefully retried after the hangup?
I highly doubt this, as we got confirmation from the module 'OK' (mentioned earlier in the logs) for exiting data and hangup, but the module keeps sending and receiving data. As long as we got ATH => OK this should kill any connection. Regarding if its public cert or not, moreover, even if its public cert, its no where mentioned all of their documentation regarding this behavior/activity.

> What's the reason for switching to command mode? Do you need to use AT interface or close the network connection?
I need to close the connection and get store the lon,lat by AT command. and do battery readings.. but as long as we keep getting such data uart will be unuasable

david-cermak, 2022-03-10:
> I need to close the connection and get the lon,lat by AT command.

You can use CMUX mode for that and keep the network/PPP still on.

david-cermak, 2022-04-01:
Hi @AdiXPS 

Is there any update from your side? This issue seems to be related to specifically your devices of `7070G/7080` family, that are not supported in IDF examples. Could you please close the issue, then? (note, that it's also okay to raise a feature request to support your devices, if you're thinking of that, please do so in the [`esp-protocols`](https://github.com/espressif/esp-protocols) repo)

closed 2022-06-29 by Alvin1Zhang