espressif/esp-idf #7806: esp-idf release/v4.3 分支；BLE例程编译报错

AzollaGit, 2021-11-01:
esp-idf:   commit 98d34e5f6db8904c31bd1e372d9896e4931d7a9b (HEAD -> release/v4.3, origin/release/v4.3)
环境： ubuntu20.04TLS
MCU: ESP32-C3
编译例程：
esp-idf/examples/bluetooth/bluedroid/ble/ble_spp_server
esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo

尝试删除代码定位错误点，找到以下代码编译会报错：
 esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
  ret = esp_bt_controller_init(&bt_cfg);   // 注释掉就不会报错了
  if (ret) {
      ESP_LOGE(GATTS_TABLE_TAG, "%s init controller failed: %s", __func__, esp_err_to_name(ret));
      return;
  }

今天执行git pull 后均报错以下信息：（以前版本是正常的）
bin/ld: /mnt/d/espressif/esp-idf_v4.3/components/esp_wifi/lib/esp32c3/libphy.a(phy_chip_v7.o): in function `phy_get_romfunc_addr':
(.text.phy_get_romfunc_addr+0x1a): undefined reference to `rom_temp_to_power'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

sunzigang, 2021-11-01:
我刚刚也遇到了同样的问题。   更新版本到v 4.3.1-245-gba15ac8634 后， esp_bt_controller_init接口崩溃。 
[image]

我在加了一些源码的调试打印后， 发现btdm_osi_funcs_register 接口导致的问题。
[image]

芯片为ESP-WROOM-32

xiewenxiang, 2021-12-03:
不能复现该问题，请把本地的 build 目录以及 sdkconfig 删除，重新编译

AzollaGit, 2021-12-14:
删除 本地的构建目录以及 sdkconfig ，重新编译就可以了，thanks !

closed 2021-12-14 by AzollaGit