espressif/esp-idf #7807: Hardware description needed for PPPOS Client using BG96

Vaishnavibhavaa, 2021-11-01:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [|ESP Command Prompt|powerShell].
- Using an IDE?: [Yes]
- Power Supply:         [USB]

## Problem Description

Not able to get hardware connection details for PPPOS client example. Pin mapping for RX,TX, RTS,CTS is not available.

Error:
E (970) esp-modem: esp_modem_dte_send_cmd(305): process command timeout
E (970) dce_service: esp_modem_dce_sync(48): send command failed
E (970) bg96: bg96_init(461): sync failed

Not able to send AT commands to BG 96
[image]
[image]

david-cermak, 2021-11-02:
Hi @Vaishnavibhavaa 

My guess is that your modem device needs an external power, so the red wire from your `ESP32-WROOM` kit is not enough to provide sufficient power supply to your `BG96`.
Connecting is described in the [example README](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client#pin-assignment) and could be changed in the menuconfig. If you leave the UART configuration unset, the default connection is simply:

| gpio | function |
|-- | --|
| GPIO25 | RX |
 |GPIO26 | TX |
| GND | GND |

(no need to configure and connect `RTS` and `CTS` if flow-control disabled)

AxelLin, 2021-11-06:
> BTW i didnt test too much with recent modem pppos, but it works better to me with code from before last update

What issue did you hit with last update?

Alvin1Zhang, 2022-03-11:
Thanks for reporting, feel free to reopen. Thanks.

closed 2022-03-11 by Alvin1Zhang