espressif/esp-idf #7808: heap_caps.c:heap_caps_get_largest_free_block() returns incorrect value

glenn20, 2021-11-01:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version(s): v4.2.1, v4.2.2, release/v4.4, v5.0-dev-79-gbcbef9a8db
- Build System:         idf.py
- Compiler version: 8.4.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

[heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)](https://github.com/espressif/esp-idf/blob/bcbef9a8db54d2deef83402f6e4403ccf298803a/components/heap/heap_caps.c#L438) always returns a value which is the largest available allocatable block rounded down to the nearest power of 2. This is in contradiction with the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv432heap_caps_get_largest_free_block8uint32_t) and the semantics prior to implementation of the TLSF allocator.

The error arises from this line in the [multi_heap_get_info_impl()](https://github.com/espressif/esp-idf/blob/bcbef9a8db54d2deef83402f6e4403ccf298803a/components/heap/multi_heap.c#L374) function:
```c
    info->largest_free_block = info->largest_free_block ? 1 << (31 - __builtin_clz(info->largest_free_block)) : 0;
```

The semantics of this function were changed in commit [bd9b921713d80](https://github.com/espressif/esp-idf/commit/bd9b921713d80f4b19931f4d94f38e87881f04cd#diff-8c61190e93dceec614cb85cc9c245bedbc37970dcf424fb4b90bb901dda7ade8) in October 2020 as part of the implementation of the TLSF allocator. (See [multi_heap.c:L374](https://github.com/espressif/esp-idf/blob/bcbef9a8db54d2deef83402f6e4403ccf298803a/components/heap/multi_heap.c#L374))

This bug has a significant impact on the Micropython port to all ESP32 platforms by significantly reducing the amount of RAM available to the Micropython heap when built against any IDF version since the TLSF allocator was implemented.

### Expected Behavior

`heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)`  should return the size of the largest allocatable block - in agreement with the documentation and the semantics prior to implementation of the TLSF allocator.

### Actual Behavior

`heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)`  returns the size of the largest allocatable block rounded down to the nearest power of 2.

igrr, 2021-11-01:
@glenn20 Thanks for reporting the issue. We did run into the same issue recently, and there is a fix submitted for this internally. Please try changing that line as follows:
```c++
    if (info->largest_free_block) {
        uint32_t sl_interval = (1 << (31 - __builtin_clz(info->largest_free_block))) / SL_INDEX_COUNT;
        info->largest_free_block = info->largest_free_block & ~(sl_interval - 1);
    }
```

glenn20, 2021-11-01:
@igrr I can confirm that fix works for me and resolves the issue for the Micropython ESP32 port. Will this fix be applied to updated v4.XX release tags, or only to future v5.X releases?

igrr, 2021-11-04:
Should be fixed on master branch in https://github.com/espressif/esp-idf/commit/0028e2c23c74872f9e424652a4758a358666b797, we will backport to release branches as well.

closed 2021-11-16 by glenn20

philippe44, 2021-11-27:
Thanks for fixing that - Took me a while to figure out because I was trying to reserve 65535 bytes for DMA. Because of this issue, it  split the block in two, but ended creating a pool of 65524 and ... trying to create another of 11 which failed obviously!