espressif/esp-idf #7809: install.sh does not work on Arch Linux

0xFEEDC0DE64, 2021-11-01:
The install.sh script has been working on my machine some time ago with older revisions of esp-idf. Running from latest master

```
$ ./esp-idf/install.sh 
Detecting the Python interpreter
Checking "python" ...
Python 3.9.7
"python" has been detected
Installing ESP-IDF tools
Selected targets are: esp32s3, esp32c3, esp32, esp32s2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20210902 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/feedc0de/.espressif/python_env/idf4.4_py3.9_env
ModuleNotFoundError: No module named 'virtualenv.activation.nushell'
Traceback (most recent call last):
  File "/home/feedc0de/bobbycar/bobbycar-boardcomputer-firmware/esp-idf/tools/idf_tools.py", line 1819, in <module>
    main(sys.argv[1:])
  File "/home/feedc0de/bobbycar/bobbycar-boardcomputer-firmware/esp-idf/tools/idf_tools.py", line 1815, in main
    action_func(args)
  File "/home/feedc0de/bobbycar/bobbycar-boardcomputer-firmware/esp-idf/tools/idf_tools.py", line 1519, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', '--seeder', 'pip', idf_python_env_path],
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--seeder', 'pip', '/home/feedc0de/.espressif/python_env/idf4.4_py3.9_env']' returned non-zero exit status 1.
```

dobairoland, 2021-11-01:
Hi  @0xFEEDC0DE64. I'm using ESP-IDF on multiple Arch Linux machines and I haven't noticed anything like this.

Could you please share the output of the following commands?

```sh
/usr/bin/python -m virtualenv --version
pacman -Qs virtualenv
/usr/bin/python -m pip freeze
```

Thanks!

0xFEEDC0DE64, 2021-11-01:
```
$ /usr/bin/python -m virtualenv --version
virtualenv 20.2.2 from /home/feedc0de/.local/lib/python3.9/site-packages/virtualenv/__init__.py
```

```
$ pacman -Qs virtualenv
local/python-pipenv 2021.5.29-1
    Sacred Marriage of Pipfile, Pip, & Virtualenv.
local/python-virtualenv 20.8.0-1
    Virtual Python Environment builder
local/python-virtualenv-clone 0.5.7-1
    A script for cloning a non-relocatable virtualenv.
```

```
$ /usr/bin/python -m pip freeze
aiofiles @ file:///build/python-aiofiles/src/aiofiles/dist/aiofiles-0.7.0-py3-none-any.whl
aiohttp==3.7.4.post0
aiohttp-cors==0.7.0
aiohttp-socks==0.4.2
aiorpcX==0.18.7
airdrop-ng==1.1
airgraph-ng==1.1
ajsonrpc==1.1.0
alabaster==0.7.12
aniso8601==9.0.1
anyio==3.3.4
anytree==2.8.0
apipkg==1.5
appdirs==1.4.4
asciinema==2.1.0
asgiref==3.4.1
asn1crypto==1.4.0
astral==2.2
astroid==2.8.4
async-timeout==3.0.1
attrs==21.2.0
awesomeversion==21.10.1
Babel==2.9.1
backcall==0.2.0
bcrypt==3.2.0
Beaker==1.11.0
beautifulsoup4==4.9.3
binwalk==2.3.3
bitstring==3.1.9
bottle==0.12.19
breezy==3.2.0
Brlapi==0.8.3
btrfsutil==5.14.2
CacheControl==0.12.6
cachetools==4.2.4
ccsm==0.8.18
ceph==1.0.0
ceph-volume==1.0.0
cephfs==2.0.0
cephfs-shell==0.0.1
certifi==2021.10.8
cffi==1.14.6
chardet==4.0.0
charset-normalizer==2.0.4
cheroot==8.5.2
CherryPy==18.6.1
ciso8601==2.2.0
click==8.0.3
colorama==0.4.4
conan==1.37.0
configobj==5.1.0.dev0
construct==2.10.67
contextlib2==0.6.0.post1
coverage==5.5
croniter==1.0.15
cryptography==35.0.0
cycler==0.10.0
Cython==0.29.24
decorator==5.1.0
defusedxml==0.7.1
deprecation==2.1.0
distlib==0.3.3
distro==1.6.0
dnspython==2.1.0
docker==5.0.3
docker-pycreds==0.4.0
docutils==0.17.1
ecdsa==0.17.0
elasticsearch==7.8.1
Electrum==4.1.3
fasteners==0.16
filelock==3.0.12
Flask==2.0.2
Flask-RESTful==0.3.9
fusion-icon==0.2.4
future==0.18.2
geographiclib==1.52
geopy==2.2.0
Glances==3.2.3
google-api-core==2.1.0
google-api-python-client==2.24.0
google-auth==1.34.0
google-auth-httplib2==0.1.0
googleapis-common-protos==1.53.0
greenlet==1.1.2
grpcio==1.41.1
gwebsockets==0.7
h11==0.12.0
homeassistant==2021.10.5
html2text==2020.1.16
html5lib==1.1
httpcore==0.13.7
httplib2==0.19.1
httptools==0.3.0
httpx==0.19.0
idna==3.2
ifaddr==0.1.7
imagesize==1.2.0
importlib-metadata==4.8.1
iniconfig==1.1.1
iotop==0.6
ipython==7.28.0
ipython-genutils==0.2.0
isc==2.0
isodate==0.6.0
isort==5.9.3
itsdangerous==2.0.1
jaraco.classes==3.2.1
jaraco.collections==3.3.0
jaraco.functools==3.3.0
jaraco.itertools==6.0.1
jaraco.logging==3.1.0
jaraco.stream==3.0.2
jaraco.text==3.5.0
jedi==0.18.0
Jinja2==3.0.2
joblib==1.1.0
jsonrpclib-pelix==0.4.3.1
kiwisolver==1.3.2
lazy-object-proxy==1.6.0
lensfun==0.3.95
libfdt==1.6.1
libvirt-python==7.8.0
lit==12.0.1.dev0
logutils==0.3.5
louis==3.19.0
lxml==4.6.3
Mako==1.1.5
mallard-ducktype==1.0.2
Markdown==3.3.4
MarkupSafe==2.0.1
marshmallow==3.14.0
matplotlib==3.4.3
matplotlib-inline==0.1.2
mccabe==0.6.1
meld==3.20.4
meson==0.60.0
mock==3.0.5
monotonic==1.6
more-itertools==8.10.0
msgpack==1.0.2
multidict==5.1.0
mutagen==1.45.1
natsort==7.1.1
netifaces==0.11.0
netsnmp-python==1.0a1
node-semver==0.8.0
nose==1.3.7
numpy==1.21.2
ordered-set==4.0.2
packaging==20.9
paramiko==2.7.2
parso==0.8.2
patch-ng==1.17.4
patiencediff==0.2.2
pbkdf2==1.3
pbr==5.6.0
pecan==1.4.0
pep517==0.11.0
pexpect==4.8.0
pickleshare==0.7.5
Pillow==8.4.0
pipenv==2021.5.29
pkgconfig==1.5.5
platformdirs==2.0.2
platformio==5.1.1
pluggy==0.13.1
pluginbase==1.0.0
ply==3.11
portend==3.0.0
prettytable==0.7.2
progress==1.6
prometheus-client==0.11.0
prompt-toolkit==3.0.20
protobuf==3.17.3
psutil==5.8.0
ptyprocess==0.7.0
pwquality==1.4.4
py==1.10.0
pyaes==1.6.1
pyasn1==0.4.8
pyasn1-modules==0.2.8
pybind11==2.8.1
pycairo==1.20.1
pycodestyle==2.7.0
pycparser==2.20
pycryptodomex==3.10.1
pyelftools==0.27
Pygments==2.10.0
PyGObject==3.42.0
pyinotify==0.9.6
PyJWT==2.2.0
pyliblo==0.10.0
pylint==2.11.1
PyNaCl==1.4.0
pyOpenSSL==20.0.1
pyparsing==2.3.1
PyQt5==5.15.6
PyQt5-sip==12.9.0
Pyro4==4.80
pyserial==3.5
PySocks==1.7.1
pystache==0.5.4
pyte==0.8.0
pytest==6.2.5
pytest-cov==2.12.1
python-dateutil==2.8.2
python-miio==0.5.6
python-pam==1.8.4
python-slugify==5.0.2
python-xapp==2.2.1
python-xlib==0.31
pytz==2021.3
pyxdg==0.27
PyYAML==5.4.1
qrcode==7.3
rados==2.0.0
rbd==2.0.0
rdflib==5.0.0
Reflector==2021.7.8.19.52.52
repoze.lru==0.7
requests==2.26.0
resolvelib==0.5.5
retrying==1.3.3
rfc3986==1.5.0
rgw==2.0.0
Routes==2.5.1
rpm==4.17.0
rsa==4.7.2
ruamel.yaml==0.17.16
ruamel.yaml.clib==0.2.6
scikit-learn==1.0
scipy==1.7.1
SCons==4.2.0
selenium==3.141.0
semantic-version==2.8.5
serpent==1.40
setproctitle==1.2.2
six==1.16.0
sniffio==1.2.0
snowballstemmer==2.1.0
soupsieve==2.2.1
speedtest-cli==2.1.3
Sphinx==4.2.0
sphinx-rtd-theme==1.0.0
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
SQLAlchemy==1.4.25
starlette==0.14.2
tabulate==0.8.9
team==1.0
tempora==4.1.2
text-unidecode==1.3
thefuck==3.31
threadpoolctl==2.2.0
tinycss2==1.1.0
toml==0.10.2
tomli==1.2.1
tqdm==4.62.3
traitlets==5.1.0
typing-extensions==3.10.0.2
uritemplate==3.0.1
urllib3==1.26.7
uvicorn==0.15.0
uvloop==0.16.0
validate==5.1.0.dev0
virtualenv==20.2.2
virtualenv-clone==0.5.7
voluptuous==0.12.2
voluptuous-serialize==2.4.0
waitress==2.0.0
wcwidth==0.2.5
webencodings==0.5.1
WebOb==1.8.7
websocket-client==0.59.0
websockets==9.1
WebTest==3.0.0
Werkzeug==2.0.2
wrapt==1.12.1
wsproto==1.0.0
wxPython==4.0.7.post2
xmlsec==1.3.12
yarl==1.6.3
youtube-dl==2021.6.6
zc.lockfile==2.0
zeroconf==0.36.11
zipp==3.6.0
```

0xFEEDC0DE64, 2021-11-01:
I was able to fix it by running:
 ```
 rm ~/.local/lib/python* ~/.espressif -Rfv
 ```

closed 2021-11-01 by 0xFEEDC0DE64

dobairoland, 2021-11-02:
Great! 

For anyone reading this later: virtualenv can misbehave if it is installed both by a package management system and pip as well. The solution is to uninstall one of them.