espressif/esp-idf #7810: Ethernet bridge working in station mode with TCP/IP

maurotaroni, 2021-11-01:
Module or chip used: ESP32-WROVER-E
IDF version: v4.3
Operating System: Windows
Hello,
I'd like to implement an ethernet bridge for a generic network device (videocamera, printer, etc.) working as station and access point simultaneously with TPC/IP stack to implement custom application services.
I have been testing ethernet bridging using the eth2ap code, and also tryed to adapt source code from https://github.com/espressif/esp-idf/issues/7206 for working initially in station mode.
I've used the eth_fork_netif_glue solution without achieving my goal.
I can see that  pkt_eth2wifi is been called succesfully, whereas the callback registered for esp_wifi_internal_reg_rxcb is never been called.
Someone thinks is possible to achieve my goal, at least working in station mode?

david-cermak, 2021-11-02:
Hi @maurotaroni 

Thank you for raising this feature request. Could you please describe in more details of what you're trying to achieve? (as you've probably seen in the referenced issue, it had taken some time and posts to exchange for us to understand the desired use-case).
Let's start with the very simple Wifi station -- Ethernet scenario: Where's the DHCP server (or servers?), are both networks on the same subnet? any requirements on static IP's or running DHCP server on any interface?

My naive understanding of a simple STA-Ethernet bridge is that you'd just inter-connect the interfaces on L2 layer, the same way as it's done in this [eth2ap](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap) example, so the nodes connected to Ethernet would get IP's form the wireless network. Is that correct? Have you tried to modify the `eth2ap` example to `sta2eth` ?
After getting this to work, you can use the `eth_fork_netif_glue`-like solution (it would be `fork-wifi-gule`) for getting IP connectivity to ESP32.

maurotaroni, 2021-11-02:
Hello David,

thanks for your attention. What I'd like to achieve is to implement a solution to transform a wired network commercial device (e.g. ip camera, printer, etc) into a wireless device. This could be achieved with one of the examples found.
But additionally I'd like to implement some custom logic on Espressif board, e.g. some rest webservice to configure the whole solution.

Going deeply, I'd like that Espressif could have an ip address different from the ip address of the attached etherned devices. 

Using the Espressif ip address, I'd like to invoke some proprietary webservices. Using the ip address of the attached ethernet device (wich could be configured both as static or dynamic) I'd like to invoke the native services exposed by the commercial network device.

I've already tryed to adapt the source code eth2ap into a sta2eth using eth_fork_netif_glue.

I've seen that Espressif board could obtain an ip address from the access point (that hosts a DHCP server) and with that ip I can ping it succesfully.

The same is not true for the commercial ethernet device (configured with static ip in non conflicting range). I wasn't able to ping it.

What I found strange is that it seems that the callback registered with esp_wifi_internal_reg_rxcb is never been called.

Do you think that goals could be achievable?

Maybe, could be of some help if I post my modified source code? If so, I need to give it a clear of various debug print statements that I've inserted for debugging purpose.

Thank you

david-cermak, 2021-11-02:
> transform a wired network commercial device (e.g. ip camera, printer, etc) into a wireless device. This could be achieved with one of the examples found.

Could you please link this example? I guess it's just a L2 bridge, so the Espressif board has no IP, but transparently bridges the two networks.

> Going deeply, I'd like that Espressif could have an ip address different from the ip address of the attached etherned devices.

Two options here, we could either create separate subnets or use the above L2 bridge and *"fork"* the station netif to L3 stack on Espressif board.

> Using the Espressif ip address, I'd like to invoke some proprietary webservices. 

If these services are meant to configure IP's (static/dynamic) of the ethernet devices, you'd probably have to go with L3 solution, i.e. create a separate L3 network for ethernet and start DHCP server there. Plus have to enable and setup NAT for that interface.

> Do you think that goals could be achievable?

If I understand correctly, you're trying to implement something like a configurable router, rather than the simple ethernet bridge. Yes, it is possible, but please do not expect the solution would be competitive or even close to off-the-shelf routers, especially for the L3 approach, where each and every packet needs to go through lwIP (seeing the IP camera as an example of your ethernet devices, just understand that it would work, but will be slow)

maurotaroni, 2021-11-03:
Hello David,
thank you again for you support! You're great.

> Could you please link this example? I guess it's just a L2 bridge, so the Espressif board has no IP, but transparently bridges the two networks.

You can find the example here: https://github.com/espressif/esp-iot-solution/tree/release/v1.0/examples/eth2wifi.
If I've understand correctly it work at L2.

> If I understand correctly, you're trying to implement something like a configurable router, rather than the simple ethernet bridge. Yes, it is possible, but please do not expect the solution would [...]

Yes, you're right. I'd like to implement a configurable router with the support for only an ethernet device. 
My first goal is to enable the ethernet bridge and working in station mode. I'm trying to use the L2 approach with eth_fork_netif_glue and implement my custom routing policies from ethernet to wifi and viceversa.
But the problem is that whereas I can see packet going from ethernet to wifi, the opposite never occurs.
At least I can't seen any hit of the callback registered with esp_wifi_internal_reg_rxcb.

Maybe I've missed something while adapting eth2ap to eth2sta. I will try to review my code and if I don't achieve my goal I will share here. The only problem is that I'm working on it only in spare time.

david-cermak, 2021-11-10:
Hello @maurotaroni 

You can refer to this demo L3 implementation here: https://github.com/david-cermak/eth-ap-nat/tree/main/eth-sta-nat

There are several issues with L2 implementation, the main trouble is that we're trying to emulate multiple 802.1X nodes on one WiFi interface (no problem for ethernet, we just set promiscuous flag). The example you referred to simply *pretended* that the wifi station *is* the device behind it:

https://github.com/espressif/esp-iot-solution/blob/f94fde73bc222863743d221e0588ad5c88a36d53/examples/eth2wifi/main/ethernet2wifi_demo_main.c#L143-L146

This is possible, but works only for 1:1 relations, so no way for adding another ethernet device, or *forking* the netif (as we did for the eth-ap example)
Other issues are just minor, as you required to have a control over the IP's assigned to your ethernet nodes.

There's no easy solution to this; perhaps if you're okay to keep the WiFi open, you can try to setup promiscuous mode and capture data packets:

https://github.com/david-cermak/eth-ap-nat/blob/f4a893ef7b35966671d476bef200506327e15651/eth-sta-L2/main/ethernet_example_main.c#L295-L296

maurotaroni, 2021-11-11:
Hello @david-cermak,
thank you for your patience and great support. I really appreciate all your efforts to find a solution that will fit my needs.
I hope to find the time to see and try the demo projects you've linked during this weekend.
Thank you again.

david-cermak, 2021-11-19:
@maurotaroni Any update? Could you please close the issue? Please note that this is not a problem in IDF and we won't add any example implementing such a bridge to IDF, since it is really a *very very* specific use case.

maurotaroni, 2021-11-19:
@david-cermak Thank you for the interest. I've tryed the examples you've provided but it seems they don't work as my expectations. I need another time to study and investigate.
But if you need to close the issue feel free to do so.
Thank you
Best regards
Taroni Mauro

david-cermak, 2021-11-19:
> But if you need to close the issue feel free to do so.

an open GitHub issue indicates there's something wrong with the project, a problem, deficiency, a bug or a grave suspicion for one, or some functionality is not supported. Also I usually prefer the reporter(s) to close their issue as an indication that you're okay with the resolution, support or assessment.

> I've tryed the examples you've provided but it seems they don't work as my expectations. 

Well, the L2 approach would never work as you've expected (as mentioned above), but the NAT example should at least work, although slowly and with potential memory issues and missing packets. What was wrong?
No problem to keep this issue open if something not working as expected.

maurotaroni, 2021-11-21:
Hello @david-cermak. I've tryed both examples because I havent't understant that L2 approach will not work as my expectations.
But also I have problem with NAT example. I can ping only the EspressIf address, while I receive no response from the attached ethernet device.
I can see this in the monitor.
```
I (6625) sta_to_eth_L3: Ethernet Link Up
I (6625) sta_to_eth_L3: Ethernet HW Addr c4:5b:be:3a:89:73
E (11575) TRANS_TCP: DNS lookup failed err=202 res=0x0
```
Maybe the problem is that the MAC address printed below is not the one of my attached device, but probably the address of the onboard Ethernet?
Or do you think that the problem could be related to the DNS lookup failed message?
I'm sorry for being not an expert in this field and let you spent your precious time.
Thank you
Best regards
Taroni Mauro

david-cermak, 2021-11-22:
Hello Taroni Mauro,

It is not possible to ping the Ethernet device if it's behind a NAT, you'd have to think of some other way to access the device (I think you mentioned a web page running on the Espressif defice, where you can post the list of active devices connected to the DHCP server, or implement some kind of a port-forwarding)

As for the DNS issue, was the device connected to internet? I added a very simple mqtt client example to the application, just to demonstrate that node can access the *outside* network. (in that sense it works the same way as [this example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/tcp)).

Alvin1Zhang, 2021-12-06:
@maurotaroni Thanks for reporting, would you please help share if any updates? Thanks.

maurotaroni, 2021-12-07:
Hello @Alvin1Zhang ,
I'm sorry for the delay but I had to focus my effort on other projects. I've briefly tested the samples but the didn't work as my expectations. As I''ve understood the IDF has been develop for other purpose.
Since I need more time to investigate, and at the moment I cannot grant all my efforts to that, I'll close the issue.

closed 2021-12-07 by maurotaroni