espressif/esp-idf #7812: IIC触发看门狗，导致程序重启

LiuWenPeng666, 2021-11-02:
# 使用官方例程 Blufi

# SDK配置如下：

1、配置工程为esp32c3，
2、分区表采用自定义，配置成如下：

###### #Name,   Type, SubType, Offset,   Size, Flags

nvs,      data, nvs,     0x9000,        0x4000,
otadata,  data, ota,     0xd000,        0x2000,
phy_init, data, phy,     0xf000,        0x1000,
ota_0,    app,  ota_0,   0x10000,        1920k,
3、esp32c3-specific中配置版本支持为 Rev3
4、使用4.3.1版本 commit:：
2e74914051d14ec2290fc751a8486fb51d73a31e

# 测试情况说明：

1、使用三个测试板进行测试，两个测试板分别使用时钟芯片pcf8563、ds1307，均需要使用IIC通讯，另外一个测试板使用esp32c3的RTC，不需要IIC;

2、测试发现，使用时钟芯片pcf8563、ds1307，也就是使用IIC通讯的测试板，均出现看门狗重启的问题，使用RTC的测试板没有出现重启；

3、进过分析判断最有可能的是IIC导致的看门狗触发；

# IIC主要代码如下：

```c
#define I2C_FREQ_HZ 400000

#define I2CDEV_TIMEOUT 1000
```

```c
esp_err_t i2c_master_init(i2c_port_t port, int sda, int scl)
{

    i2c_config_t i2c_config = {

        .mode = I2C_MODE_MASTER,

        .sda_io_num = sda,

        .scl_io_num = scl,

        .sda_pullup_en = GPIO_PULLUP_ENABLE,

        .scl_pullup_en = GPIO_PULLUP_ENABLE,

        .master.clk_speed = 100000

    };

    //i2c_param_config(I2C_NUM_0, &i2c_config);

    //i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);

    i2c_param_config(port, &i2c_config);

    return i2c_driver_install(port, I2C_MODE_MASTER, 0, 0, 0);

}
```

```c
esp_err_t i2c_dev_read(const i2c_dev_t *dev, const void *out_data, size_t out_size, void *in_data, size_t in_size)
{

  if (!dev || !in_data || !in_size) return ESP_ERR_INVALID_ARG;

[... 285 lines cut ...]
I (08:45:48.883) wifi_init: tcp rx win: 5744
I (08:45:48.888) wifi_init: tcp mss: 1440
I (08:45:48.893) wifi_init: WiFi IRAM OP enabled
I (08:45:48.898) wifi_init: WiFi RX IRAM OP enabled
I (08:45:48.904) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
I (830) wifi:set rx active PTI: 0, rx ack PTI: 12, and default PTI: 1
I (830) wifi:mode : sta (7c:df:a1:b9:05:18)
I (830) wifi:enable tsf
I (08:45:49.028) BLUFI_EXAMPLE: wifi event id 2

W (08:45:49.033) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (840) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (850) wifi:state: init -> auth (b0)
I (08:45:49.051) BTDM_INIT: BT controller compile version [9c99115]
I (08:45:49.057) coexist: coexist rom version 9387209
I (08:45:49.064) BTDM_INIT: Bluetooth MAC: 7c:df:a1:b9:05:19

I (880) wifi:state: auth -> assoc (0)
I (08:45:49.085) BLUFI_EXAMPLE: BD ADDR: 7c:df:a1:b9:05:19

I (08:45:49.086) BLUFI_EXAMPLE: BLUFI VERSION 0102

I (08:45:49.088) BLUFI_EXAMPLE: BLUFI init finish

...

miroslavpetrov, 2021-11-05:
I am experiencing exactly the same issue when reading i2c devices with esp32-c3 and idf v.4.3.1. 
After few hours of working fine the tast_wdt gets triggered 

`Voltage: 236.53 V, Current: 0.00 A, Active Power: 0.00 
Temperature: 25.74 C, Humidity: 43.37 
heap:158836, max:110180
Time to ade7953_read_sensor: 46 ms
Time to read i2c: 50 ms
I (32821065) app_driver: No significant difference in Voltage. Not updating.
I (32821075) app_driver: No significant difference in Current. Not updating.
I (32821085) app_driver: No significant difference in Active Power. Not updating.
I (32821095) app_driver: No significant difference in Temperature. Not updating.
Time to report to rmaker: 31 ms
E (32842995) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (32842995) task_wdt:  - IDLE (CPU 0)
E (32842995) task_wdt: Tasks currently running:
E (32842995) task_wdt: CPU 0: esp_timer`

Oreobird, 2021-11-10:
Any progress of this issue ? Similarly issue happened to me after writing i2c device failure with esp32c3 and idf v4.3.1.

LiuWenPeng666, 2021-11-10:
> Any progress of this issue ? Similarly issue happened to me after writing i2c device failure with esp32c3 and idf v4.3.1.

我的做法是把i2c_hw_fsm_reset(i2c_num);函数屏蔽了，禁用掉了IIC的异常情况复位的功能，期待乐鑫官方给出解决方法。

miroslavpetrov, 2021-11-10:
I switched to master branch of esp-idf and the problem hasn't shown for 3 hours so far. 

I also found that if on 4.3.1 I use I2C together with Wifi and MQTT and I start a ping flood against the ESP32-C3 it crashes in absolutely the same way as it would after few hours/days.

I have done some other tests - same fw on 4.3.1 but without communicating with I2C sensors. The task_wdt still triggers and outputs the same core dump, but requires some more ping flooding(from 2PCs).

This means to me that either the WiFi or MQTT components are using esp_timer in an inappropriate way which starves the CPU and combined with CPU intensive I2C communication the problem appears and the task_wdt triggers.

I have made the above tests again with the master branch and the task_wdt doesn't get triggered when ping flooding the esp32-c3.

KaeLL, 2021-11-10:
@miroslavpetrov do a bisect on the master branch.

FanhuaCloud, 2021-12-20:
I have same ploblem: Task watchdog got triggered.
my idf version 4.4-dev

mythbuster5, 2021-12-21:
[i2c_patch_1210 .txt](https://github.com/espressif/esp-idf/files/7749484/i2c_patch_1210.txt)

A patch may help here, change the *.txt to *.patch and git apply i2c_patch_1210.patch. Thanks

AxelLin, 2022-03-31:
> > Any progress of this issue ? Similarly issue happened to me after writing i2c device failure with esp32c3 and idf v4.3.1.
> 
> 我的做法是把i2c_hw_fsm_reset(i2c_num);函数屏蔽了，禁用掉了IIC的异常情况复位的功能，期待乐鑫官方给出解决方法。

Above is the same as my observation:
https://github.com/espressif/esp-idf/issues/8120#issuecomment-1061475029

This fixes the issue (the patch is against master):
https://github.com/espressif/esp-idf/issues/8120#issuecomment-1080328035

LiuWenPeng666, 2022-03-31:
> > > Any progress of this issue ? Similarly issue happened to me after writing i2c device failure with esp32c3 and idf v4.3.1.
> > 
> > 
> > 我的做法是把i2c_hw_fsm_reset(i2c_num);函数屏蔽了，禁用掉了IIC的异常情况复位的功能，期待乐鑫官方给出解决方法。
> 
> Above is the same as my observation: [#8120 (comment)](https://github.com/espressif/esp-idf/issues/8120#issuecomment-1061475029)
> 
> This fixes the issue (the patch is against master): [#8120 (comment)](https://github.com/espressif/esp-idf/issues/8120#issuecomment-1080328035)

> > > Any progress of this issue ? Similarly issue happened to me after writing i2c device failure with esp32c3 and idf v4.3.1.
> > 
> > 
> > 我的做法是把i2c_hw_fsm_reset(i2c_num);函数屏蔽了，禁用掉了IIC的异常情况复位的功能，期待乐鑫官方给出解决方法。
> 
> Above is the same as my observation: [#8120 (comment)](https://github.com/espressif/esp-idf/issues/8120#issuecomment-1061475029)
> 
> This fixes the issue (the patch is against master): [#8120 (comment)](https://github.com/espressif/esp-idf/issues/8120#issuecomment-1080328035)

兄弟，使用IIC，祝你好运！！！

ginkgm, 2022-10-11:
Issue fixed on:

<html>
<body>

Target Branch | Merged | Supported | Last N.A.
-- | -- | -- | --
master | 0b00831 | v5.0 | null
release/v4.4 | f86be61 | v4.4.2 | v4.4.1
release/v4.3 | 1b40e61 | v4.3.4 | v4.3.3
release/v4.2 | 2900d0d | null | v4.2.4

</body>
</html>
Target Branch	Merged	Supported	Last N.A.
master	[0b00831](https://github.com/espressif/esp-idf/commit/0b00831703ec351c4c3b1040b004b358e4396067)	v5.0	null
release/v4.4	[f86be61](https://github.com/espressif/esp-idf/commit/f86be6141b3b92f84d0645cba94636ba8944c9f3)	v4.4.2	v4.4.1
release/v4.3	[1b40e61](https://github.com/espressif/esp-idf/commit/1b40e6173e4b58920160e200cb9e735d3da33e25)	v4.3.4	v4.3.3
release/v4.2	[2900d0d](https://github.com/espressif/esp-idf/commit/2900d0da2e7a31175677e62a1a5162ccb55b912b)	null	v4.2.4

Close this issue. If you find it still existing, please fell free to report again.