espressif/esp-idf #7814: Multi connection issue with esp32s3

twomas, 2021-11-02:
## Environment

- Development Kit:      ESP32-S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-S3-WROOM
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3401-gb86fe0c66c
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

This issue is only seen for esp2s3. Same application for esp32 seems to be ok.
See attached picture
[image]

The sequence working is when central  connection is made before the peripheral connection.

The sequence that is not working is when the peripheral connection already exist when trying to establish the central connection, marked red in picture.

After esp_ble_gattc_open this is seen "BT_L2CAP: L2CA_CancelBleConnectReq - no connection pending"

xiewenxiang, 2021-11-26:
Hi @twomas 

I want to reproduce this issues. 

The IDF version is release/v4.4  commit:[b86fe0c66ca](https://github.com/espressif/esp-idf/tree/b86fe0c66ca17d5907ae26ecdffc9d4eb79cec07).

I test with [gattc_gatts_coex](https://github.com/espressif/esp-idf/tree/b86fe0c66ca17d5907ae26ecdffc9d4eb79cec07/examples/bluetooth/bluedroid/coex/gattc_gatts_coex) example and [gatt_server_service_table](https://github.com/espressif/esp-idf/tree/b86fe0c66ca17d5907ae26ecdffc9d4eb79cec07/examples/bluetooth/bluedroid/ble/gatt_server_service_table) example.

ESP32-S3-DevKitC-1 download the gattc_gatts_coex.bin, ESP32-DevKit-V4 download the gatt_server_service_table.bin. 

When the chip was power on, I did two tests：

`Test 1:`

ESP32S3(central) connects to the ESP32(peripheral), then, the PC(central) connects to ESP32S3(peripheral).

`Test 2:`

the PC(central) connects to ESP32S3(peripheral), then, ESP32S3(central) connects to the ESP32(peripheral).

Both of these conditions worked in my tests. I am not sure whether my retest method is different from yours. Could you please help confirm it.

twomas, 2021-12-01:
Hi @xiewenxiang 
It seems to be device related.
I just noticed if don't use our Windows BLE device in the picture above it seems to work better.
So if I use our Android app instead of the Windows BLE device I manage to do the central connection after peripheral connection.

zztiger123, 2021-12-09:
Hi @twomas ,

Good day.

May I know if the issue was solved or if you still have any questions about this issue?

twomas, 2021-12-09:
Hi @zztiger123

Seems to work better now. I close this.

twomas, 2021-12-09:
We modified our connection parameter config. Seems to be related to this

closed 2021-12-09 by twomas