espressif/esp-idf #7815: Otatool not working

MaazSk, 2021-11-02:
- Development Kit:     ESP32-DevKit
- Kit version:     v1
- Module or chip used:  ESP32-WROOM-32
- IDF version:   v4.3
- Build System:         [Make|CMake|idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Using an IDE :  VS Code.
- Power Supply:         USB

## Problem Description
Hello everyone,

When I am running `python otatool_example.py` which is available in `examples/systems/ota/otatool` it is giving me the following error.

```
Traceback (most recent call last):
  File "/home/maaz/esp/esp-idf/examples/system/ota/otatool/otatool_example.py", line 118, in <module>
    main()
  File "/home/maaz/esp/esp-idf/examples/system/ota/otatool/otatool_example.py", line 52, in main
    from otatool import OtatoolTarget
ModuleNotFoundError: No module named 'otatool'
```
And when I am running `otatool_example.sh` I am getting the following error

```
Writing factory firmware to ota_0
python: can't open file '/components/app_update/otatool.py': [Errno 2] No such file or directory
Writing factory firmware to ota_1
python: can't open file '/components/app_update/otatool.py': [Errno 2] No such file or directory
python: can't open file '/components/app_update/otatool.py': [Errno 2] No such file or directory
python: can't open file '/components/app_update/otatool.py': [Errno 2] No such file or directory
stat: cannot stat 'build/otatool.bin': No such file or directory
stat: cannot stat 'app0.bin': No such file or directory
!!!!!!!!!!!!!!!!!!!
FAILURE: Slot 0 app does not match factory app
!!!!!!!!!!!!!!!!!!!
stat: cannot stat 'build/otatool.bin': No such file or directory
stat: cannot stat 'app1.bin': No such file or directory
!!!!!!!!!!!!!!!!!!!
FAILURE: Slot 1 app does not match factory app
!!!!!!!!!!!!!!!!!!!
Switching to factory app
python: can't open file '/components/app_update/otatool.py': [Errno 2] No such file or directory
!!!!!!!!!!!!!!!!!!!
FAILURE: Running partition 'No module named esptool' does not match expected 'factory'
!!!!!!!!!!!!!!!!!!!
```
How can I use this example and is there any other way to test OTA?

KonstantinKondrashov, 2021-11-03:
Hi @MaazSk !
Probably IDF env is not installed properly. Please follow steps 3 and 4 in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-3-set-up-the-tools.

I think these steps were missed by you.
cd ~/esp/esp-idf
./install.sh
. ./export.sh

To run this example test:
cd ~/esp/esp-idf/examples/system/ota/otatool
idf.py build
idf.py flash
python otatool_example.py

MaazSk, 2021-11-03:
Yes, the problem was with the IDF I resolved it last night only. 

I will try these following steps and will let you know

> cd ~/esp/esp-idf/examples/system/ota/otatool
> idf.py build
> idf.py flash
> python otatool_example.py

Also while running one of the files I am getting the following error
`import ttfw_idf ModuleNotFoundError: No module named 'ttfw_idf'`

KonstantinKondrashov, 2021-11-03:
@MaazSk 
export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages"
sudo python -m pip install -r $IDF_PATH/tools/ci/python_packages/ttfw_idf/requirements.txt

MaazSk, 2021-11-03:
@KonstantinKondrashov 
I tried the above commands and they were installed successfully

`DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Requirement already satisfied: pyserial in /usr/local/lib/python2.7/dist-packages (from -r /home/maaz/esp/esp-idf/tools/ci/python_packages/ttfw_idf/../tiny_test_fw/requirements.txt (line 1)) (3.5)
Collecting pyyaml
  Downloading PyYAML-5.4.1-cp27-cp27mu-manylinux1_x86_64.whl (574 kB)
     |████████████████████████████████| 574 kB 4.7 MB/s 
Collecting junit_xml
  Downloading junit_xml-1.9-py2.py3-none-any.whl (7.1 kB)
Collecting netifaces
  Downloading netifaces-0.11.0-cp27-cp27mu-manylinux_2_5_x86_64.manylinux1_x86_64.whl (31 kB)
Collecting pexpect
  Downloading pexpect-4.8.0-py2.py3-none-any.whl (59 kB)
     |████████████████████████████████| 59 kB 4.1 MB/s 
Collecting python-gitlab
  Downloading python-gitlab-1.15.0.tar.gz (124 kB)
     |████████████████████████████████| 124 kB 9.6 MB/s 
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting ptyprocess>=0.5
  Downloading ptyprocess-0.7.0-py2.py3-none-any.whl (13 kB)
Collecting requests>=2.4.2
  Downloading requests-2.26.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 991 kB/s 
Collecting idna<3,>=2.5; python_version < "3"
  Downloading idna-2.10-py2.py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 5.5 MB/s 
Collecting certifi>=2017.4.17
  Downloading certifi-2021.10.8-py2.py3-none-any.whl (149 kB)
     |████████████████████████████████| 149 kB 9.8 MB/s 
Collecting chardet<5,>=3.0.2; python_version < "3"
  Downloading chardet-4.0.0-py2.py3-none-any.whl (178 kB)
     |████████████████████████████████| 178 kB 9.3 MB/s 
Collecting urllib3<1.27,>=1.21.1
  Downloading urllib3-1.26.7-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 11.0 MB/s 
Building wheels for collected packages: python-gitlab
  Building wheel for python-gitlab (setup.py) ... done
  Created wheel for python-gitlab: filename=python_gitlab-1.15.0-py2-none-any.whl size=70274 sha256=e23e3c16a4f72e5d760ce377d31d8d4e801fc585dfd00719fcb9f11507c17a68
  Stored in directory: /root/.cache/pip/wheels/d4/fb/67/b3ede2677af845f38c25507c993bf06bada4ec1d0dc03782cf
Successfully built python-gitlab
Installing collected packages: pyyaml, six, junit-xml, netifaces, ptyprocess, pexpect, idna, certifi, chardet, urllib3, requests, python-gitlab
Successfully installed certifi-2021.10.8 chardet-4.0.0 idna-2.10 junit-xml-1.9 netifaces-0.11.0 pexpect-4.8.0 ptyprocess-0.7.0 python-gitlab-1.15.0 pyyaml-5.4.1 requests-2.26.0 six-1.16.0 urllib3-1.26.7
`

But still, when I am running the python script I am getting the following error

`Traceback (most recent call last):
  File "/home/maaz/new_ota_update_tool/simple_ota_example/example_test.py", line 8, in <module>
    import ttfw_idf
ModuleNotFoundError: No module named 'ttfw_idf'`

KonstantinKondrashov, 2021-11-03:
@MaazSk 
You can try to diagnostic your case running:
> printenv  PYTHONPATH
make sure that PYTHONPATH is correct.

for 4.3 your python path should be like:
which python
/home/{user}/.espressif/python_env/idf4.3_py3.8_env/bin/python

MaazSk, 2021-11-03:
I think my PYTHONPATH is not set because when I try `printenv PYTHONPATH` I don't get any output I will set the python path and will retry

MaazSk, 2021-11-03:
@KonstantinKondrashov 
How can I set the environment variables permanently and also when the environment variables are set to "/home/{user}/.espressif/python_env/idf4.3_py3.8_env/bin/python" I get the error 
 

> import ttfw_idf ModuleNotFoundError: No module named 'ttfw_idf'

and when the environment variables are set to 

> export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages"

I don't get any errors.

Thank you!

KonstantinKondrashov, 2021-11-03:
@MaazSk Sorry, we do not have a way to set these environment variables permanently. But we can improve example_scripts  by adding at least some tips on what need to do to fix this issue.

MaazSk, 2021-11-03:
@KonstantinKondrashov  Okay so every time we need to run this example we will need to set environment variables to 

> export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages"

And then execute the python script.

Thank you for your help

MaazSk, 2021-11-04:
Hi @KonstantinKondrashov,
Is there a way in which I can write to different OTA partitions manually and then switch between those OTA partitions 

Thanks in advance

KonstantinKondrashov, 2021-11-04:
Hi @MaazSk !
Please take a look at an example of how it is done here https://github.com/espressif/esp-idf/blob/master/examples/system/ota/otatool/otatool_example.sh#L46-L67.

I think you are asking about how to do it from CLI.

Writing factory firmware to ota_0
 > python $IDF_PATH/components/app_update/otatool.py write_ota_partition --name ota_1 --input $BINARY

Switch to slot 0
 > python $IDF_PATH/components/app_update/otatool.py   switch_ota_partition --slot 0

Additionally, a few useful commands:
idf.py  read-otadata  
idf.py  erase-otadata

MaazSk, 2021-11-05:
Yes, the above-mentioned commands did the job for me.

Thank you
@KonstantinKondrashov

closed 2021-11-05 by MaazSk