espressif/esp-idf #7816: Disabling IPv6

fbucafusco, 2021-11-02:
## Problem Description

I tried to disable IPv6 support in lwip, and many compilations errors arose (most of them related with asio and coap modules, that are not enabled in menuconfig ).

I did it via menu config, so I imagined that it could be done, but it seems it needs some polish. 

[Here ](https://www.esp32.com/viewtopic.php?t=2260) it says that could be disabled 

[ref1](https://www.esp32.com/viewtopic.php?t=17150), [ref2](https://www.esp32.com/viewtopic.php?t=2260)
  
I am using the ESPIDF platformio version that currently is the 4.3.0

Maybe I am doing something wrong.

david-cermak, 2021-11-02:
@fbucafusco Could you tell us more about your project and the build system? You can start with the issue template, this will help us understand how you configure and build the project.
IPv4-only configuration is normally supported, but these two components (`asio`, `coap`) are excluded from the build, see the docs here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#lwip-ipv6

fbucafusco, 2021-11-03:
Hi! My build system es PlatformIO, that eventually uses espidf build system somehow. 
Sorry I didn't use the template cos I thought it would be something easy to see from the source.
Just to be sure I took an example from the repo examples\protocols\https_server\simple y configure it to compile almost with the default options (I removed the certificate validation from mbedtls component). When I was sure that it compiled, I went to the LWIP ip v6 option and I disabled it (CONFIG_LWIP_IPV6 not set). That macro is copied to LWIP_IPV6 definition, that controls the definition of [this structure](https://github.com/espressif/esp-lwip/blob/2749568fe15df2003f6c3f37f0dfd44f8f01fcd6/src/include/lwip/sockets.h#L81). Then, that structure is not declared anywere and is used many times within many components, just to mention [this place](https://github.com/espressif/asio/blob/f31694c9f1746ba189a4bcae2e34db15135ddb22/asio/include/asio/detail/socket_types.hpp#L115), but there are others.  
The result is that the comilations drops many messages similar to "error: 'sockaddr_in6' does not name a type; did you mean ..."
Maybe I am doing something wrong, anyway

david-cermak, 2021-11-03:
Thanks for your reply, I see that these components are not automatically excluded from the build on IDF `v4.3.0`. it seems the we forget to backport https://github.com/espressif/esp-idf/commit/e305f2938278c2a39e75c21a3ed59d8f4d4e62fa to `v4.3`.

As a workaround, you can disable it manually, by adding this line to the project makefile
```cmake
set(EXCLUDE_COMPONENTS "coap" "asio")
```

fbucafusco, 2021-11-03:
Thanks for the help!

david-cermak, 2023-08-28:
This issue has been fixed in 9d72bdd

@fbucafusco Unfortunately the coap library (and our IDF port) doesn't support IPv4 only mode. If you need this, please create a separate issue.

closed 2023-08-28 by david-cermak