espressif/esp-idf #7817: esp_ble_gap_config_adv_data_raw should distinguish memory/size erros.

dachalco, 2021-11-02:
**Is your feature request related to a problem? Please describe.**
When using bluedroid stack, and updating advertisement PDU data, I wish [esp_ble_gap_config_adv_data_raw](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/esp_gap_ble_api.c#L431) would return `ESP_ERR_NO_MEM` to indicate that my attempted data length exceeds `ESP_BLE_ADV_DATA_LEN_MAX`, instead of it being consolidated with other error conditions and simply returning `ESP_ERR_INVALID_ARG`.

**Describe the solution you'd like**
I'd like mentioned functioned to return `ESP_ERR_NO_MEM` when `raw_data_len > ESP_BLE_ADV_DATA_LEN_MAX`, so I can percolate the exact error condition up my superposed stack, and add indicative logging for users.

**Describe alternatives you've considered**
One alternative is to check whether `raw_data_len > ESP_BLE_ADV_DATA_LEN_MAX` before calling `esp_ble_gap_config_adv_data_raw`, however I don't prefer this solution as this is work seems in function's purview.

**Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.**
NimBLE stack already suffices, by returning `BLE_HS_EMSGSIZE` for same error condition under similar circumstances.

xiewenxiang, 2021-11-05:
Thank you for your suggestion. We will evaluate it and decide whether to change it.

zhp0406, 2024-07-15:
Considering the impact on previous user and the actual usage scenarios, this feature will be not supported now.

closed 2024-07-15 by Indastri