espressif/esp-idf #7820: The `export.ps1` script used undefined variable: `$IsWindows`

cmumford, 2021-11-03:
## Environment

- Development Kit:      N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  N/A
- IDF version (run ``git describe --tags`` to find it): v4.3.1
- Build System:         N/A
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System:     Windows 10
- (Windows only) environment type: PowerShell
- Using an IDE?: no
- Power Supply:         N/A

## Problem Description

`$IsWindows` is used, but never defined.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. Install IDE.
2. Enable strict debugging: `Set-PSDebug -strict`
3. `./export.ps``

dobairoland, 2021-11-03:
Hi @cmumford. Thanks for the report. Do you know the proper solution for this? IsWindows should have been defined on other platform to $False: 

https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_automatic_variables?view=powershell-7.1#iswindows

Shouldn't this be a PowerShell issue? It should not complain about his own internal variable.

dobairoland, 2021-11-08:
@cmumford I think this issue is caused by old version of PowerShell. The link given above is for 7.1. The variable is missing for version 5.1. 

I can reproduce this in PowerShell 5.1. Just execute `$IsWindows`. It won't print anything. If you repeat this after setting the debug flag then it will produce the same error as `export.ps1`.

cmumford, 2021-11-09:
Yes, here's my version:

```
Major  Minor  Build  Revision
-----  -----  -----  --------
5      1      19041  1320
```

I think I'll submit a PR to set `$IsWindows` to true if not defined. I think that PowerShell 6.x shipped on non-windows platforms so this won't be technically correct.

cmumford, 2021-11-09:
Actually it is defined in "PowerShell Core 6 and PowerShell 7", so maybe were OK here.

closed 2021-11-13 by espressif-bot