espressif/esp-idf #7821: ESP32S3 Cannot start after downloading the program

Kevincoooool, 2021-11-03:
After I downloaded the program as usual, the restart prompt failed to start. It has happened twice, and it can only be solved by replacing the module. Is the module damaged?

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x178c
load:0x3b600000,len:0x8c8ff
Invalid image block, can't boot.
ets_main.c 329
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x178c
load:0x3b600000,len:0x8c8ff
Invalid image block, can't boot.
ets_main.c 329
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x178c`

0xjakob, 2021-11-04:
@Kevincoooool Do I understand this correctly that this problem only persist on one particular module and if you flash the same project to other modules these other modules don't have problems running the application?

Alvin1Zhang, 2021-11-22:
@Kevincoooool Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Kevincoooool, 2021-12-04:
> @Kevincoooool Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Hello, when I use the D2 version of ESP32S3, I downloaded the program normally as usual. After the download, I restarted and the above error occurred. It was accidental and there was no specific condition. It appeared twice. I opened the shield. After replacing the FLASH chip, re-downloading is normal, indicating that the FLASH chip is damaged by ESP32

RobVlaar, 2021-12-07:
@Kevincoooool we had the same problem, but with the ESP32C3, and we put the module in boot mode and then start ESP-IDF cmd tool. After this we go to our project folder and run `idf.py set-target esp32c3` when this is done we run `idf.py flash` after this reset the board without boot mode and we can use our board again. 

We "broke" our flash while we tried to program with the wrong target settings. 

Hope this can help you out

zhangtemplar, 2021-12-27:
Same error when following the tutorials in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-8-build-the-project

# command
```
idf.py set-target esp32s3
idf.py build 
~/.espressif/python_env/idf4.3_py3.9_env/bin/python ../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.usbserial-14410 -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello-world.bin
idf.py -p /dev/cu.usbserial-14410 monitor
```

# error message:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15c8
load:0x65202e6e,len:0x3d207272
Invalid image block, can't boot.
ets_main.c 329
```

zhongwei, 2022-02-05:
I have the same problem. I solved this problem by unplugging and reinserting the USB.

gmarcc, 2022-04-24:
I have the same problem, but unplugging and reinserting the USB didn't work for me:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15c8
load:0x3ff48000,len:0x40009258
Invalid image block, can't boot.

BrandonNVZNLabs, 2022-05-15:
Having the same issue.  Any solution?

tvanfossen, 2022-05-23:
Bump, seeing the same issue - board was working fine, and then started getting this invalid image block issue

Power cycling does not resolve, reverting to an older project doesnt resolve, erasing flash, fresh clone/build doesnt fix.

Any possibility its an issue with secure boot? Damage to hardware?

demitriaswilbert, 2022-08-18:
Same issue with mine. Any solutions? or do i have to solder a new module?

lgh0322, 2022-10-06:
经常出现， 太难受了

shawvyu, 2022-10-31:
> 也有同样的问题。我通过拔下并重新插入 USB 解决了这个问题
This problem occurs when I use the trust module of the esp32c3. It can be solved by unplugging the USB

patrickrb, 2023-10-27:
just encountered this.  ordering a replacement board for now

F1642358, 2023-11-07:
使用PlatformIO+Arduino按网上的方法配置Falsh和PSRAM后，下载程序就不停的发送以下内容：
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377f79
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

在platformio.ini文件中配置内如如下：
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
; 指定为16MB的FLASH分区表
board_build.arduino.partitions = default_16MB.csv
; 指定FLASH和PSRAM的运行模式
board_build.arduino.memory_type = opi_opi
; 预定义宏，启用PSRAM
build_flags = -DBOARD_HAS_PSRAM
; 指定FLASH容量为16MB
board_upload.flash_size = 16MB

如果把上面新增的配置内容删掉就正常了。

dlakatos847, 2024-02-02:
I've had a similar issue with the ESP32-C6 chip. My resolution was fixing the bootloader's flash address. It used to be `0x1000` for previous models but with the C6 it is `0x0`.

I see that the S3 has the same bootloader address so it might be the issue for you too.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/bootloader.htm

> Bootloader is located at the address 0x0 in the flash.

It is always a good idea to erase the flash entirely when flashing a new image, eg.:
`esptool.py --chip esp32s3 --port /dev/ttyACM0 write_flash --flash_mode dio --erase 0x0 bootloader.bin 0x8000 partition-table.bin 0x10000 example-app.bin`

ramalhais, 2024-02-17:
I have the same issue on an ESP32S3 (YD-ESP32-23 2022-V1.3) similar to esp32s3 devkitc:

```
02:23:00.623 -> ESP-ROM:esp32s3-20210327
02:23:00.670 -> Build:Mar 27 2021
02:23:00.670 -> rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
02:23:00.670 -> SPIWP:0xee
02:23:00.670 -> mode:DIO, clock div:1
02:23:00.670 -> load:0x3fce3808,len:0x44c
02:23:00.670 -> load:0x3c970000,len:0xbd8ff
02:23:00.670 -> Invalid image block, can't boot.
02:23:00.670 -> ets_main.c 329 
```

I've tried erasing the flash, flashing manually, espressif tool to flash (flash_download_tool_3.9.5) and nothing works.
What i think happened is that opened an arduino project and it was set to a normal ESP32 and then flashing broke the flash chip or some setting like efuse.

projectatrts, 2024-03-22:
solved by flashing at 120Mhz

projectatrts, 2024-03-22:
Solved by setting flash frequency at 120Mhz

gitcnd, 2024-05-11:
The problem is the bootloader address

    Bootloader is located at the address 0x0 in the flash.

Find wherever you performed the flash, and change your incorrect 0x1000 to 0x0 and re-flash.

diversit, 2024-12-27:
I have been struggling with this same issue for several days now.
I was happily building up an application using ESPHome as Home Assistant Addon when suddenly after a new firmware upload the device did not boot anymore.

When connecting with an ArduinoIDE the output logged things like
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3c0a0020,len:0x255c0
ets_loader.c 78 
```

Reverting to a previous firmware which had worked fine before also gave issues like:
```
[12:01:40]ESP-ROM:esp32s3-20210327
[12:01:40]Build:Mar 27 2021
[12:01:40]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[12:01:40]Saved PC:0x4037aa90
[12:01:40]SPIWP:0xee
[12:01:40]mode:DIO, clock div:1
[12:01:40]load:0x3c0a0020,len:0x25b20
[12:01:40]load:0x3fc97580,len:0x3ffc
[12:01:40]load:0x40374000,len:0x64cc
[12:01:40]load:0x42000020,len:0x922b8
[12:01:40]load:0x4037a4cc,len:0xd0b0
[12:01:40]SHA-256 comparison failed:
[12:01:40]Calculated: dcfd316f719bf1fd2654e62e212315415068e40c1773945961f48e481897a999
[12:01:40]Expected: 384b9ba526d3a589c290e918e018f8f1c2a062bc5fc55fe85d89073adacd11f7
[12:01:40]Attempting to boot anyway...
[12:01:40]entry 0x403763d0
```

Today I found that via ArduinoIDE I was able to flash an example `blink` or `esp32/gpio/blinkrgb` sketch to the device and both worked fine.
Re-uploading an ESPHome firmware again did not work and gave similar output as above.

Erase the flash (`python -m esptool erase_flash`) also did not make any difference.

As a last resort, after Googling for days did not get me the solution and after a short walk outside listening to 'The AI Fix' podcast (which I can recommend), I thought what the hell, I'll ask ChatGPT.
The 6th bullet point suggestion was to fix the bootloader.
After asking how to do that it came with clear instructions.
Via these I found out that ESPHome generates the bootloader.bin, partitions.bin and firmware.bin files.
In the Home Assistant Addon, these can be found in the ESPHome container in the folder `data/build/<devicename>/.pioenvs/<devicename>/` (<devicename> is the name of the device in ESPHome).

In Home Assistant I have the 'Advanced SSH & Web Terminal' to have a normal terminal on the HA host.
Using `docker cp` these `.bin` files can be copied from the ESPHome container to the Home Assistant host.
Lookup the correct container id using `docker ps`.
Since my Home Assistant is a Linux container on Proxmox, I then used the File Editor (another HA add-on) to download the `.bin` files to my laptop.

Then, using `esptool` I was able to flash the device using the command
```
python -m esptool --chip esp32s3 write_flash --flash_mode dio --erase 0x0 bootloader.bin 0x8000 partitions.bin 0x10000 firmware.bin
```

And now the device works again 🎉 .
Hope this helps someone solving the same issue (as well as future-me 😉 )

MaffooClock, 2025-04-16:
I ran into this today, the repeating `Invalid image block, can't boot` in the serial console after uploading firmware in CDC mode (flashing without touching any buttons).

I found that if I flash the device in DFU mode (holding down `IO0` while pressing `RST`), the device boots fine.  I also tried erasing the flash entirely, but that made no difference.  So, flash via CDC: invalid image block; flash after manually entering DFU mode: boots normally.

Now I'm trying to learn the technical differences between flashing via DFU and CDC modes 🤔.  My only guess is that I've got something goofy in my sdkconfig, but I can't figure out what...

closed 2026-02-05 by Kevincoooool