espressif/esp-idf #7822: Memory & Heap issues with 4.3 (FreeRTOS functions, const char * in DRAM .data and TLSF control_t)

philippe44, 2021-11-03:
While investigating and issue with available heap decrease between 4.3 and 4.0 while porting my application, I've opened #7813 and #7818 that I've closed now that I understand the problem and I think it's better to make a summary here. I observed a loss of 10-15kB of the precious internal RAM for any application and I've seen 3 causes:

1- On 4.3, the FREERTOS_PLACE_FUNCTIONS_INTO_FLASH seems to be disabled by default where it is not an option in 4.0. I missed that, my bad, but that was a reduction of ~7kB of available on-chip RAM for heap. That is simple to "solve"

2- The TLSF has a ton of asserts() with a string in heap_tlsf.c. That's nice but unfortunately, as these functions are placed in IRAM for efficiency, it seems that (logically?) the linker also places the strings (which should normaly be in .rodata in Flash) in DRAM .data, so that's another 1.5kB loss. The tlsf_assert can be changed to nothing, but that does not seem wise, maybe there is a way to still move these string to .rodata in Flash.

3- The TLSF itself requires a gigantic header/control block of 1.6kB for each heap zone, and as esp32 seems to have 5, then we end up "wasting" ~9kB. This one seems really awful to me (no offense meant) because it's a lot of scarce memory (for a better algorithm, I understand).

sle118, 2021-11-04:
Looking forward to seeing that commit reviewed and merged!

0xjakob, 2021-11-16:
@philippe44 regarding 2-:

The idea of asserts is to use them only during development. Once the application is stable, you can disable them. Since they only abort the application and reset the device, they're also not very meaningful for a user. That being said, if you put their string into flash and the assert triggers when flash operations are disabled, then the assert will trigger an exception which is also not very pleasant.

For the rest of the points my knowledge is too limited, I can't make any judgement but I'll pass this to colleagues who are more familiar.

philippe44, 2021-11-17:
@0xjakob: absolutely, I forgot about that but I was surprised because it was a change in 4.3 vs 4.0. I can disable them in general.

philippe44, 2021-12-20:
I've moved to make them simply silent. The issue is that our project is becoming quite big and every piece of ram is needed and now every piece of flash is needed as well...

closed 2022-01-01 by philippe44

0xjakob, 2022-01-12:
> > then the assert will trigger an exception which is also not very pleasant.
> 
> But very useful. It'll pinpoint where the assert failed, so even without a debugger, with gdbstub, you'll see where the assert failed. If it doesn't trigger an exception, it's very likely the assert will go unnoticed (what's a line in a hundred of lines of logs?) and your code will crash later on because the allocated memory is missing or corrupted.

I'm afraid not. If the assert *does not* trigger an exception, then yes, the default is to print out the register stack trace which makes it possible to see where the assert fails. That's what the assert is for.

If, however, there's an exception while the assert function is running and printing the stack trace, then you lose all this information. You may see a stack trace but to the assert code itself, not the code the assert is back tracing at that moment. All you'll know is that an assert triggered, but not where.

AxelLin, 2022-01-15:
> @philippe44 regarding 2-:
> 
> The idea of asserts is to use them only during development. Once the application is stable, you can disable them. Since they only abort the application and reset the device, they're also not very meaningful for a user. 

No, this is not ture.
It's common to have assertion enabled for relase code.
It can avoid undefined behavior if it indeed hit assertion in some cases.
e.g. reset device is better than have a dead device runing in a endless loop.

philippe44, 2022-01-15:
Well, what I hope really is that my PR will be accepted...

AxelLin, 2022-01-16:
> 2- The TLSF has a ton of asserts() with a string in heap_tlsf.c. That's nice but unfortunately, as these functions are placed in IRAM for efficiency, it seems that (logically?) the linker also [...]

Does CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT=y help?

AxelLin, 2022-12-23:
v5.0: c173845ff35985317160413c3c9c529f7703f3d0
v4.4: 345a12c1f981bf20f3541b24eddb83b6972c15eb