espressif/esp-idf #7825: ESP32S3 SPI Master Mode 3 not working with any tested device

nx518, 2021-11-03:
## Environment

- Module used: ESP32-S3-WROOM-1 with ROM esp32s3-20210327 chip revision: 0
- IDF version: v5.0-dev-79-gbcbef9a8db, also tried with v4.4
- Build System: idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description

We are replacing ESP32 with ESP32S3 in some designs. The pcba layouts are almost identical. We checked the signals with scope and CLK, MOSI, MISO, CS levels looks fine.
ESP32S3 SPI Master in mode 3 does not read the MISO signals correctly.
No slave devices supporting mode 3 only can be used.

### Expected Behavior
SPI-master of ESP32S3 in mode 3 works like ESP32 with SPI-Slaves (e.g. ESP32, QCA7000).

### Actual Behavior
SPI-master of ESP32S3 SPI-Master in mode 3 reads false bytes on MISO.
| Master Device         | Evaluated MISO Signal hex "00 01 10 41" from mode 3 SPI-Slave  |
|-----------------------|----------------------------------------------------|
| Scope set to mode 3   | 00 01 10 41 ✔                                      |
| Scope set to mode 1   | 00 02 20 00 ✘                                      |
| ESP32S3 set to mode 3 | 00 02 20 00 ✘                                      |
| ESP32S3 set to mode 1 | 00 00 00 00 ✘                                      |
| ESP32 SPI mode 3      | 00 01 10 41 ✔                                      |

It seems like the values of the falling CLK edges are getting triggered by the ESP32S3 in mode 3 and not the rising edges as supposed. 
Changing [input_delay_ns](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#_CPPv4N29spi_device_interface_config_t14input_delay_nsE) does not seem to change anything on ESP32S3. Technical Reference Manual is not available for ESP32S3 SPI yet.

### Steps to reproduce

1. Connect ESP32S3 as master and ESP32 as slave over SPI.
2. Run example below with SPI-mode 3.

### Code to reproduce this issue

[SPI Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave) with ESP32S3 as sender and ESP32 as receiver or any mode 3 SPI slave device.

### Interesting SPI behaviour between ESP32S3-Master and ESP32-Slave

|                          | ESP32-Slave SPI-Mode 0 | ESP32-Slave SPI-Mode 1 | ESP32-Slave SPI-Mode 2 | ESP32-Slave SPI-Mode 3 |
|--------------------------|-------------------|-------------------|-------------------|-------------------|
| ESP32S3-SPI Master-Mode 0 | ✔                 |                   |                   |                   |
| ESP32S3-SPI Master-Mode 1 |                   | ✘                 |                   | ✔                 |
| ESP32S3-SPI Master-Mode 2 |                   |                   | ✔                |                   |
| ESP32S3-SPI Master-Mode 3 |                   | ✔                 |                   | ✘                 |

Setting the mode to 1 on other slave hardware (not ESP32) did not resolve the mode 3 issue. Probably related: #7698

Icarus113, 2021-11-09:
Hi @nx518 , could you increase SPI Master `cs_ena_pretrans`  (FYI, it's in `spi_device_interface_config_t`)? Try to set it to 1 or much higher.

nx518, 2021-11-12:
Hi @Icarus113 , I tried setting `cs_ena_pretrans` to 1, 5 and 12 and tested it with 4 different slave mode 3 configurations (esp32, esp32 with changed spi_ll.h, QCA7000 and another proprietary device). 
On the scope I can see the increased time between CS and CLK (TCSSCK) generated by `cs_ena_pretrans`.

Unfortunately it does not fix the communication with any of the slave devices, still not getting correct frames, I still think ESP32S3 samples the wrong edges in mode 3.

Icarus113, 2021-11-15:
Hi @nx518 , could you share the following scope results?
```
1. Using the SPI Example you mentioned above, and use ESP32S3 as the Master, ESP32 as the Slave
2. Set both of the Master and Slave to mode 3.
3. Set Master `cs_ena_pretrans` to 1 and 0.
```
I want to see if the ESP32 (Slave) 's timing is different.

nx518, 2021-11-16:
> Hi @nx518 , could you share the following scope results?

`cs_ena_pretrans = 0;`
[image]
Here the slave does not behave properly, so a change in the [example code](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_slave/sender/main/app_main.c) should be considered.

`cs_ena_pretrans = 1;`
[image]

Setting `cs_ena_pretrans = 1;` results in successfully reading the frames. However ESP32S3 is still reading the falling edges, which turns out the same result as reading the rising edges from the scope.

Setting `hw->ctrl2.miso_delay_mode = 4;` for the slave in `spi_ll.h` leads to reading the frames correctly with the ESP32S3, whereas the scope does not get the frames when reading the rising edges:
[image]

With this last slave configuration the scope and the ESP32 read the frames correctly, whereas the ESP32S3 fails:
[image]

Icarus113, 2021-11-18:
@nx518 
Hi! Thanks for the detailed graphs.

If you set `cs_ena_pretrans = 0`, the CS changing and CLK first edge is almost the same time. Under this condition, after Slave(ESP32) sees the CS changing, the CLK could either be 0 or 1 (in Slave's view). If Slave misses the first CLK change, it will launch its data at next launching edge. If you set `cs_ena_pretrans = 1`, Slave has enough time to see the frst CLK change. So its behaviour is right.
       
> However ESP32S3 is still reading the falling edges, 

Your conservation is right. 
    
> Here the slave does not behave properly, so a change in the example code should be considered.

Actually different SPI Slave devices have different timing requirements. This example is a general example and can't meet all requirements. You can modify the `cs_setup_time` (namely, cs_ena_pretrans) to meet this requirement.

For `miso_delay_mode`, I still suggest you setting it according to our TRM

nx518, 2021-11-18:
Ok, so the ESP32S3-Master to ESP32-Slave is resolved with setting cs_ena_pretrans and configurable slave. Thank you!

> >However ESP32S3 is still reading the falling edges,

> Your observation is right.

So there is still a problem for non-configurable mode3 SPI-Slave devices, like QCA7000 on MISO-line:
ESP32S3-master with QCA7000-slave ✘
ESP32-master with QCA7000-slave ✔

[image]

How can I set up ESP32S3 to work like ESP32 as SPI-mode3-master?

Icarus113, 2021-11-19:
@nx518  Hi, this timing is weird. It looks Slave violates the SPI protocol.  Is the Slave set to mode 3? Otherwise maybe there is a huge delay. In reality, the MISO data change should happen after A LAUNCH edge. It's true there is setup time limitation, but physically there are some delays in either PCB or digital circuit. Therefore protocol rule is to launch at an edge, and sample and the next edge.

nx518, 2021-11-26:
@Icarus113 I made some analog measurements with ESP32 and ESP32S3 and QCA7000 as slave. The results still suggest to me that ESP32S3 is reading on the wrong edge.
[image]
The setups are identical, on console ESP32 reads 0x55 => 01010101 and the ESP32S3 reads 0x67 => 01100111, which has the same effect setting the scope to read on the other edge.

The only workaround I found is to use ESP32S3 like this:

1. `spi_bus_add_device` in mode 3 half duplex
2. `spi_device_polling_transmit` to send MOSI data.
3. `spi_bus_remove_device`
4. `spi_bus_add_device` in mode 1 half duplex
5. `spi_device_polling_transmit` to receive MISO data.
6. `spi_bus_remove_device`
... and so on

This reads the MISO data correctly but is too slow.

Icarus113, 2021-11-29:
@nx518 
Hi thanks for this detailed graph. From you graph:
The 4th negedge is the launch edge, there is possibly a timing violation. Slave changes its data nearly the same time as the 4th negedge (or even earlier). At the moment, Master can see either 1 or 0. If it's earlier (this will definitely violate the SPI timing requirements), Master will see 1.
    
As I said, Slave's data change should usually happen after the launch edge (see my previous reply), just as the 2nd and 3rd negedge shows. I think your datasheet should also specify a number which stands for the `CLK to output delay`. This usually is 2ns to 10ns. This means Slave's data will change `2~10ns behind the Clock launch edge`.
       
Reason why ESP32 can sample correct data:
If you're using ESP32 GPIO Matrix, it has some delay. Even if you're using IOMUX, due to PCB design and pin loads, the ESP32 Master still has some input delay. So the actual time when Master sees the MISO data change is delayed.

Icarus113, 2022-09-02:
Hi @nx518 , sorry for the late reply.

> Does this apply for ESP32S3?

You can refer to my previous reply, it actually samples after a half cycle delay.

> Is "A LAUNCH edge" the same as a "rising edge"?

When in mode 3, it's negedge.

Current IDF chips don't guarantee this situation, when:
Slave violates SPI protocol under mode 3, Slave changes its data at the same time with / earlier than a negedge
This may be improved, when there's some delay on this signal, but this isn't guaranteed.

Icarus113, 2022-09-19:
As the SPI related timing issues are clarified. I think we can close this issue.

Alvin1Zhang, 2022-09-19:
Thanks for reporting, feel free to reopen.

closed 2022-09-19 by Alvin1Zhang

kiibimees, 2023-01-30:
@nx518 , did you manage to get the SPI working on S2/S3 versions or is there still this _half-duplex bus-remove-add work-around_ needed?

nx518, 2023-01-30:
Well... it is working with another hack... delaying MISO with another MCU by 1 cycle...
My resume of this bug is that the SPI protocoll is not well standardized, everybody does what he wants.
And espressif just changed the behaviour for ESP32 vs ESP32S3 by 180 degree.

stefannagelchargebyte, 2023-11-08:
Dear developers,
I also face a problem with a non-stable MISO reception on ESP32-S3.

Just to verify:
In SPI-mode 3, the ESP samples the MISO data on the rising edge of the clock like it is mentioned in the Reference manual:

• Mode 3: CPOL = 1, CPHA = 1; SCK is 1 when the SPI is in idle state; data is changed on the negative
edge of SCK and sampled on the positive edge.

If I read the above conversation, I understand that it is vice versa. Or am I wrong.

Kind regards