espressif/esp-idf #7827: BLE scan response is empty

twomas, 2021-11-03:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-D0WD-V3 (revision 3)
- IDF version (run ``git describe --tags`` to find it): v4.3.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
In our application we are using scan response and it seems to work only once after a reboot.

See attached file (scan-rsp.txt) for logging code and output.
Compare logs for "First time after esp_ble_gap_start_scanning" and "Second time after esp_ble_gap_start_scanning"
The second log shows no scan reponse data when scanning.

[scan-rsp.txt](https://github.com/espressif/esp-idf/files/7467901/scan-rsp.txt)

zztiger123, 2021-11-04:
Hi @twomas 

The Bluetooth bottom layer of ESP will report the received broadcast packet once in a while, so you need to do additional processing in your application layer.

For example, the broadcast packet + scan reply packet may be received in the first period of time, and maybe only the broadcast packet may be scanned in the second period.

twomas, 2021-11-04:
Hi @zztiger123

I just noticed that if I do a "esp_ble_gap_set_scan_params" just before every "esp_ble_gap_start_scanning" it works

zztiger123, 2021-11-05:
Hi @twomas 

May I know your specific application requirements?

We don’t need to distinguish too much whether the received packet is adv_data or scan rsp data. Both have a maximum of 31 bytes, and the structure inside is exactly the same. When the packet is received, adv data and scan rsp data can be processed in the same way. .

For example, if there is only adv data, only adv data will be processed, and only scan rsp data will only process scan rsp data. If both are available, they will be processed separately and processed twice, first processing adv data, then scan rsp data

xiewenxiang, 2021-11-05:
@twomas 

Adv data and scan data are reported sometimes together and sometimes separately, because active scanning does not guarantee that each packet can be sent a Scan_REQ.

The application layer does not necessarily require simultaneous reporting because the two pieces of data are independent of each other.

twomas, 2021-11-11:
The question I have if it is needed to do esp_ble_gap_set_scan_params every time before esp_ble_gap_start_scanning?
Without this fix scan responses are not seen.

zhp0406, 2023-09-11:
Have you resolved your issue?
esp_ble_gap_set_scan_params only needs to be called once.

closed 2024-03-10 by Alvin1Zhang