espressif/esp-idf #7832: BLE Python test tools for examples don't support ESP32-C3

dazzalicious, 2021-11-04:
I'm trying to run the BLE nimble examples and they flash onto the ESP32-C3 chip no worries and I can use other devices to connect so I believe they are working fine. It is just when I try to use the python scripts for testing such as blecent_test.py in the blecent example I get the error report below which I guess means that the script doesn't support the C3 chip. It would be great if these could be updated.

[image]

krzychb, 2021-11-04:
Hi @dazzalicious, "blecent_test.py" (and other "_test.py" scripts in the [examples](https://github.com/espressif/esp-idf/tree/master/examples) directory) are intended primarily for continuous integration (CI), not for using locally. See some background information in https://esp32.com/viewtopic.php?t=22552.

dobairoland, 2021-11-04:
I'm not entirely familiar with the BLE examples but they don't seem to support any chip beside ESP32. Please notice the first line of the examples' READMEs. They state that the supported target is ESP32 (only).

dazzalicious, 2021-11-07:
> I'm not entirely familiar with the BLE examples but they don't seem to support any chip beside ESP32. Please notice the first line of the examples' READMEs. They state that the supported target is ESP32 (only).

I've been able to get them to flash and function on the ESP32-C3 chip. I'm a noob with this though.

closed 2024-12-17 by Alvin1Zhang