espressif/esp-idf #7833: Trouble with Encrypted NVS When Updating from 4.2 -> 4.3

0xjakob, 2021-11-04:
This is a follow-up ticket for https://github.com/espressif/esp-idf/issues/5747, since the original issue seems different and has been solved.

## Environment
- IDF version v4.3

## Problem Description
See https://github.com/espressif/esp-idf/issues/5747, after the closing activity.

0xjakob, 2021-11-04:
@baamiis @igrr @mahavirj I opened a new ticket to better track this issue, since it seems quite different than the original one. Please continue the discussion here.

mahavirj, 2021-11-23:
FYI: We have added a new config option for backward compatibility behavior for "encrypted" label to "nvs" partition. Please see https://github.com/espressif/esp-idf/commit/e67128dd0af57bade44cc626530ee21e8308dcaf

AxelLin, 2021-11-23:
> FYI: We have added a new config option for backward compatibility behavior for "encrypted" label to "nvs" partition. Please see [e67128d](https://github.com/espressif/esp-idf/commit/e67128dd0af57bade44cc626530ee21e8308dcaf)

That fix is not yet in v4.3.

mahavirj, 2021-11-24:
> That fix is not yet in v4.3.

Fix is merged internally in `release/v4.4` and `release/v4.3`. It should appear on GH with next sync.

mahavirj, 2021-12-09:
Fixes now available at https://github.com/espressif/esp-idf/commit/893342794f62a26218cefbcf10ad892f2cb188d0 and https://github.com/espressif/esp-idf/commit/e2fb582cf85cfb3cde0fb6c038fbf566b64adfcd. Closing this issue.

closed 2021-12-09 by mahavirj

baamiis, 2021-12-09:
Cool Thanks very much