espressif/esp-idf #7834: Ethernet gets old IP after plugging into other router

silardg, 2021-11-04:
IDF version: 4.2, platformio version 3.1.1
Ethernet chip used: LAN8720
Board is designed based on the LAN8720 eval board

I have two routers, one is giving out IP range 192.168.1.x other is 10.1.3.x
I plug my ethernet into the first router, gets 192.168.1.x IP, which is good.
After plugging it into the other router, I still get the 192.168.1.x, but it should give me 10.1.3.x

I tried reseting the board, thus the LAN8720 with hardware reset pin set. 

I managed to get them to switch, but after fully erasing the flash, first it connected as 192.168.1.x and then after reset 10.1.3.x, but at the third reset it kept being 192.168.1.x.

david-cermak, 2021-11-05:
Hi @silardg 

Could you please share the debug log from your board? Are you getting Link-Down and Link-Up events from Ethernet? When getting the wrong address, do you receive an event or it just keeps the address silently? Do you use any special configuration? (like for example this [CONFIG_LWIP_DHCP_RESTORE_LAST_IP](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-dhcp-restore-last-ip) would explain the behavior)

Switching between subnets is normally supported (just quickly tested on `v4.2` and works as expected), as the link-down event stops the dhcp client and link-up restarts it.

Alvin1Zhang, 2021-11-22:
@silardg Thanks for reporting, would you please help share if any updates for the issue? Thanks.

silardg, 2021-11-26:
The problem has nothing to do with ESP, so we can close the issue.

The problem was that I had two routers giving out IP's and they were "fighting" to send an IP to my ESP. Sometimes the first and other times the second succeeded.

closed 2021-11-26 by silardg