espressif/esp-idf #7835: Build with COMPILER_OPTIMIZATION_NONE and call nvs_flash_init() hit panic

AxelLin, 2021-11-04:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3.1-354-g98d34e5f6db8
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Build with COMPILER_OPTIMIZATION_NONE and found if my application calls nvs_flash_init it will hit panic.

### Expected Behavior
Should work when build with COMPILER_OPTIMIZATION_NONE
### Actual Behavior
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
### Steps to reproduce

Use hello_world to reproduce
1. idf.py menuconfig and select COMPILER_OPTIMIZATION_NONE
2. Test build with hello_word example code, with below code modification
```
diff --git a/examples/get-started/hello_world/main/hello_world_main.c b/examples/get-started/hello_world/main/hello_world_main.c
index ebc5eb0b048b..3c529b76207e 100644
--- a/examples/get-started/hello_world/main/hello_world_main.c
+++ b/examples/get-started/hello_world/main/hello_world_main.c
@@ -12,6 +12,8 @@
 #include "freertos/task.h"
 #include "esp_system.h"
 #include "esp_spi_flash.h"
+#include "nvs.h"
+#include "nvs_flash.h"
 
 void app_main(void)
 {
@@ -32,6 +34,7 @@ void app_main(void)
             (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");
 
     printf("Minimum free heap size: %d bytes\n", esp_get_minimum_free_heap_size());
+nvs_flash_init();
 
     for (int i = 10; i >= 0; i--) {
         printf("Restarting in %d seconds...\n", i);
```
3. My testing device has flash encryption enabled, so I use encrypted-flash below.
(I'm not sure if disable flash encryption has any impact or not.)

Note, If i compile with CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y, it works.

$ idf.py encrypted-flash monitor

ELF file SHA256: 34402adaeefea299

Rebooting...
I (117) cpu_start: Pro cpu up.
I (117) cpu_start: Starting app cpu, entry point is 0x40081114
0x40081114: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (131) cpu_start: Pro cpu start user code
I (131) cpu_start: cpu freq: 160000000
I (131) cpu_start: Application information:
I (136) cpu_start: Project name:     hello-world
I (141) cpu_start: App version:      v4.3.1-354-g98d34e5f6db8-dirty
I (148) cpu_start: Compile time:     Nov  4 2021 17:22:31
I (154) cpu_start: ELF file SHA256:  34402adaeefea299...
I (160) cpu_start: ESP-IDF:          v4.3.1-354-g98d34e5f6db8-dirty
I (167) heap_init: Initializing. RAM available for dynamic allocation:
I (174) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (180) heap_init: At 3FFB3AE8 len 0002C518 (177 KiB): DRAM
I (187) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (193) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (199) heap_init: At 40092A2C len 0000D5D4 (53 KiB): IRAM
I (207) spi_flash: detected chip: generic
I (210) spi_flash: flash io: dio
W (214) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (223) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 1, 4MB external flash
Minimum free heap size: 289932 bytes
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400ea77c  PS      : 0x00060b34  A0      : 0x8008411c  A1      : 0x3ffb64e0  
0x400ea77c: portENTER_CRITICAL_NESTED at /home/axel/esp/esp-idf/examples/get-started/hello_world/build/../../../../components/freertos/port/xtensa/include/freertos/portmacro.h:108

A2      : 0x00000001  A3      : 0x15122500  A4      : 0x00060b20  A5      : 0x00000000  
A6      : 0x00ff0000  A7      : 0x3ffb64e0  A8      : 0x80082010  A9      : 0x3ffb6490  
A10     : 0x00000000  A11     : 0x15122500  A12     : 0x00060d20  A13     : 0x00000000  
A14     : 0x3ffb7a70  A15     : 0x3ffb6490  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace:0x400ea779:0x3ffb64e0 0x40084119:0x3ffb6510 0x40084660:0x3ffb6540 0x400842d4:0x3ffb65e0 0x400d76ea:0x3ffb6640 0x400d74f4:0x3ffb6720 0x400d753e:0x3ffb6750 0x400d7659:0x3ffb6790 0x400db4a2:0x3ffb67d0 0x400d8ba0:0x3ffb6810 0x400d7de8:0x3ffb6850 0x400d7e04:0x3ffb6890 0x400d6700:0x3ffb68b0 0x400ebb6e:0x3ffb68e0
0x400ea779: s_compare_reserved_regions at /home/axel/esp/esp-idf/components/soc/memory_layout_utils.c:60

0x40084119: Cache_Flush at /home/axel/esp/esp-idf/examples/get-started/hello_world/build/../../../../components/esp_rom/include/esp32/rom/cache.h:141

0x40084660: spi_flash_mmap_pages at /home/axel/esp/esp-idf/components/spi_flash/flash_mmap.c:274

0x400842d4: spi_flash_mmap at /home/axel/esp/esp-idf/components/spi_flash/flash_mmap.c:151

0x400d76ea: load_partitions at /home/axel/esp/esp-idf/components/spi_flash/partition.c:179

0x400d74f4: ensure_partitions_loaded at /home/axel/esp/esp-idf/components/spi_flash/partition.c:74

0x400d753e: esp_partition_find at /home/axel/esp/esp-idf/components/spi_flash/partition.c:87

0x400d7659: esp_partition_find_first at /home/axel/esp/esp-idf/components/spi_flash/partition.c:135

0x400db4a2: nvs::partition_lookup::lookup_nvs_partition(char const*, nvs::NVSPartition**) at /home/axel/esp/esp-idf/components/nvs_flash/src/nvs_partition_lookup.cpp:14

0x400d8ba0: nvs::NVSPartitionManager::init_partition(char const*) at /home/axel/esp/esp-idf/components/nvs_flash/src/nvs_partition_manager.cpp:53

0x400d7de8: nvs_flash_init_partition at /home/axel/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:135

0x400d7e04: nvs_flash_init at /home/axel/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:171

0x400d6700: app_main at /home/axel/esp/esp-idf/examples/get-started/hello_world/build/../main/hello_world_main.c:38 (discriminator 4)

0x400ebb6e: main_task at /home/axel/esp/esp-idf/components/freertos/port/port_common.c:133

0xjakob, 2021-11-09:
@AxelLin Sorry for the long follow-up time. But we're discussing this internally already. It looks like problem with the underlying SPI flash API.

AxelLin, 2022-02-23:
> @AxelLin Sorry for the long follow-up time. But we're discussing this internally already. It looks like problem with the underlying SPI flash API.

Just wondering if this is related to https://github.com/espressif/esp-idf/issues/8301#issuecomment-1023072309

AxelLin, 2022-02-24:
Fixed in master: ca6f76d831b4fe6761bf58b964920104743434eb

closed 2022-03-22 by AxelLin