espressif/esp-idf #7837: Corrupted blob causes crash that can't be resolved by system

DCSBL, 2021-11-04:
## Environment

- Development Kit: None
- Kit version: -
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: External 5V

## Problem Description
TL;DR: At some point I think the NVS got corrupted. When the ESP boots it crashes while trying to read a blob from NVS. This crash as caused by an assertion that can't be handled (eg. clear NVS and try again)

Context:
A project uses a large data blob with about 8KB of JSON data. This blob us updated when the user needs to change some functionality. At some point something went wrong (don't know what exactly) and now the ESP refuses to boot.

### Expected Behavior
I expect that the `nvs_get_blob` method returns a non `ESP_OK` status so I can take action like clear the NVS.

### Actual Behavior
The ESP crashes due to a assert:
https://github.com/espressif/esp-idf/blob/417ef24b0641856ea71a7aeb44ccee62b11b5cc8/components/nvs_flash/src/nvs_storage.cpp#L479

### Code to reproduce this issue

```cpp
esp_err_t flash_read_blob_preallocated(const char *section_name, const char *key, void **data, size_t length)
{
    esp_err_t err;
    nvs_handle_t handle;

    // Validate argument
    if (section_name == NULL || strlen(section_name) > 15 || key == NULL || data == NULL || strlen(key) > 15 || length == 0)
    {
        ESP_LOGE(TAG, "Invalid argument");
        return ESP_ERR_INVALID_ARG;
    }

    // Open NVS
    err = nvs_open(section_name, NVS_READONLY, &handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Error opening NVS handle: (%s)", esp_err_to_name(err));
        return err;
    }

    // Read data
    {   
        err = nvs_get_blob(handle, key, data, &length); // <-- CRASH HAPPENS WHEN THIS METHOD IS CALLED

        if (err != ESP_OK)
        {
            ESP_LOGW(TAG, "Error getting data: (%s)", esp_err_to_name(err));
            goto close;
        }
    }

close:
    // Close handle
    nvs_close(handle);

    return err;
}

char value_out[8000];
flash_read_blob_preallocated(NVS_NAMESPACE, "flash_test", (void**) &value_out, 8000);
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
assertion "offset == dataSize" failed: file "/<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_storage.cpp", line 480, function: esp_err_t nvs::Storage::readMultiPageBlob(uint8_t, const char*, void*, size_t)

abort() was called at PC 0x401663c7 on core 0
0x401663c7: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x4008a7fb:0x3ffbda80 0x4008ae55:0x3ffbdaa0 0x400927f6:0x3ffbdac0 0x401663c7:0x3ffbdb30 0x400ecaf9:0x3ffbdb60 0x400ecb3b:0x3ffbdbe0 0x400ed17d:0x3ffbdc40 0x400eb9c5:0x3ffbdc60 0x400ebe79:0x3ffbdca0 0x400db04e:0x3ffbdcc0 0x400e0bc0:0x3ffbdd00 0x400db9cb:0x3ffbdd20 0x400da8bc:0x3ffbdd60 0x4017ef41:0x3ffbdd80 0x4008e655:0x3ffbdda0
0x4008a7fb: panic_abort at /<omitted>/esp/esp-idf-v4.3.1/components/esp_system/panic.c:368

0x4008ae55: esp_system_abort at /<omitted>/esp/esp-idf-v4.3.1/components/esp_system/system_api.c:112

0x400927f6: abort at /<omitted>/esp/esp-idf-v4.3.1/components/newlib/abort.c:46

0x401663c7: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400ecaf9: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_storage.cpp:480 (discriminator 1)

0x400ecb3b: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_storage.cpp:539

0x400ed17d: nvs::NVSHandleSimple::get_typed_item(nvs::ItemType, char const*, void*, unsigned int) at /<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_handle_simple.cpp:36

0x400eb9c5: nvs_get_str_or_blob(unsigned int, nvs::ItemType, char const*, void*, unsigned int*) at /<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_api.cpp:504

0x400ebe79: nvs_get_blob at /<omitted>/esp/esp-idf-v4.3.1/components/nvs_flash/src/nvs_api.cpp:514

0x400db04e: flash_read_blob_preallocated at <omitted>/components/flash/flash.c:295
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Let me know if I can somehow get more information from NVS. I can't supply any real code or what is written in NVS.

negativekelvin, 2021-11-04:
Did you try first calling nvs_get_blob with zero length to check the size instead of assuming the size?

DCSBL, 2021-11-04:
Hi, thanks! Yes I do. I simplified the code example to much.

negativekelvin, 2021-11-04:
Looks like more checking should be done on each chunk in the read loop and asserts should be eliminated where possible.

Alvin1Zhang, 2022-07-26:
Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available https://github.com/espressif/esp-idf/commit/582f9548d628f8fb0fd60d90021e6373816abd6a, feel free to reopen.

closed 2022-07-26 by Alvin1Zhang

AxelLin, 2022-07-26:
> Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available [582f954](https://github.com/espressif/esp-idf/commit/582f9548d628f8fb0fd60d90021e6373816abd6a), feel free to reopen.

This issue was reported on v4.3.1, please also fix release/4.3 branch.

Alvin1Zhang, 2022-07-26:
> > Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available [582f954](https://github.com/espressif/esp-idf/commit/582f9548d628f8fb0fd60d90021e6373816abd6a), feel free to reopen.
> 
> This issue was reported on v4.3.1, please also fix release/4.3 branch.

Thanks for the notes, regarding release/4.3, will analyse and update later. Thanks.

DCSBL, 2022-07-26:
Thanks for the feedback!