espressif/esp-idf #7838: explanation to ESP_APP_ID define in gatt_server_service_table example

Szymono, 2021-11-04:
Hello , 

I try to understand what mean 
```*.c
#define ESP_APP_ID                  0x55
```
I cant find any explanation what exactly the value do and also why it's set to 0x55. 

I hope u can easy explain and maybe it's good idea to update [DS]:(https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatts.html?highlight=esp_ble_gatts_app_register) or [Gatt_Server_Service_Table_Example_Walkthrough.md]:(https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table/tutorial/Gatt_Server_Service_Table_Example_Walkthrough.md)

xiewenxiang, 2021-11-05:
The `ESP_APP_ID ` is application ID. It is used to distinguish between application layer callbacks. We will fill in more details in doc later.

zhp0406, 2024-03-10:
see:
a5a7e60ad0bd0b00839310b6d28a87ae80fd0e94

closed 2024-07-01 by Indastri