espressif/esp-idf #7839: SDKV4 and OTA compatibility with existing SDKV3 products using NVS and flash encryption

MartinTJDK, 2021-11-04:
We have a product using flash encryption, but no NVS encryption.

The (nvs) partitions are defined as:
otadata,   data,      ota,    0x18000,    0x2000,  encrypted
phy_init,  data,      phy,    0x1A000,    0x1000,  encrypted
nvs_keys,  data, nvs_keys,    0x1B000,    0x1000,  encrypted
nvs,       data,      nvs,    0x1C000,   0x14000,  encrypted
#factory,   app,  factory,    0x30000,   0x80000,  encrypted
ota_0,      app,    ota_0,   0x0B0000,  0x3A0000,  encrypted
ota_1,      app,    ota_1,   0x450000,  0x3A0000,  encrypted
coredump,  data, coredump,   0x7F0000,   0x10000,

The sdkconfig of SDKV3 is:
#
# Bootloader config
#
CONFIG_APP_ROLLBACK_ENABLE=y
CONFIG_APP_ANTI_ROLLBACK=y
CONFIG_APP_SECURE_VERSION=0
CONFIG_APP_SECURE_VERSION_SIZE_EFUSE_FIELD=32
CONFIG_EFUSE_SECURE_VERSION_EMULATE=

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_ENABLED=y
CONFIG_SECURE_BOOTLOADER_ONE_TIME_FLASH=y
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=
CONFIG_SECURE_BOOT_VERIFICATION_KEY="XXX-SECUREBOOT-PUBLIC-V1.bin"
CONFIG_SECURE_BOOT_INSECURE=
CONFIG_FLASH_ENCRYPTION_ENABLED=y
CONFIG_FLASH_ENCRYPTION_INSECURE=

#
# NVS
#
CONFIG_NVS_ENCRYPTION=

****************************************************************************************

The sdkconfig of SDKV4 is:
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_SIGNED_APPS_ECDSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V1_ENABLED=y
CONFIG_SECURE_BOOTLOADER_ONE_TIME_FLASH=y
# CONFIG_SECURE_BOOTLOADER_REFLASHABLE is not set
# CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES is not set
CONFIG_SECURE_BOOT_VERIFICATION_KEY="WIFI2LIN-SECUREBOOT-PUBLIC-V1.bin"
# CONFIG_SECURE_BOOT_INSECURE is not set
CONFIG_SECURE_FLASH_ENC_ENABLED=y
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
# end of Security features

#
# NVS
#
# CONFIG_NVS_ENCRYPTION is not set
# end of NVS

****************************************************************************************

When updating a SDKV3 device to an SDKV4 application via OTA, the new application is launched, but as soon as NVS initialization code is called:

// Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

The bootloader reverts to old version.

Is it impossible to build a SDKV4 software, which is compatible with existing NVS content and partition setup, which worked in SDKV3 application?

Any clues are appreciated.

negativekelvin, 2021-11-04:
> as soon as NVS initialization code is called: The bootloader reverts to old version.

You mean on the next boot the old app will run or...? Do you have logs?

BitProgress, 2021-11-04:
Yes.
During the first run of new app, the NVS initialization fails with an error, that nvs partition cannot be encrypted:

ESP_ERROR_CHECK failed: esp_err_t 0x1119 (ESP_ERR_NVS_WRONG_ENCRYPTION) at 0x40095388
file: "../src/main/esp32init.c" line 20

That causes device to reboot, and the the old app runs again (due to bootloader feature, where new app must confirm successful boot).
But I have disabled NVS encryption, which was also disabled in old app.
The flash is, however, encrypted via the built-in hardware encryption.
I guess I need to add debug code and then figure out why the NVS partition cannot have the encrypted flag set...
But I find it strange, and not optimal, if I cannot update SDKV3 devices....

negativekelvin, 2021-11-04:
Nvs partition is never supposed to have encrypted flag set, only keys partition.  Encrypted nvs uses separate software encryption not hardware flash encryption.

BitProgress, 2021-11-04:
I understand that - now. But it was not a problem in SDKV3. And did not report any compile error.
But when NVS encryption is not used, why did the SDK change behaviour?
It worked before...

negativekelvin, 2021-11-04:
Probably due to the unified partition api, it should be easy to patch it to ignore the encrypted flag on nvs partitions here
https://github.com/espressif/esp-idf/blob/417ef24b0641856ea71a7aeb44ccee62b11b5cc8/components/spi_flash/partition.c#L227-L237

kayzen9, 2021-11-05:
https://github.com/espressif/esp-idf/issues/5747#issuecomment-956244068

igrr, 2021-11-05:
@negativekelvin 's suggestion is probably better than ad-hoc patching of partition data structure, I think we'll implement it.

BitProgress, 2021-11-09:
@negativekelvin @igrr Thank you so much for the help.
We have tried to make patching of partitions before, but I think we would need temporary storage of the NVS data as well. So not an easy task.

@igrr When do you think Espressif will decide, if the SDK is updated to allow secure "nvs" partition, if encrypted NVS is disabled? And do you think it is risky to follow this "incorrect" use of NVS? The more devices, we have out, the more risky the partition update is...

negativekelvin, 2021-11-09:
@BitProgress nvs partition is never hardware encrypted, so it is safe for the sdk to ignore the flag in the partition table.  That's all the patch would do, there is no rewriting of partitions.

BitProgress, 2021-11-09:
@negativekelvin Thank you for the information.

I thought flash handling obeyed the "flash" encryption flag.
But to update existing devices with NVS encryption, we first need to store the NVS content somewhere else, then load a new ota image (with NVS encryption enabled) and then save the data in NVS again. Right?

negativekelvin, 2021-11-09:
Yes if you want to achieve "encrypt in place" you would have to copy the data, reinitialize the partition, and rewrite all the data.  You may be able to come up with a way to do it more efficiently one page at a time if you want to dig into low level stuff.

mahavirj, 2021-11-11:
@BitProgress 

Could you please help to check attach patch 
[0001-nvs-add-config-to-ignore-encrypted-flag-of-nvs-parti.patch.tar.gz](https://github.com/espressif/esp-idf/files/7518429/0001-nvs-add-config-to-ignore-encrypted-flag-of-nvs-parti.patch.tar.gz)?

- This is added under new config option, please enable it before building the application
- Patch is against `release/v4.3` branch

MartinTJDK, 2021-11-11:
@mahavirj I confirm, that the system now allows a new update built on SDK V4.3.1.
We will move forward to perform an update of partition table (also ad-hoc patching), and will try to update NVS content to use encryption.

Thanks a lot for the help.

closed 2021-11-12 by espressif-bot