espressif/esp-idf #7840: esp-idf "sniffer" core panic

JorgeDismuntel, 2021-11-04:
** Environment**

- Module or chip used: ESP32-WROVER-B
- IDF version: ESP-IDF v4.4-dev-3235-g3e370c4296-dirty
- Operating System: Windows
- Environment type: ESP Command Prompt
- Using an IDE?: Yes, Eclipse-workspace
- Power Supply: external 5V

 **Problem Description**
I'm using an example of a WiFi Sniffer, in which the mic randomly panics and reboots on its own. 

**### Expected Behavior**
It should work without a problem and without restarting.

**### Actual Behavior**
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

**### Steps to reproduce**
Use "simple_sniffer" to reproduce.
1. Create a project with that example.
2. Compile the project with this small modification in the cmd_sniffer.c file
`static void wifi_sniffer_cb(void *recv_buf, wifi_promiscuous_pkt_type_t type)
{
	sniffer_packet_info_t packet_info;
	wifi_promiscuous_pkt_t *sniffer = (wifi_promiscuous_pkt_t *)recv_buf;
	/* prepare packet_info */
	packet_info.seconds = sniffer->rx_ctrl.timestamp / 1000000U;
	packet_info.microseconds = sniffer->rx_ctrl.timestamp % 1000000U;
	packet_info.length = sniffer->rx_ctrl.sig_len;
	
	size_t VAR_1 = 0;
	size_t VAR_2 = 0;
	size_t VAR_3 = 0;
	size_t VAR_4 = 0;
	
	VAR_1 = heap_caps_get_free_size(MALLOC_CAP_8BIT);
	//	ESP_LOGI(TAG, "MALLOC_CAP_8BIT = %d", VAR);
	VAR_2 = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
	//	ESP_LOGI(TAG, "MALLOC_CAP_SPIRAM = %d", VAR);
	VAR_3 = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
	//	ESP_LOGI(TAG, "MALLOC_CAP_INTERNAL = %d", VAR);
	VAR_4 = uxTaskGetNumberOfTasks();
	//	ESP_LOGI(TAG, "Number of tasks: %d", VAR);

	/* For now, the sniffer only dumps the length of the MISC type frame */
	if (type != WIFI_PKT_MISC && !sniffer->rx_ctrl.rx_state) {
		packet_info.length -= SNIFFER_PAYLOAD_FCS_LEN;
		/*queue_packet(recv_buf, &packet_info);*/
		ESP_LOGW("MEMORY STATUS", "%zu:%zu\t%zu\t%10d\t%10d\t%10d\t%10d", packet_info.seconds, packet_info.microseconds, packet_info.length, VAR_1, VAR_2, VAR_3, VAR_4);
	}
}`
3. When it will working, just wait few minutes and it will hit panic.

## Debug Logs

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400927ca  PS      : 0x00060c33  A0      : 0x800932be  A1      : 0x3ffc6e70
0x400927ca: remove_free_block at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:213
 (inlined by) block_locate_free at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:757

A2      : 0x3ffb6edc  A3      : 0x00000158  A4      : 0x3ffb3b64  A5      : 0x00060c20
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x3ffba3dc  A9      : 0x00000000
A10     : 0x73cca255  A11     : 0x00000006  A12     : 0xbc34f7be  A13     : 0x3ffdd830
A14     : 0x3ffdd84c  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
EXCVADDR: 0xbc34f7ca  LBEG    : 0x4008a6e4  LEND    : 0x4008a700  LCOUNT  : 0xffffffff
0x4008a6e4: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x4008a700: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace:0x400927c7:0x3ffc6e700x400932bb:0x3ffc6e90 0x40082195:0x3ffc6eb0 0x400821d8:0x3ffc6ed0 0x40094a91:0x3ffc6ef0 0x4008730d:0x3ffc6f10 0x4011a4a5:0x3ffc6f30 0x4009a205:0x3ffc6f70 0x4009a42b:0x3ffc6fb0 0x40097682:0x3ffc6ff0 0x40091089:0x3ffc7020
0x400927c7: remove_free_block at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:212
 (inlined by) block_locate_free at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_tlsf.c:757

0x400932bb: multi_heap_malloc_impl at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/multi_heap.c:197

0x40082195: heap_caps_malloc at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_caps.c:145

0x400821d8: heap_caps_malloc_default at C:/Users/jorge.ibanez/Desktop/esp-idf/components/heap/heap_caps.c:181

0x40094a91: malloc at C:/Users/jorge.ibanez/Desktop/esp-idf/components/newlib/heap.c:31

0x4008730d: wifi_malloc at C:/Users/jorge.ibanez/Desktop/esp-idf/components/esp_wifi/esp32/esp_adapter.c:80

0x4011a4a5: wDev_SnifferRxData at ??:?

0x4009a205: wDev_ProcessRxSucData at ??:?

0x4009a42b: wdevProcessRxSucDataAll at ??:?

0x40097682: ppTask at ??:?

0x40091089: vPortTaskWrapper at C:/Users/jorge.ibanez/Desktop/esp-idf/components/freertos/port/xtensa/port.c:159

ELF file SHA256: 2c7ac03d5afd6523

Rebooting...
ets Jun  8 2016 00:22:57

```

## Other items if possible
https://drive.google.com/file/d/1d8bupp_MYrGIjRr4-ycocScSOuNHKVbK/view?usp=sharing

JorgeDismuntel, 2021-11-05:
I attach the modified program of the example. 
https://drive.google.com/file/d/1Nqui8i1d3trVatKKBvKprsUmRKqYs97Y/view?usp=sharing

mhdong, 2021-11-05:
hi @JorgeDismuntel  sniffer 802.11 error frame, such as the frame with a CRC error, may cause crash.
You can decide which specific type of packets can be filtered to the application by using esp_wifi_set_promiscuous_filter() and esp_wifi_set_promiscuous_ctrl_filter().

JorgeDismuntel, 2021-11-08:
Hi @mhdong, it's working, now it don't reset.
Thank you very much

Alvin1Zhang, 2021-11-29:
@JorgeDismuntel Thanks for reporting, and sharing the updates, feel free to reopen.

closed 2021-11-29 by Alvin1Zhang