espressif/esp-idf #7841: Unable to read GPIO13 state on ESP32-WROVER-E

romatetemadze, 2021-11-04:
- Development Kit:      Custom ESP32-WROVER-E Board
- IDF version: idf4.2.1
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows 10
- ESP Command Prompt
- Using an IDE?: Eclipse IDE
- Power Supply: external 30A+

**I am unable to read GPIO13 high/low state**
Tested with GPIO4 and GPIO35 same code - everything works fine.

My configuration uses Secure boot V2 with JTAG disabled. I do realize that GPIO is part of JTAG but I couldn't find any way to use it as an INPUT only pin and read its state. Below you will find the schematics along with the code.

_Initialization code. Called pretty much by the end of the app_main():_

`#define GPIO_INPUT_IGNITION     GPIO_NUM_13  // Works as expected with GPIO_NUM_4 and GPIO_NUM_35
#define GPIO_INPUT_IGNMSK  ((1ULL<<GPIO_INPUT_IGNITION))

void initSensor() {
    gpio_config_t io_conf;
    io_conf.pin_bit_mask = GPIO_INPUT_IGNMSK;
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 0;
    io_conf.pull_down_en = 0;
    gpio_config(&io_conf);
    logToRam(true,"*SYS* MONITOR ON.\r\n");
    enableIgnitionMonitor = true;
    //PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[GPIO_INPUT_IGNITION], PIN_FUNC_GPIO);
}`

_Here goes the pooling in the main loop:_
`    	
while (true) {
    	if (true) {
    		gpio_pad_select_gpio(GPIO_INPUT_IGNITION);
    		int v = gpio_get_level(GPIO_INPUT_IGNITION);
    		if (v==1) {
    			logToRam(true,"*SIGNAL* ON\r\n");
    		} else {
    			logToRam(true,"*SIGNAL* OFF\r\n");
    		}
    	}
       vTaskDelay(pdMS_TO_TICKS(1000));
}
`

_I have tried to reinit GPIO matrix for 13th with following call inside the initSensor but it didn't help:_
`PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[GPIO_INPUT_IGNITION], PIN_FUNC_GPIO);`

Output is always LOW when GPIO13 is used (unless I pull it UP with gpio_config, and everything is okay if I use GPIO4 or GPIO35 (pretty sure more will work too). Attached you may find the sdkconfig file, fuses summary and relevant schematics.

HIghly appreciate any help.

![fuses](https://
[image]
user-images.githubusercontent.com/65067064/140418301-f698c379-29de-41b9-8a20-260c9b6575c5.png)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7478416/sdkconfig.txt)

nx518, 2021-11-05:
On IO13 you have 10K/57K*12V=2.1V which could be read as low. You need at least 2.475V @ 3V3 Supply to get a defined high level. Also double check diode polarity.

[image]

romatetemadze, 2021-11-05:
> On IO13 you have 10K/57K*12V=2.1V which could be read as low. You need at least 2.475V @ 3V3 Supply to get a defined high level. Also double check diode polarity.
> 
> [image]

Thanks for your reply. Apparenly I misread the datasheet (only read the 0.75 and assumed that was the V threshold for HIGH). 

Any suggestion why pins GPIO4 ans GPIO35 are okay still reading as high in identical setup?

nx518, 2021-11-05:
> Any suggestion why pins GPIO4 ans GPIO35 are okay still reading as high in identical setup?

Input voltage is in range of undefined behaviour, this should be fixed first. 

Also GPIO4 and GPIO35 are function0 and GPIO13 is function2 in IO_MUX according to ESP32 datasheet.

romatetemadze, 2021-11-05:
> > Any suggestion why pins GPIO4 ans GPIO35 are okay still reading as high in identical setup?
> 
> Input voltage is in range of undefined behaviour, this should be fixed first.
> 
> Also GPIO4 and GPIO35 are function0 and GPIO13 is function2 in IO_MUX according to ESP32 datasheet.

Thanks. I did change the input to always output VDD-0.2v, which I believe is sufficient, which solved the issue of GPIO13. Looking for a clarification on different behaviour of GPIO13 vs 4 and 35.

Where could I observe more information about the function0 vs function2? I want to understand the difference in the HIGH/LOW reading among the IO's mentioned above.

nx518, 2021-11-05:
> I want to understand the difference in the HIGH/LOW reading among the IO's mentioned above.

The reason for difference in HIGH/LOW reading is the undefined behaviour for this input voltage, resulting in highly implementation-specific and enviroment-dependent circuit behavior. In this state even small changes in temperature or load on other GPIOs may lead to reading different input levels.

If you understood the concept of undefined behaviour and still have doubt check the [Technical Reference Manual Chapter 4 IO_MUX and GPIO Matrix
](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) for the difference between function0 vs function2.

Alvin1Zhang, 2021-11-25:
@romatetemadze Thanks for reporting and thanks to @nx518  for the notes, feel free to reopen.

closed 2021-11-25 by Alvin1Zhang