espressif/esp-idf #7842: How to use function esp_ble_gatts_add_char twince in examples\bluetooth\bluedroid\ble\gatt_server

GKoSon, 2021-11-05:
- Development Kit:    ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  ESP32-WROOM-32
hi

when i run 
examples\bluetooth\bluedroid\ble\gatt_server\main\gatts_demo.c
examples\bluetooth\bluedroid\ble\gatt_client\main\gattc_demo.c
they are work well.

Now i wanna to connect with Nordic soc ,I wanna change the UUID to Nordic NUS.
I have nrf Connect APP in my android phone.
the right ble i wanna is like below(use nrf Connect APP )
[image]

when i change the code examples\bluetooth\bluedroid\ble\gatt_server
i find the function  esp_ble_gatts_add_char  only used once
so only have Rx    the Tx cannot added

[image]

my code
`

static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param) {
    uint8_t custs1_svc1[16]= {0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5,    0x01, 0x00, 0x40, 0x6E};
    uint8_t adv_char_uuid128[16]= {0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5,    0x02, 0x00, 0x40, 0x6E};
    uint8_t adv_char_uuid1282[16]= {0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5,    0x03, 0x00, 0x40, 0x6E};

    switch (event) {

    case ESP_GATTS_CREATE_EVT:
    

        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT--, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_128;
        memcpy(gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid128,adv_char_uuid128,ESP_UUID_LEN_128);
        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);
        a_property = ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY;
        esp_err_t add_char_ret = esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                                                        ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                                        a_property,
                                                        &gatts_demo_char1_val, NULL);
        if (add_char_ret){
            ESP_LOGE(GATTS_TAG, "add char failed, error code =%x",add_char_ret);
        }

        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT----, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_128;
        memcpy(gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid128,adv_char_uuid1282,ESP_UUID_LEN_128);
        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);
        a_property = ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY;
        esp_err_t add_char_ret2 = esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                                                        ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                                        a_property,
                                                        &gatts_demo_char1_val, NULL);
        if (add_char_ret2){
            ESP_LOGE(GATTS_TAG, "add char failed, error code =%x",add_char_ret2);
        }

        break;`
How can i add 2 char to a service ？
thanks

GKoSon, 2021-11-05:
PS:
i read nordic code  he run function characteristic_add twince to add 2 char in 1 service

\nRF5_SDK_17.1.0_ddde560\components\ble\ble_services\ble_nus\ble_nus.c

#define NUS_BASE_UUID                  {{0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5, 0x00, 0x00, 0x40, 0x6E}} /**< Used vendor specific UUID. */
ble_nus_init
    sd_ble_uuid_vs_add
    sd_ble_gatts_service_add
    characteristic_add(p_nus->service_handle, &add_char_params, &p_nus->rx_handles)
    characteristic_add(p_nus->service_handle, &add_char_params, &p_nus->tx_handles);

GKoSon, 2021-11-05:
examples\bluetooth\bluedroid\ble\gatt_server_service_table\main\gatts_table_creat_demo.c
can work out

please close this issue 
thanks

closed 2021-11-05 by GKoSon

Azuresword, 2024-05-01:
but in my program i can not use gatts_table,how i can slove this problem