espressif/esp-idf #7844: ESP32-WROOM-32UE doesn't work with CYW89359 bluetooth chip

7erduk, 2021-11-05:
Hello Guys,

Could you help me? ESP32_DevKitc_v4 doesn't work with Bluetooth chip (My VW ID.3 car) CYW89359 - https://www.cypress.com/file/350976/download

I have ESP32_DevKitc_v4 (ESP32-WROOM-32UE).

Build example from the master https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart_esp32 and flashed the device.

Connected it to BlueZ stack 

sudo btattach -B /dev/ttyUSB1 -S 921600

My ESP32 device:

sudo hciconfig -a hci1

hci1:	Type: Primary  Bus: UART
	BD Address: E0:E2:E6:B2:E3:06  ACL MTU: 1021:9  SCO MTU: 255:4
	UP RUNNING 
	RX bytes:1847 acl:0 sco:0 events:106 errors:0
	TX bytes:4038 acl:0 sco:0 commands:105 errors:0
	Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'ubuntu #2'
	Class: 0x0c0000
	Service Classes: Rendering, Capturing
	Device Class: Miscellaneous, 
	HCI Version: 4.2 (0x8)  Revision: 0x30e
	LMP Version: 4.2 (0x8)  Subversion: 0x30e
	Manufacturer: RivieraWaves S.A.S (96)

It works well with my Raspberry Pi Bluetooth, Android phone, iPhone...

But it doesn't work with Bluetooth chip (My VW ID.3 car) CYW89359.

This is the information I requested from Raspberry Pi Bluetooth, [b]not via esp32!!![/b]
root@raspberrypi-bt:~# hcitool info "18:48:CA:10:7E:AC"
Requesting information ...
        BD Address:  18:48:CA:10:7E:AC
        Device Name: My VW 3816
        LMP Version: 4.2 (0x8) LMP Subversion: 0x420d
        Manufacturer: Broadcom Corporation (15)
        Features page 0: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
                <3-slot packets> <5-slot packets> <encryption> <slot offset>
                <timing accuracy> <role switch> <sniff mode> <RSSI>
                <channel quality> <SCO link> <HV2 packets> <HV3 packets>
                <u-law log> <A-law log> <CVSD> <paging scheme> <power control>
                <transparent SCO> <broadcast encrypt> <EDR ACL 2 Mbps>
                <EDR ACL 3 Mbps> <enhanced iscan> <interlaced iscan>
                <interlaced pscan> <inquiry with RSSI> <extended SCO>
                <EV4 packets> <EV5 packets> <AFH cap. slave>
                <AFH class. slave> <LE support> <3-slot EDR ACL>
                <5-slot EDR ACL> <sniff subrating> <pause encryption>
                <AFH cap. master> <AFH class. master> <EDR eSCO 2 Mbps>
                <EDR eSCO 3 Mbps> <3-slot EDR eSCO> <extended inquiry>
                <LE and BR/EDR> <simple pairing> <encapsulated PDU>
                <err. data report> <non-flush flag> <LSTO> <inquiry TX power>
                <EPC> <extended features>
        Features page 1: 0x09 0x00 0x00 0x00 0x00 0x00 0x00 0x00
        Features page 2: 0x7f 0x0b 0x00 0x00 0x00 0x00 0x00 0x00

Could you help me, maybe I should tunning some parameters in ESP32 firmware?

`Executing action: monitor
Running idf_monitor in directory /home/user/esp/esp-idf/examples/bluetooth/hci/controller_hci_uart_esp32
Executing "/home/user/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/user/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/user/esp/esp-idf/examples/bluetooth/hci/controller_hci_uart_esp32/build/controller_hci_uart.elf -m '/home/user/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/user/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6744
load:0x40078000,len:14812
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4-dev-3401-gb86fe0c66c-dirty 2nd stage bootloader
I (29) boot: compile time 00:52:30
I (30) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b954h ( 47444) map
I (131) esp_image: segment 1: paddr=0001b97c vaddr=3ffbdb60 size=04604h ( 17924) load
I (138) esp_image: segment 2: paddr=0001ff88 vaddr=40080000 size=00090h (   144) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=35e14h (220692) map
I (225) esp_image: segment 4: paddr=00055e3c vaddr=40080090 size=16888h ( 92296) load
I (263) esp_image: segment 5: paddr=0006c6cc vaddr=50000000 size=00010h (    16) load
I (274) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (286) cpu_start: Pro cpu up.
I (286) cpu_start: Starting app cpu, entry point is 0x4008115c
0x4008115c: call_start_cpu1 at /home/user/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (300) cpu_start: Pro cpu start user code
I (300) cpu_start: cpu freq: 160000000
I (300) cpu_start: Application information:
I (305) cpu_start: Project name:     controller_hci_uart
I (311) cpu_start: App version:      v4.4-dev-3401-gb86fe0c66c-dirty
I (318) cpu_start: Compile time:     Oct 15 2021 02:07:23
I (324) cpu_start: ELF file SHA256:  7870c22910fcaf00...
I (330) cpu_start: ESP-IDF:          v4.4-dev-3401-gb86fe0c66c-dirty
I (337) heap_init: Initializing. RAM available for dynamic allocation:
I (344) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (350) heap_init: At 3FFB7CD8 len 00000328 (0 KiB): DRAM
I (356) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (363) heap_init: At 3FFC37E8 len 0001C818 (114 KiB): DRAM
I (369) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (375) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (382) heap_init: At 40096918 len 000096E8 (37 KiB): IRAM
I (389) spi_flash: detected chip: generic
I (392) spi_flash: flash io: dio
W (396) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (411) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (444) CONTROLLER_UART_HCI: HCI UART1 Pin select: TX 5, RX, 18, CTS 23, RTS 19
I (454) BTDM_INIT: BT controller compile version [078d492]
I (454) system_api: Base MAC address is not set
I (454) system_api: read default base MAC address from EFUSE
I (464) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
`

Best Regards.

7erduk, 2021-11-08:
Hello for fast answer,

Doesn’t work mean, I can’t connect, request name, create pair.
For example i can’t request name

hcitool name  18:48:CA:10:7E:AC

I don’t see any communication with chip with hcidump, I think problem in Inquiry/Page procedure.

How can I collect additional logs Inquiry/Page procedure?

Best Regards

Wth-Esp, 2021-11-08:
@7erduk  We will look into this issue. Please wait for the reply.

BetterJincheng, 2021-11-08:
Hi @7erduk, 
It will do help if you provide some LOGs with `btmon`. I need the logs in btmon CLI when you run `hcitool name 18:48:CA:10:7E:AC` in another CLI.

7erduk, 2021-11-08:
Hello,

btmon logs

hcitool name 18:48:CA:10:7E:AC

`Bluetooth monitor ver 5.53
= Note: Linux version 5.8.0-50-generic (x86_64)                                                                                                                                                   0.567270
= Note: Bluetooth subsystem version 2.22                                                                                                                                                          0.567272
= New Index: E0:E2:E6:B2:E3:06 (Primary,UART,hci1)                                                                                                                                         [hci1] 0.567273
= Open Index: E0:E2:E6:B2:E3:06                                                                                                                                                            [hci1] 0.567273
= Index Info: E0:E2:E6:B2:E3:06 (RivieraWaves S.A.S)                                                                                                                                       [hci1] 0.567273
= New Index: C8:58:C0:25:B9:BB (Primary,USB,hci0)                                                                                                                                          [hci0] 0.567274
@ RAW Open: dumpcap (privileged) version 2.22                                                                                                                                     {0x0002} [hci0] 0.567274
@ MGMT Open: bluetoothd (privileged) version 1.17                                                                                                                                        {0x0001} 0.567275
@ MGMT Open: btmon (privileged) version 1.17                                                                                                                                             {0x0003} 0.567285
@ RAW Open: hcitool version 2.22                                                                                                                                                         {0x0004} 3.608963
@ RAW Close: hcitool                                                                                                                                                                     {0x0004} 3.609330
@ RAW Open: hcitool version 2.22                                                                                                                                                  {0x0004} [hci1] 3.609386
< HCI Command: Remote Name Request (0x01|0x0019) plen 10                                                                                                                                #1 [hci1] 3.609737
        Address: 18:48:CA:10:7E:AC (Murata Manufacturing Co., Ltd.)
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                               #2 [hci1] 3.623396
      Remote Name Request (0x01|0x0019) ncmd 5
        Status: Success (0x00)
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                                   #3 [hci1] 8.750761
        Status: Page Timeout (0x04)
        Address: 18:48:CA:10:7E:AC (Murata Manufacturing Co., Ltd.)
        Name: 
@ RAW Close: hcitool                                                                                                                                                              {0x0004} [hci1] 8.750893
@ RAW Open: hcitool version 2.22                                                                                                                                                        {0x0004} 10.148010
@ RAW Close: hcitool                                                                                                                                                                    {0x0004} 10.148103
@ RAW Open: hcitool version 2.22                                                                                                                                                 {0x0004} [hci1] 10.148131
< HCI Command: Remote Name Request (0x01|0x0019) plen 10                                                                                                                               #4 [hci1] 10.148173
        Address: 18:48:CA:10:7E:AC (Murata Manufacturing Co., Ltd.)
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                              #5 [hci1] 10.157088
      Remote Name Request (0x01|0x0019) ncmd 5
        Status: Success (0x00)
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                                  #6 [hci1] 15.288187
        Status: Page Timeout (0x04)
        Address: 18:48:CA:10:7E:AC (Murata Manufacturing Co., Ltd.)
        Name: 
[... cut ...]
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                             #11 [hci1] 58.842467
      Remote Name Request (0x01|0x0019) ncmd 5
        Status: Success (0x00)
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                                 #12 [hci1] 63.979359
        Status: Page Timeout (0x04)
        Address: 18:48:CA:10:7E:AC (Murata Manufacturing Co., Ltd.)
        Name: 
@ RAW Close: hcitool 
`

7erduk, 2021-11-09:
> @7erduk We will look into this issue. Please wait for the reply.
> 
> For logs, maybe you can capture some HCI logs using `WireShark`.

snoop file with hcidump tool
https://mega.nz/file/9jpWFarS#JHGPQDVgyzHKDeE6lG0hX_duSjRa2SzMG_4JmHi3VAM

7erduk, 2021-11-09:
Also, you can see HCI log from raspberry pi, it is **not ESP32**

The car chip CYW89359 answer on hcitool name 18:48:CA:10:7E:AC
https://mega.nz/file/U3hwwYAC#gPwZNs0_CAoxLCzITBA0kZl8e7bJpEY4XCR0VYLpRJg

7erduk, 2021-11-09:
Maybe I can somehow to make a log with Inquiry/Page procedure. Do you have any idea?

BetterJincheng, 2021-11-11:
@7erduk,

From the btmon LOGs, the command was just correctly passed to controller (ESP32) from host stack. And the Command Status Event returned successfully. All the things looks like the systems worked well. 

As for "HCI Event: Remote Name Req Complete" with error code "Page Timeout", it was just "page timeout".

From my perspective, there is no problem on Interoperability between the host and controller. Maybe you should check it again.

Thanks.

7erduk, 2021-11-11:
> @7erduk,
>
> From the btmon LOGs, the command was just correctly passed to controller (ESP32) from host stack. And the Command Status Event returned successfully. All the things looks like the systems [...]

Hello BetterJincheng,

I have tested it **many many  many** times, I have only got "Page Timeout".  Other my devices don't have such a problem with CYW89359.

7erduk, 2021-11-11:
Hello Guys,

I have connected an antenna to my esp32, Now everything is working well. Thank you!

Alvin1Zhang, 2021-11-12:
Thanks for reporting, and glad to hear everything is working well now, feel free to reopen.

closed 2021-11-12 by Alvin1Zhang