espressif/esp-idf #7845: tp_read_main.c locks itself out ...

franz-ms-muc, 2021-11-05:
https://github.com/espressif/esp-idf/tree/v4.3.1/examples/peripherals/touch_pad_read/main/esp32s2
tp_read_main.c

Hello, 

it seems the internal Driver locks itself out when it sees VERY High values. 

so the Touch pad Reading working very well, but if you attatch very large Pads with a very big Finger (more Hand) 
you get Reading Values of e.g. 2699993 and then you get the same Values for all Channels until restart. 

it is like the Measuring is in a Deadlock, not updating the internal Measurements anyhow, and just giving a "freezed" value out. 

Tested on v4.3.1

Thanks, 
Franz

franz-ms-muc, 2021-11-05:
just for proofing: 

i tested with the "master" branch: ESP-IDF:          v5.0-dev-178-g417ef24b06

ESP32S2, 

```
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
T1: [85459] T2: [55743] T3: [71849] T4: [63331] T5: [1010637] T6: [1848183] T7: [2120671] T8: [1709241] T9: [249383] T10: [28832] T11: [30569] T12: [25658] T13: [7025] T14: [3315495]
```

it does also lock out. 

same as on v4.3.1

CarlosGS, 2021-11-14:
The lock seems to be happening within the `printf()` function.
But there is still an odd behavior within IDF, as the values are increasing with the pad capacity. The values should become lower when touching the pads, not larger (specially this huge).

Hopefully someone from Espressif can explain what happens when the pad capacity gets too large, and what causes this behavior. The docs are excellent and this would be a helpful addition :)

franz-ms-muc, 2021-11-19:
I made a very very simple Test: 

connected 2 Wires, 1 to GND, one to G14, 

run the Example. 

when i Touch Wire G14 all OK, Value goes up and down. 

when i Touch G14 to GND (Hard, no Resistor in Between) the Lock is there.

franz-ms-muc, 2021-11-19:
The Programm still outputs Values, but always the same !

franz-ms-muc, 2021-11-19:
so, it seems we have 2 Problems: if he see a GND: Lock, if he see a Very high Touch: Lock.

franz-ms-muc, 2021-11-19:
i found a temporary solution to get out of the Lock: 

touch\_pad\_fsm\_stop();  
touch\_pad\_fsm\_start();

whenever you do this .. the Lok is away ..

franz-ms-muc, 2021-11-19:
BUT: if one Electrode is bound to GND, the Error is not away. this helps only ig one Electrode is TEMPORARY bound to GND.