espressif/esp-idf #7846: CONFIG_ESP32_PANIC_GDBSTUB not handled by component Makefiles

HenriChataing, 2021-11-05:
## Environment

- Development Kit:  ESP32-LyraT
- Module or chip used: ESP32
- IDF version: v4.3.1-354-g98d34e5f6d
- Build System:  Make
- Compiler version: (Crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux

## Problem Description

Using the option CONFIG_ESP32_PANIC_GDBSTUB, the file `esp_system/panic.c` fails to compile
because the headers `esp_gdbstub_arch.h` and `gdbstub_target_config.h` are not found in the include paths. 
Comparing `components/esp_gdbstub/CMakeLists.txt` and `components/esp_gdbstub/component.mk` it appears
that `xtensa` and `esp32` are missing in `COMPONENT_ADD_INCLUDEDIRS`; modifying this configuration variable
to add these include paths fixes the issue. 

### Expected Behavior

Build succeeds

### Actual Behavior

Build fails

### Steps to reproduce

Build esp-idf with Make with option CONFIG_ESP32_PANIC_GDBSTUB enabled in sdkconfig.defaults

0xjakob, 2021-11-08:
@HenriChataing I could reproduce the error. We'll work on a fix for this.

HenriChataing, 2021-11-08:
Thank you for the quick followup :)

Alvin1Zhang, 2022-01-05:
Thanks for reporting, sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/6085518bcb4cdc4a639ce0135c54862a8e264f9d and is included in the IDF v4.3.2 release https://github.com/espressif/esp-idf/releases/tag/v4.3.2, feel free to reopen if the issue still happens.

closed 2022-01-05 by Alvin1Zhang