espressif/esp-idf #7847: esp32s2 wifi crash

chegewara, 2021-11-06:
To be honest im not sure if this is a real bug or problem with my app.

I am working on USBIP server app. More or less it is working, but with ACM device connected (esp32 S3) my app just crash with this backtrace:
```
abort() was called at PC 0x40024a2f on core 0
0x40024a2f: lock_acquire_generic at /home/chegewara/esp/master/components/newlib/locks.c:139

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4002cd84  PS      : 0x00060234  A0      : 0x8002befe  A1      : 0x3ffd2200  
0x4002cd84: vListInsert at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/list.c:166 (discriminator 1)

A2      : 0x3ff9ec24  A3      : 0x3ffd255c  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x0000000a  A7      : 0x000014c7  A8      : 0x3ffd255c  A9      : 0x3ffd255c  
A10     : 0x00000002  A11     : 0x00000002  A12     : 0x3ffd23b0  A13     : 0x00000003  
A14     : 0x00000000  A15     : 0x00000009  SAR     : 0x00000009  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x3ffd23b0  LEND    : 0x00000003  LCOUNT  : 0x400232f0  
0x400232f0: xt_highint4 at /home/chegewara/esp/master/components/esp_system/port/soc/esp32s2/highint_hdl.S:58

Backtrace:0x4002cd81:0x3ffd22000x4002befb:0x3ffd2220 0x4002b09b:0x3ffd2240 0x4003011c:0x3ffd2280 0x400ef562:0x3ffd22a0 0x40030096:0x3ffd22d0 0x400d5516:0x3ffd2320 0x400d5564:0x3ffd23b0 0x400272df:0x3ffd2400 0x40027381:0x3ffd2460 0x40028e08:0x3ffd2480 0x400c780b:0x3ffd24a0 0x400eb162:0x3ffd24e0 0x4002cdfd:0x3ffd2510 
0x4002cd81: vListInsert at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/list.c:166 (discriminator 1)

0x4002befb: vTaskPlaceOnEventList at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/tasks.c:3567

0x4002b09b: xQueueSemaphoreTake at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/queue.c:1688

0x4003011c: esp_log_impl_lock_timeout at /home/chegewara/esp/master/components/log/log_freertos.c:48

0x400ef562: esp_log_writev at /home/chegewara/esp/master/components/log/log.c:192

0x40030096: esp_log_write at /home/chegewara/esp/master/components/log/log.c:210

0x400d5516: lib_printf at /home/chegewara/esp/master/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)

0x400d5564: phy_printf at /home/chegewara/esp/master/components/esp_wifi/src/lib_printf.c:57

0x400272df: pll_vol_cal at ??:?

0x40027381: wifi_track_pll_cap at ??:?

0x40028e08: tx_pwctrl_background at ??:?

0x400c780b: lmacProcessTxComplete at ??:?

0x400eb162: ppTask at ??:?

0x4002cdfd: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130
```

Most logs are suppressed by `esp_log_level_set("*", ESP_LOG_NONE);` and i have just few `printf` lines. 

INFO:
- esp-idf: master,
- tcp/ip is basically copy/paste example,

Thanks

chegewara, 2021-11-07:
Ive found the cause of crash. It was related to tcp/ip buffer too small (1224 bytes), just i am not sure it should endup with crash.

xueyunfei998, 2021-11-12:
hi @chegewara 

Please provide the  commit iD and Recurrent demo, and SDK config.

I think this has something to do with phy. Maybe if it is updated to the latest commit, the problem will be solved.

chegewara, 2021-11-13:
I tested with current master branch, commit 4a011f318377ab717b9dc459019c1055602c93ff and still the same crash.

```
abort() was called at PC 0x40024a2f on core 0
0x40024a2f: lock_acquire_generic at /home/chegewara/esp/master/components/newlib/locks.c:131

Backtrace:0x400236ea:0x3ffd77000x4002a1d1:0x3ffd7720 0x40031455:0x3ffd7740 0x40024a2f:0x3ffd77c0 0x40024bad:0x3ffd77f0 0x40024c26:0x3ffd7810 0x400deca6:0x3ffd7840 0x400e1cd9:0x3ffd7b50 0x400eb1a5:0x3ffd7b80 0x40030082:0x3ffd7bb0 0x400d5776:0x3ffd7c00 0x400d57c4:0x3ffd7c90 0x400272df:0x3ffd7ce0 0x40027381:0x3ffd7d40 0x40028e08:0x3ffd7d60 0x400c79bb:0x3ffd7d80 0x40032c8d:0x3ffd7dc0 0x4002cdfd:0x3ffd7df0 
0x400236ea: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:403

0x4002a1d1: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:123

0x40031455: abort at /home/chegewara/esp/master/components/newlib/abort.c:46

0x40024a2f: lock_acquire_generic at /home/chegewara/esp/master/components/newlib/locks.c:131

0x40024bad: _lock_acquire_recursive at /home/chegewara/esp/master/components/newlib/locks.c:159

0x40024c26: __retarget_lock_acquire_recursive at /home/chegewara/esp/master/components/newlib/locks.c:315

0x400deca6: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x400e1cd9: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x400eb1a5: esp_log_writev at /home/chegewara/esp/master/components/log/log.c:200

0x40030082: esp_log_write at /home/chegewara/esp/master/components/log/log.c:210

0x400d5776: lib_printf at /home/chegewara/esp/master/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)

0x400d57c4: phy_printf at /home/chegewara/esp/master/components/esp_wifi/src/lib_printf.c:57

0x400272df: pll_vol_cal at ??:?

0x40027381: wifi_track_pll_cap at ??:?

0x40028e08: tx_pwctrl_background at ??:?

0x400c79bb: lmacProcessTxComplete at ??:?

0x40032c8d: ppTask at ??:?

0x4002cdfd: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130
```

but i found it may be issue with my code, dont know why yet, where i have `ESP_LOG_BUFFER_HEX_LEVEL` to log request about 900 bytes and at the same time other part of code trying to send data back to socket `write(_sock, (void*)req, to_write);`.
When i disable this buffer log or it is fairly short then i dont have crash, so i will close the issue and let espressif decide if its bug in wifi library or just my code.

closed 2021-11-13 by chegewara

brainstorm, 2022-04-08:
@chegewara I just faced this same issue, it turns out that if there's too much load/noise in the main 5V power rail, this problem occurs.

In other words, when I connect my power meter board to the ESP32S2 Saola-1 board (as in being an external load), I get infinite reboots referring to `phy_printf`. Then if I only connect the Saola board alone, everything works fine, the application runs on ESP32S2 just fine without reboots related to PHY.

Conclusion: Feeding your devices with a stable power source instead of the board's PSU should solve this issue (in my repeatable setup/experience):

[image]

```
I (1360) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
D (1370) nvs: nvs_open_from_partition rmaker_time 0
D (1370) nvs: nvs_get_str_or_blob tz_posix
D (1370) nvs: nvs_get_str_or_blob tz_posix
D (1380) nvs: nvs_close 11
I (1380) esp_rmaker_core: Starting RainMaker Work Queue task
I (1390) esp_rmaker_work_queue: RainMaker Work Queue task started.
D (1400) wifi_prov_mgr: execute_event_cb : 0
D (1400) event: running post WIFI_PROV_EVENT:0 with handler 0x4008e1a8 and context 0x3ffd7f0c on loop 0x3ffd6f98
D (1410) event: running post WIFI_PROV_EVENT:0 with handler 0x40088ff0 and context 0x3ffe0e88 on loop 0x3ffd6f98
D (1420) wifi_prov_mgr: Found Wi-Fi SSID     : pinha
D (1420) wifi_prov_mgr: Found Wi-Fi Password : 6*************Q
I (1430) app_wifi: Already provisioned, starting Wi-Fi STA
D (1440) wifi_prov_mgr: Provisioning not running
D (1440) wifi_prov_mgr: execute_event_cb : 6
D (1450) event: running post WIFI_PROV_EVENT:6 with handler 0x4008e1a8 and context 0x3ffd7f0c on loop 0x3ffd6f98
D (1460) nvs: nvs_get opmode 1
D (1460) ADC: Wi-Fi takes adc2 lock.
I (1460) phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07
D (1470) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 18 bit
D (1480) phy_init: loading PHY init data from partition at offset 0x18000
D (1480) phy_init: PHY data partition validated
D (1490) nvs: nvs_open_from_partition phy 0
D (1490) nvs: nvs_get cal_version 4
D (1500) nvs: nvs_get_str_or_blob cal_mac
D (1500) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit
D (1510) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit
D (1510) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit
D (1520) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit
D (1530) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit
D (1540) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit
D (1540) nvs: nvs_get_str_or_blob cal_data
D (1550) nvs: nvs_close 12
D (1550) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit
D (1560) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit
D (1560) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit
D (1570) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit
D (1580) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit
D (1580) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit
I (1600) phy: pll_cap_ext 10
I (1600) phy: pll_cap_ext 10
I (1600) phy: pll_cap_ext 10
I (1610) phy: pll_cap_ext 10
I (1610) phy: pll_cap_ext 10
I (1610) phy: pll_cap_ext 10
I (1620) phy: pll_cap_ext 10
I (1620) phy: pll_cap_ext 10
I (1620) phy: pll_cap_ext 10
I (1630) phy: pll_cap_ext 10
I (1630) phy: pll_cap_ext 10
I (1630) phy: pll_cap_ext 10
I (1640) phy: pll_cap_ext 10
I (1650) phy: pll_cap_ext 10
I (1650) phy: pll_cap_ext 10
I (1650) phy: pll_cap_ext 10
I (1650) phy: pll_cap_ext 10
I (1650) phy: pll_cap_ext 10
I (1670) phy: pll_cap_ext 10
I (1680) phy: pll_cap_ext 10
I (1680) phy: pll_cap_ext 10
I (1680) phy: pll_cap_ext 10
[... 42 lines cut ...]
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40023651
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1478
load:0x4004c000,len:0xb84
load:0x40050000,len:0x3430
entry 0x4004c240
I (24) boot: ESP-IDF 4.4.0 2nd stage bootloader
I (24) boot: compile time 19:59:39
I (24) boot: chip revision: 0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (71) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (79) boot:  3 phy_init         RF data          01 01 00018000 00001000
```