espressif/esp-idf #7848: When installing specific tools, export.sh fails

shehzaman, 2021-11-06:
Hello,

I am following instructions from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-set-up-tools

The `install.sh` script supports installing specific tools, I tried installing `esp32` and none of the other tools.

OS: Ubuntu 18.04 (Docker container)
Python: 3.6.9
esp-idf version: 417ef24b0641856ea71a7aeb44ccee62b11b5cc8

Here are the commands I ran:
```bash

cd ~/esp

git clone -b master --depth=1 --recursive https://github.com/espressif/esp-idf.git

cd esp-idf

./install.sh esp32                            

Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Installing ESP-IDF tools
WARNING: File /root/.espressif/idf-env.json was not found. 
Creating /root/.espressif/idf-env.json
WARNING: File /root/.espressif/idf-env.json can not be created. 
Selected targets are: 
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2021r2-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2021r2-linux-amd64.tar.gz to /root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-linux-amd64.tar.gz.tmp
Done
Extracting /root/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-linux-amd64.tar.gz to /root/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0
Installing esp32ulp-elf@2.28.51-esp-20191205
Downloading binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205.tar.gz to /root/.espressif/dist/binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205.tar.gz.tmp
Done
Extracting /root/.espressif/dist/binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205.tar.gz to /root/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205
Installing openocd-esp32@v0.10.0-esp32-20210902
Downloading openocd-esp32-linux64-0.10.0-esp32-20210902.tar.gz to /root/.espressif/dist/openocd-esp32-linux64-0.10.0-esp32-20210902.tar.gz.tmp
Done
Extracting /root/.espressif/dist/openocd-esp32-linux64-0.10.0-esp32-20210902.tar.gz to /root/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902
Installing Python environment and packages
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessful: b''
...
... (things that ran successfully)
...
All done! You can now run:

  . ./export.sh

. ./export.sh

Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Adding ESP-IDF tools to PATH...
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/root/esp/esp-idf/install.sh' to install it.
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '/root/esp/esp-idf/install.sh' to install it.
ERROR: tool riscv32-esp-elf has no installed versions. Please run '/root/esp/esp-idf/install.sh' to install it.
ERROR: tool esp32s2ulp-elf has no installed versions. Please run '/root/esp/esp-idf/install.sh' to install it.
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessful: b''
```

The `. ./export.sh` command fails because it is unable to find tools, that I did not install. I do not need any of the other tools. 

Did it fail because `idf-env.json` could not be generated? I searched this repository, issues, commits and could not find any mention of `idf-env.json` anywhere.

I looked at command line tools for `export` in `idf_tools.py` - https://github.com/espressif/esp-idf/blob/417ef24b0641856ea71a7aeb44ccee62b11b5cc8/tools/idf_tools.py#L1703 
There is no command line option to mention which tools have been installed.

I'd be very obliged if anyone could help me run the export script correctly.

shehzaman, 2021-11-08:
Hi @dobairoland thank you for taking a look into this. 

You are absolutely right! 

The script that creates `idf-env.json` expects `/root/.espressif` folder to be present. Since creation of `idf-env.json` precedes the directory creation in `idf_tools.py` it fails.

Adding `mkdir /root/.espressif` before `install.sh` fixes it.

 Ah, I should've taken a look at the docker folder makes it so much easier!

I'll close the issue as it is solved.

closed 2021-11-08 by shehzaman

dobairoland, 2021-11-08:
Thanks @shehzaman for the feedback. I'll create a patch so `mkdir` won't be necessary in the future.