espressif/esp-idf #7849: Failed to open new connection with OTA after upgrade to v4.3.1

z7jzk, 2021-11-07:
## Environment

- Module or chip used:    ESP32-WROOM-32D
- IDF version:    v4.3.1-354-g98d34e5f6d
- Build System:    idf.py
- Compiler version:    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     macOS
- Using an IDE?:    Yes: VS Code
- Power Supply:         USB

## Problem Description

OTA is no longer working in my project after upgrading to v4.3.1 just over a month ago. Getting the below error (in Debug Logs) after getting an IP HTTP callback and after running `esp_https_ota_begin(&ota_config, &ota_handle)`.

### Expected Behavior

Expect to connect to WiFi network, start OTA connection, populate OTA partition and restart device.

### Actual Behavior

Getting the above error response from device.

### Code to reproduce this issue

```cpp
esp_http_client_config_t clientConfig = {
        .url = "url_to_s3_bucket_location", 
        .event_handler = ota_event_handler,
        .cert_pem = (char *)server_cert_pem_start,
        .timeout_ms = 20000, // default is 5000
        .keep_alive_enable = true};

esp_https_ota_config_t ota_config = {
        .http_config = &clientConfig};

esp_https_ota_handle_t ota_handle = NULL;

if (esp_https_ota_begin(&ota_config, &ota_handle) != ESP_OK)
{
        ESP_LOGE(GNAR_TAG, "esp_https_ota_begin failed");
        esp_restart();
        continue;
}

esp_app_desc_t incoming_ota_desc;
if (esp_https_ota_get_img_desc(ota_handle, &incoming_ota_desc) != ESP_OK)
{
        ESP_LOGE(GNAR_TAG, "esp_https_ota_get_img_desc failed");
        esp_https_ota_finish(ota_handle);
        esp_restart();
        continue;
}

if (validate_image_header(&incoming_ota_desc) != ESP_OK)
{
        ESP_LOGE(GNAR_TAG, "validate_image_header failed");
        esp_https_ota_finish(ota_handle);
        esp_restart();
        continue;
}

while (true)
{
        esp_err_t ota_result = esp_https_ota_perform(ota_handle);
        if (ota_result != ESP_ERR_HTTPS_OTA_IN_PROGRESS)
        break;
}

if (esp_https_ota_finish(ota_handle) != ESP_OK)
{
        ESP_LOGE(GNAR_TAG, "esp_https_ota_finish failed");
        esp_restart();
        xSemaphoreTake(ota_semaphore, portMAX_DELAY);
        continue;
}
else
{
        esp_restart();
}
ESP_LOGE(GNAR_TAG, "Failed to update firmware");
```

## Debug Logs

```
E (26990) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x256a
I (26990) esp-tls-mbedtls: Certificate verified.
E (26990) esp-tls: Failed to open new connection
E (26990) TRANS_SSL: Failed to open a new connection
E (27000) HTTP_CLIENT: Connection failed, sock < 0
E (27000) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (27010) esp_https_ota: Failed to establish HTTP connection
E (27020) GNAR OTA: esp_https_ota_begin failed
```

z7jzk, 2021-11-08:
**Update** After running `idf.py fullclean`, removing the sdkconfig, and rebuilding, I'm now getting the following error before the ESP restarts on failure:

```
W (71650) wifi:<ba-add>idx:1 (ifx:0, f0:72:ea:71:6d:ca), tid:0, ssn:0, winSize:64
E (72100) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3b00
I (72100) esp-tls-mbedtls: Certificate verified.
E (72100) esp-tls: Failed to open new connection
E (72100) TRANS_SSL: Failed to open a new connection
E (72110) HTTP_CLIENT: Connection failed, sock < 0
E (72110) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (72120) esp_https_ota: Failed to establish HTTP connection
E (72130) esp_https_ota_begin failed
```

Here is what I'm getting from heap output also:
```
W (71640) free heap size: 45208
W (71640) max free heap size: 43032
```

mahavirj, 2021-11-08:
@z7jzk 

> E (26990) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x256a

This error points to insufficient dynamic memory and hence allocation failure.

> E (72100) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3b00

This points to some error in parsing certificate file. Could you please double check on credentials passed to `esp_https_ota` API once?

Alternatively, please try OTA examples with local HTTPS server once per instructions at  https://github.com/espressif/esp-idf/tree/master/examples/system/ota#run-https-server and https://github.com/espressif/esp-idf/tree/master/examples/system/ota#local-https-server

z7jzk, 2021-11-09:
Hi @mahavirj, 
Thank you for the feedback. I was able to get OTA working again today! I had two main issues for anyone that stumbles upon this thread:

I was battling insufficient memory (as you pointed out above) in some instances where I had been calling `xTaskCreate()` on the same task more than once in some cases (fixed, available memory is now consistent at runtime). I was also using the wrong certificate for my custom domain that points to an AWS S3 Bucket location. I resolved this by using the correct certificate by referencing the reply on [this thread](https://esp32.com/viewtopic.php?t=15788#p61358).

Thanks again!

mahavirj, 2021-11-09:
@z7jzk Great to hear that your issue is resolved. Thanks for sharing additional insights. I will go ahead and close the issue.

closed 2021-11-09 by mahavirj