espressif/esp-idf #7851: How to define   image_size, base_dir, output_file for spiffsgen.main()

dustralia, 2021-11-07:
How to define   image_size, base_dir, output_file for spiffsgen.main()

spiffsgen.main('0x170000', 'data', 'data.bin') ---- TypeError: main() takes 0 positional arguments but 3 were given

spiffsgen.main() ---- error: the following arguments are required: image_size, base_dir, output_file

igrr, 2021-11-28:
Hi @dustralia, I'm sorry that it took a long time to reply. `main` function of spiffsgen.py uses `argparse` module to parse the arguments supplied on the command line. The arguments are taken from `sys.argv` array (implicitly, in the call to parser.parse_args()): https://github.com/espressif/esp-idf/blob/ef98a363e35dbc6bfd1caadfab8706874930f278/components/spiffs/spiffsgen.py#L511-L578.

You can do something like this to call spiffsgen.py main function from another script:

```
import sys
import spiffsgen

sys_argv_saved = sys.argv
spiffsgen_args = ['spiffsgen.py', '0x170000', 'data', 'data.bin']
try:
    sys.argv = spiffsgen_args
    spiffsgen.main()
finally:
    sys.argv = sys_argv_saved
```

the first argument in argv array (argv[0]) is the program name (as passed to the program itself), subsequent arguments are parsed by the program.

closed 2021-12-13 by igrr

dustralia, 2021-12-17:
OK，thank you