espressif/esp-idf #7852: .gpio_cs needed if only 1 SDCARD

franz-ms-muc, 2021-11-08:
Hello, 

using Example: 
https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi

then i am not able to set 
`#define PIN_NUM_CS   GPIO_NUM_NC `

Error: 
`I (331) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (340) example: Initializing SD card
I (340) example: Using SPI peripheral
I (350) example: Mounting filesystem
E (350) gpio: GPIO_PIN mask error 
E (360) vfs_fat_sdmmc: Failed to attach sdspi device onto an SPI bus (rc=0x102), please initialize the bus first and check the device parameters.
E (370) vfs_fat_sdmmc: slot init failed (0x102).
E (380) example: Failed to initialize the card (ESP_ERR_INVALID_ARG). Make sure SD card lines have pull-up resistors in place.
`

but setting it to some unused Pin it works: 

`
I (331) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (340) example: Initializing SD card
I (340) example: Using SPI peripheral
I (350) example: Mounting filesystem
I (350) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (400) sdspi_transaction: cmd=5, R1 response: command not supported
I (520) example: Filesystem mounted
Name: USDU1
Type: SDHC/SDXC
Speed: 20 MHz
Size: 60290MB
I (520) example: Opening file /sdcard/hello.txt
`

the CS Pin is not used on the SDCARD, but pulled down with a Resistor. 

so, why this Error ? 
if only 1 SDCARD it seems not using the CS.

Thanks, 
Franz

igrr, 2021-11-08:
@franz-ms-muc Thanks for the suggestion, we can add support in SDSPI driver for CS line being set to GPIO_NUM_NC.

AxelLin, 2022-06-18:
> @franz-ms-muc Thanks for the suggestion, we can add support in SDSPI driver for CS line being set to GPIO_NUM_NC.

Any update?

igrr, 2022-06-18:
@AxelLin not yet, sorry. This is still in the backlog with a relatively low priority.

closed 2022-08-12 by espressif-bot

adokitkat, 2022-08-12:
Hello @franz-ms-muc. I hope your issue is resolved now. If not, feel free to reopen this.