espressif/esp-idf #7853: Lose of precision on pdTICKS_TO_MS convertions.

lisekt84, 2021-11-08:
## Environment

ESP-IDF:          v4.3.1-246-g37a736e419

## Problem Description

Lose of precision on pdTICKS_TO_MS convertions in **xTicks * 1000** multiply operation (output could exceed TickType_t max value).
Line 45: /components/freertos/include/freertos/projdefs.h 
```cpp
	#define pdTICKS_TO_MS( xTicks )   ( ( uint32_t ) ( xTicks ) * 1000 / configTICK_RATE_HZ )
```

### Expected Behavior
```cpp
//configTICK_RATE_HZ = 1000
uint32_t output = pdTICKS_TO_MS(4,294,967,295)
```
`**output: 4,294,967,295**` 

### Actual Behavior 
```cpp
//configTICK_RATE_HZ = 1000
uint32_t output = pdTICKS_TO_MS(4,294,967,295)
```
`**output: 4,294,967**` 

### Code to resolve this issue

```cpp
	#define pdTICKS_TO_MS( xTicks )   ( ( TickType_t ) ( ( uint64_t ) ( xTicks ) * 1000 / configTICK_RATE_HZ ) )
```

closed 2021-12-01 by espressif-bot

AxelLin, 2022-01-01:
v4.3 still does not include this fix.

AxelLin, 2022-01-15:
Just remind that v4.4-rc1 does not include this fix.

igrr, 2022-01-15:
Thanks for the reminder @AxelLin. We will merge this fix into release/v4.4 soon after v4.4 release is tagged. This fix will be part of v4.4.1.

Alvin1Zhang, 2022-01-17:
Fix on release/4.3 is available at https://github.com/espressif/esp-idf/commit/6dce875bf9ab3672508d0bb74cd360e60f622395, and will be part of IDF v4.3.3 release, thanks.