espressif/esp-idf #7859: Firmware image being automatically set to valid after OTA update, rollback not being possible.

GKFernandes, 2021-11-09:
## Environment

- Development Kit:|ESP32-DevKitC|
- Kit version DevKitC):v2
- Module or chip used: ESP32-WROOM-32E
- IDF version: 4.2.1
- Build System: CMake through platformio
- Compiler version:  toolchain-xtensa32 2.80400.210211 (8.4.0)
- Operating System: Windows
- (Windows only) environment type: Git Bash/PlatformIO Terminal
- Using an IDE?: Yes, using VSCode
- Power Supply: USB 5V

## Problem Description

Hi everyone,

So, after reading some documentation over here (Over The Air Updates (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html) I was able to perform the OTA update using the advanced_https_ota example code.

At the rollback process part (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html#rollback-process) it is mentioned that when you perform OTA, the first boot with the new partition is monitored, therefore it receives the ESP_OTA_IMG_PENDING_VERIFY flag and in case of reboot before calling the esp_ota_mark_app_valid_cancel_rollback() function the application would rollback to the previous working version.

Unfortunately, that is not what is happening, instead, after the OTA update, both images are shown as valid, so the rollback never occurs.

Did somebody encounter the same problem?

Thanks!

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

/* Advanced HTTPS OTA example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_http_client.h"
#include "esp_https_ota.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"

#if CONFIG_EXAMPLE_CONNECT_WIFI
#include "esp_wifi.h"
#endif

static const char *TAG = "advanced_https_ota_example";
extern const uint8_t server_cert_pem_start[] asm("_binary_ca_cert_pem_start");
extern const uint8_t server_cert_pem_end[] asm("_binary_ca_cert_pem_end");

#define OTA_URL_SIZE 256

[... 152 lines cut ...]
User-Agent: ESP32 HTTP Client/1.0
Host: gkzinotabucket.s3.amazonaws.com
Content-Length: 0

D (44903) HTTP_CLIENT: on_message_begin
D (44903) HTTP_CLIENT: HEADER=x-amz-id-2:7cbAkdG4tXwFwYl3swElOzkomudJ+OLBXNh9gKFZxRknyf/5a0g3cTsE9KptN0g1ZZ6RkYydnis=
D (44913) HTTP_CLIENT: HEADER=x-amz-request-id:VFRQ483KZHZG2CDG
D (44913) HTTP_CLIENT: HEADER=Date:Thu, 04 Nov 2021 20:07:53 GMT
D (44923) HTTP_CLIENT: HEADER=Last-Modified:Tue, 31 Aug 2021 15:16:23 GMT
D (44933) HTTP_CLIENT: HEADER=ETag:"d7e2d1e286862db7251234ffe548147d"
D (44933) HTTP_CLIENT: HEADER=x-amz-version-id:okvWFIuhTSn4ZJiFLypGIlPqXP2iSTlP
D (44943) HTTP_CLIENT: HEADER=Accept-Ranges:bytes
D (44943) HTTP_CLIENT: HEADER=Content-Type:application/octet-stream
D (44953) HTTP_CLIENT: HEADER=Server:AmazonS3
D (44963) HTTP_CLIENT: HEADER=Content-Length:1270288
D (44963) HTTP_CLIENT: http_on_headers_complete, status=200, offset=427, nread=427
D (44973) HTTP_CLIENT: content_length = 1270288
I (44973) esp_https_ota: Starting OTA...
I (44983) esp_https_ota: Writing to partition subtype 17 at offset 0x160000
D (44983) HTTP_CLIENT: Data processed 0 != Data specified in content length 1270288
D (44993) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1270288
D (45003) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=85, ridx=0
D (45013) HTTP_CLIENT: http_on_body 85
D (45013) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1270288
Current time: Thursday, 04-Nov-2021 21:07:52 UTC

KaeLL, 2021-11-09:
> when CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE option is enabled

GKFernandes, 2021-11-09:
@KaeLL Thanks for reminding me to upload my sdkconfig file, I've attached it in the bottom part of the issue.

laukik-hase, 2021-11-11:
Hello, @GKFernandes 

I tried reproducing this issue on tag `v4.2.1` but was unable to do it. The `advanced_https_ota` example (with your configurations) ran just fine.

Alvin1Zhang, 2021-11-23:
Thanks for reporting, sorry that we could not reproduce the issue locally, you could also raise a post on our forum https://esp32.com, feel free to reopen, thanks.

closed 2021-11-23 by Alvin1Zhang