espressif/esp-idf #7860: SPI

YangJiawei, 2021-11-09:
## Environment

- Module or chip used:  [ESP32C3]
- IDF version (run ``git describe --tags`` to find it):
 v4.3.1
- Build System:         [idf.py]
- Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]

## Problem Description
```
typedef struct mj_spi_init_s
{
    spi_bus_config_t buscfg;              
    spi_device_interface_config_t devcfg;
    spi_device_handle_t device_handle;   

} mj_spi_data_t;

typedef mj_spi_data_t *mj_spi_handle_t;
```
```
esp_err_t mj_spi_read_buff(mj_spi_handle_t mj_spi_handle, uint8_t reg, uint8_t *buff, int length)
{
    esp_err_t ret;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t));      
    t.addr = (MPU_READ_FLAG | reg); 
    t.length = length * 8;       
    t.tx_buffer = NULL;        
    t.rx_buffer = buff;           
    t.user = (void *)0;      
    ret = spi_device_polling_transmit(((mj_spi_handle_t)mj_spi_handle)->device_handle, &t);
    // gpio_set_level(PIN_NUM_CS, 1);
    return ret;
}
esp_err_t mj_spi_write_buff(mj_spi_handle_t mj_spi_handle, uint8_t reg, uint8_t *buff, int length)
{
    if (length == 0)
        return ESP_OK;
    esp_err_t ret;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t)); 
    t.addr = reg;
    t.length = length * 8;
    t.tx_buffer = buff;
    t.rx_buffer = NULL;    
    ret = spi_device_polling_transmit(((mj_spi_handle_t)mj_spi_handle)->device_handle, &t);
    return ret;
}
```
```
esp_err_t mj_spi_write8(mj_spi_handle_t mj_spi_handle, uint8_t reg, uint8_t val)
{
    esp_err_t ret;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t)); 
    t.length = 8;           
    t.addr = (MPU_WRITE_FLAG & reg);
    t.tx_data[0] = val;
    t.flags = SPI_TRANS_USE_TXDATA;
    ret = spi_device_polling_transmit(((mj_spi_handle_t)mj_spi_handle)->device_handle, &t); 
    return ret;
}

esp_err_t mj_spi_write_buff2(mj_spi_handle_t mj_spi_handle, uint8_t reg, uint8_t *buff, int length)
{
    for (int i = 0; i < length; i++)
    {
        ESP_ERROR_CHECK(mj_spi_write8(mj_spi_handle, reg + i, buff[i]));
    }
    return ESP_OK;
}
```
```
void mj_p(char *p, uint8_t *buff, int count)
{
    printf("%s", p);
    for (int i = 0; i < count; i++)
    {
        printf("%02X", buff[i]);
    }
    printf("\n");
}

  unsigned char rrrr[6] = {0};
  unsigned char wwww[6] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06};

  mj_spi_read_buff(mj_spi_handle, 0x13, rrrr, 6);
  mj_p("rrrr1:", rrrr, 6);
  mj_spi_write_buff(mj_spi_handle, 0x13, wwww, 4);
  mj_spi_read_buff(mj_spi_handle, 0x13, rrrr, 6);
  mj_p("rrrr2:", rrrr, 6);

```
Use SPI to read MPU6500.
According to the above code, the value written to the register using `mj_spi_write_buff` is always 0x01.
However, the value written to the register using `mj_spi_write_buff2` is correct.
So where is the configuration error of the `mj_spi_write_buff` function?

### Expected Behavior
```
rrrr1:000000000000
rrrr2:010203040000
```
### Actual Behavior

```
rrrr1:000000000000
rrrr2:010101010000
```

YangJiawei, 2021-11-12:
esp32C3 will have this problem when using DMA. If SPI_DMA_DISABLED is used, the transfer byte is fixed at 64B.

nopnop2002, 2022-01-04:
In my experience, some ESP32-C3 development boards do not have the SPI connected correctly.

Rigth one: Work correct.
Left one: Dont work correct.
[image]

closed 2025-03-13 by Alvin1Zhang