espressif/esp-idf #7862: "esp_partition_erase_range" causes esp32 to crash/restart, when BLE is enabled

MartinTJDK, 2021-11-09:
SDK: V4.3.1
CPU: ESP32
Module: WRover

When calling "esp_partition_erase_range", the ESP32 reboots immediately.

I have tried to call "esp_partition_erase_range" 10 times, with a delay for 1 second, at multiple places in the "main()".
It succeeds successfully for many times, but as soon as BLE is enabled, it causes the ESP32 to reboot.

I use "esp_partition_erase_range" for multiple purposes on a running device, and it is an issue, that BLE cannot be enabled, while erasing a partition. If I only start Wi-Fi part, it is not an issue.

When I do not start any BLE stuff, if works perfectly.

The exact same application worked without any issue on SDK V3.3.5.

I see other with same bug (https://github.com/espressif/esp-idf/issues/6068), but that bug is 1 year old.

MartinTJDK, 2021-11-09:
I tried to reduce the size to erase to only 0x1000. But the system still reboots as soon as BLE is enabled.

0xjakob, 2021-11-10:
@MartinTJDK Do you have a log of the crash? It would be good to have information such as backtrace and in particular the register dump which will tell us something about the state of the processor and what type of exception occurred.

BitProgress, 2021-11-10:
I will create it now. Because it is 100% replicable (and frustrating)...

BitProgress, 2021-11-10:
The only output I get during the restart is:
(I have enabled coredump output to uart, because I see no dump to flash)

*******************************************************************************************
I (33365) ble_impl: Initializing Completed
I (33365) ble_driver_server: BLE Server Initialized
D (33375) ble_impl: esp_gattc_cb
D (33375) ble_impl: ESP_GATTC_REG_EVT
D (33385) ble_driver_client: ************** BLE_CLIENT_STATE__READY_IDLE
I (33385) ble_driver_client: BLE Client Initialized
[00000031.127319] BleMgr : [Informational] Starting server
D (33405) ble_driver_server: Added to Whitelist
I (33405) ble_impl: Starting advertisement
I (33415) ble_impl: Starting advertisement
[00000031.149655] APP : [Notice] Erase Coredump
[00000031.172475] APP : [Notice] Erase Coredump Success
[00000032.170369] APP : [Notice] Erase Coredump
[00000032.186948] APP : [Notice] Erase Coredump Success
[00000033.180641] APP : [Notice] Erase Coredump
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7072
load:0x40078000,len:14876
load:0x40080400,len:3764
0x40080400: _init at ??:?

entry 0x40080680
I (160) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (161) boot: compile time 14:24:24
I (161) boot: chip revision: 1
I (164) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (171) boot.esp32: SPI Speed      : 40MHz
I (176) boot.esp32: SPI Mode       : DIO
I (181) boot.esp32: SPI Flash Size : 8MB
W (185) boot.esp32: PRO CPU has been reset by WDT.
W (191) boot.esp32: WDT reset info: PRO CPU PC=0x4008e0a0
0x4008e0a0: _xt_panic at C:\ESP\iotgwhc\build/../platform/esp/esp-idf/components/esp_system/port/arch/xtensa/panic_handler_asm.S:30

W (197) boot.esp32: WDT reset info: APP CPU PC=0x4008c99e
0x4008c99e: spi_flash_op_block_func at C:\ESP\iotgwhc\build/../platform/esp/esp-idf/components/spi_flash/cache_utils.c:115 (discriminator 1)

I (203) boot: Enabling RNG early entropy source...
*******************************************************************************************

My settings for watchdogs are:
[image]

Does it give any clues? What else can I get for you?

BitProgress, 2021-11-10:
@0xjakob 

Reverting this setting: "CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL" to 4096 makes system run normally again.
I was using 64 to free more internal RAM.
Perhaps SPI flash does not allocate memory from the internal pool? (and value less than 4096 results in external RAM allocation)

BitProgress, 2021-11-10:
@0xjakob Setting the value to 128 makes the system survive. But it would be interesting to know, if there was a good explanation...

fromeijn, 2023-02-15:
experiencing the same issue but ble is not on, but uart is (in iram)

my backtace looks like this:
```
0x4008d5cd: uart_rx_intr_handler_default at esp-idf/components/driver/uart.c:992
0x40083db1: _xt_lowint1 at esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4008b866: spi_flash_op_block_func at esp-idf/components/spi_flash/cache_utils.c:117
0x4008b774: ipc_task at esp-idf/components/esp_ipc/ipc.c:77
```

im on idf version V4.3.4

setting CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL to 128 seems to work indeed