espressif/esp-idf #7863: ESP32S2 - RISC-V crashes with ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG

bw38, 2021-11-09:
## Environment

- Development Kit:      [ESP32S2-Wrover-Module & featherS2]
- Module or chip used:  [ESP32S2 WROVER]
- IDF version  ESP-IDF v4.4-dev-3042-g220590d599
- Build System:         [CMake|idf.py]
- Compiler version  xtensa-esp32s2-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Eclipse 2020-12]
- Power Supply:         [external 3.3V or Battery 3.6V]

## Problem Description
A program ends with deepsleep and waits for the next ulp-riscv-wakeup. It works fine as long the ulp-riscv-programm ends with a wake of the main-processor.
If ulp-programm don't wakeup the main (because the measurement result isn't different from the last one), an exception raises after some ulp-runs and the mainproc start with "ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG".

**But it only happens** when the power-domain "rtc-periph" is switched on before deepsleep

### Expected Behavior
The ULP-RiscV wakes, runs and goes to halt in the predefined wakeup-loop. The MainProc sleeps until a wake is triggered from the RiscV

### Actual Behavior
The MainProc booted after some RiscV-loops with reset-cause 11.
But only when the rtc-perith is switched ON before deepsleep in mainproc.

### Steps to reproduce

Use from idf-examples: 
"examples/system/ulp_riscv/ds18b20_onewire"

insert in main-code: 
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

run the program and monitor the uart

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
void app_main()
{
    ......
    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());

   //insert this line:
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
    
    esp_deep_sleep_start();
}
```
hope it helps

Alvin1Zhang, 2022-07-11:
Thanks for reporting, the fix on master branch is available at https://github.com/espressif/esp-idf/commit/0eafe4d326b96c72932e8859de88fc8d527056ab, we are also back porting this fix to release/4.4, thanks.

AxelLin, 2022-08-10:
@Alvin1Zhang 
Just remind that v4.4.2 does not include this fix.

sudeep-mohanty, 2022-08-11:
Hello @AxelLin, The change made under this commit was reverted under the commit 9c4a12b11ec7d79142900b07b36cc31609ba957e as it was no longer needed. The issue reported under this ticket is not observed after some subsequent ULP fixes which have been backported duly. Therefore, this particular commit needn't be backported to other IDF release branches as of now. In case the issue is seen again, we will have a new investigation on the same.

closed 2022-08-11 by sudeep-mohanty

yangdx2022, 2023-01-02:
> Hello @AxelLin, The change made under this commit was reverted under the commit [9c4a12b](https://github.com/espressif/esp-idf/commit/9c4a12b11ec7d79142900b07b36cc31609ba957e) as it was no longer needed. The issue reported under this ticket is not observed after some subsequent ULP fixes which have been backported duly. Therefore, this particular commit needn't be backported to other IDF release branches as of now. In case the issue is seen again, we will have a new investigation on the same.

Hello @sudeep-mohanty , If ulp dont wake up main processor after using i2c, ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG still occurr. but without i2c working, ulp can work. 
```

uint32_t running_cnt = 0;
uint8_t data[1 << 4];
int main (void) {
    int i;
    running_cnt += 1;
    // ulp_riscv_i2c_master_set_slave_addr(0x32);
    // ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
    // ulp_riscv_i2c_master_read_from_device(data, 1 << 4);
    if(0 == (running_cnt % 4)) ulp_riscv_wakeup_main_processor();

    return 0;
}

```

yangdx2022, 2023-01-02:
> Hello @AxelLin, The change made under this commit was reverted under the commit [9c4a12b](https://github.com/espressif/esp-idf/commit/9c4a12b11ec7d79142900b07b36cc31609ba957e) as it was no longer [...]

this is my esp idf version. 
commit 0025915dc489a9d45f99aed74920346f8ac4ec09 (HEAD -> master, origin/master, origin/HEAD)
Merge: c09cbfe414 094997a2ec
Author: Shu Chen <chenshu@espressif.com>
Date:   Sat Dec 31 09:26:23 2022 +0800

    Merge branch 'feature/public_ot_port' into 'master'
    
    openthread: open source openthread port layer
    
    See merge request espressif/esp-idf!21803

sudeep-mohanty, 2023-01-02:
Hi @yangdx2022 Thanks for reporting the issue.
We had a similar issue reported under #10301 for which a fix is bring worked upon currently. However, there is a patch that is shared [here](https://github.com/espressif/esp-idf/issues/10301#issuecomment-1357167592) for the same. Could you try it out and let us know if it helped resolve your issue?

yangdx2022, 2023-01-02:
> Hi @yangdx2022 Thanks for reporting the issue. We had a similar issue reported under #10301 for which a fix is bring worked upon currently. However, there is a patch that is shared [...]

thanks for your help, i will try to fix it using the patch.

yangdx2022, 2023-01-03:
> Hi @yangdx2022 Thanks for reporting the issue. We had a similar issue reported under #10301 for which a fix is bring worked upon currently. However, there is a patch that is shared [...]

sorry, it dont work, with the patch, when ulp first halt , ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG ocurrs. 
`diff --git a/components/ulp/ulp_riscv/ulp_core/ulp_riscv_utils.c b/components/ulp/ulp_riscv/ulp_core/ulp_riscv_utils.c
index c2f4a3de92..992d46338b 100644
--- a/components/ulp/ulp_riscv/ulp_core/ulp_riscv_utils.c
+++ b/components/ulp/ulp_riscv/ulp_core/ulp_riscv_utils.c
@@ -26,12 +26,9 @@ void ulp_riscv_halt(void)
     /* Setting the delay time after RISCV recv `DONE` signal, Ensure that action `RESET` can be executed in time. */
     REG_SET_FIELD(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_2_CLK_DIS, 0x3F);
 
-    /* suspends the ulp operation*/
-    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE);
-
-    /* Resets the processor */
-    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);
-
+    /* Suspends the ulp operation and reset the ULP core. Must be the final operation before going to halt. */
+    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE | RTC_CNTL_COCPU_SHUT_RESET_EN);
+ 
     while(1);
 }
 
diff --git a/components/ulp/ulp_riscv/ulp_riscv.c b/components/ulp/ulp_riscv/ulp_riscv.c
index 0e6bda3ded..4a8e4f095e 100644
--- a/components/ulp/ulp_riscv/ulp_riscv.c
+++ b/components/ulp/ulp_riscv/ulp_riscv.c
@@ -75,8 +75,6 @@ esp_err_t ulp_riscv_config_and_run(ulp_riscv_cfg_t* cfg)
 #elif CONFIG_IDF_TARGET_ESP32S3
     /* Reset COCPU when power on. */
     SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);
-    esp_rom_delay_us(20);
-    CLEAR_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);
 
      /* The coprocessor cpu trap signal doesnt have a stable reset value,
        force ULP-RISC-V clock on to stop RTC_COCPU_TRAP_TRIG_EN from waking the CPU*/
diff --git a/examples/system/ulp_riscv/adc/main/ulp/main.c b/examples/system/ulp_riscv/adc/main/ulp/main.c
index f049c1122a..dc91dd6d98 100644
--- a/examples/system/ulp_riscv/adc/main/ulp/main.c
+++ b/examples/system/ulp_riscv/adc/main/ulp/main.c
@@ -27,7 +27,7 @@ int main (void)
 {
     int32_t last_result = ulp_riscv_adc_read_channel(EXAMPLE_ADC_UNIT, EXAMPLE_ADC_CHANNEL);
 
-    if (last_result > adc_threshold) {
+    if (last_result < adc_threshold) {
         wakeup_result = last_result;
         ulp_riscv_wakeup_main_processor();
     }`

sudeep-mohanty, 2023-01-03:
Thanks for trying out @yangdx2022. Would you mind opening a new GH issue for us to investigate this behaviour further? It would be great if you could share your target platform details, sample code (main CPU and ULP) and your sdkconfig as well for us to reproduce the issue at our end. Thanks!

yangdx2022, 2023-01-04:
> Thanks for trying out @yangdx2022. Would you mind opening a new GH issue for us to investigate this behaviour further? It would be great if you could share your target platform details, sample code (main CPU and ULP) and your sdkconfig as well for us to reproduce the issue at our end. Thanks!

thanks for your help, i has opened a new issue, its url is https://github.com/espressif/esp-idf/issues/10482. do you need any info?

sudeep-mohanty, 2023-01-04:
Thanks for opening the issue @yangdx2022. I have everything to investigate the issue.