espressif/esp-idf #7865: Smart Config does not works on some wifi configurations:

ivan-kuzlo-chisw, 2021-11-10:
Smart config does not works on some wifi configurations:
Router Microtik, Band: 2.4Ghz b/g/n Tested channels: [8,9]
Channel width:

20MHz → Works as expected
20/40MHZ eC → Works as expected
20/40MHZ Ce → Device don't receive configuration commands
20/40MHZ XX → Device don't receive configuration commands (looks like it auto selected Ce mode)

About eC/Ce mode -> found some answers on Microtik router form:
*20/40MHz channel width also have index (Ce, eC, XX). The index determines the direction of increasing the channel size (C - main, e - additional). Ce (extension by next channel), eC (extension by previous channel) and XX (automatic direction selection).

mhdong, 2021-11-10:
hi @ivan-kuzlo-chisw Does 20/40MHZ Ce mean router will use channel 9 and channel 10 transmit packet?If so, please confirm that channel 10 is available in the current area.

ivan-kuzlo-chisw, 2021-11-15:
@mhdong yes, channel 12 and 13 is available
make some testing on different channels and get this results

Channel width:
20MHz works all channels (1-13)
40MHz Ce works only 1-8 channels (channel 9 not working)
40MHz eC works all channels (5-13)

mhdong, 2021-11-15:
hi @ivan-kuzlo-chisw  You can use Wireshark to capture packets to make sure whether your router transmit data use 40MHz Ce channel 9. By the way, the ASUS router I use in China does not support setting to 40MHz Ce channel 9.

closed 2022-03-04 by Alvin1Zhang