espressif/esp-idf #7870: after setting fixed IP address, some services won't start

dannybackx, 2021-11-10:
## Environment

- Development Kit:      any esp32
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): 4.3.1
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Using an IDE?:  No
- Power Supply:        USB

## Problem Description

When using the code below to allocate a fixed IP address, some services (own FTP server, but also standard SNTP client) won't start. Looks as if not everything is ok when the IP_EVENT_STA_GOT_IP event handler runs.

So when running this, things fail :
```
void Network::con_event_handler(void *ctx, esp_event_base_t event_base, int32_t event_id, void *event_data) {
  if (config == 0) {
    ESP_LOGE(snetwork_tag, "%s: no config, bailing", __FUNCTION__);
    return;
  }

  ESP_LOGI(snetwork_tag, "trying preset ip %s gw %s mask %s",
    config->getPresetIP() ? config->getPresetIP() : "",
    config->getPresetGW() ? config->getPresetGW() : "",
    config->getPresetSubnetMask() ? config->getPresetSubnetMask() : "");

  if (config == 0 || config->getPresetIP() == 0 || config->getPresetGW() == 0
   || config->getPresetSubnetMask() == 0)
    return;

  esp_netif_t *netif = (esp_netif_t *)ctx;

  if (esp_netif_dhcpc_stop(netif) != ESP_OK) {
    ESP_LOGE(snetwork_tag, "%s: failed to stop dhcp client", __FUNCTION__);
    return;
  }

  esp_netif_ip_info_t ip;
  memset(&ip, 0 , sizeof(esp_netif_ip_info_t));
  ip.ip.addr = ipaddr_addr(config->getPresetIP());
  ip.netmask.addr = ipaddr_addr(config->getPresetSubnetMask());
  ip.gw.addr = ipaddr_addr(config->getPresetGW());

  if (esp_netif_set_ip_info(netif, &ip) != ESP_OK) {
    ESP_LOGE(snetwork_tag, "%s: failed to set IP address", __FUNCTION__);
    return;
  }
}
```

### Expected Behavior

Output from a run without that code (a return statement just after the "trying to preset" line). Remark e.g. that SNTP updates the log timestamp at some point.
```
I (01:00:01.227) Network: Waiting for wifi
I (01:00:01.232) Network: Wifi 0, ssid [HolidayHome59B]
E (01:00:03.398) Network: event_handler: wifi STA_DISCONNECTED
I (01:00:03.399) Network: discon_event_handler
E (01:00:03.399) Network: Failed to connect to SSID xxxxxxx (reason 201 NO_AP_FOUND)
I (01:00:03.409) Network: Waiting for wifi
I (01:00:03.413) Network: Wifi 1, ssid [yyyyyy]
E (01:00:04.064) Network: event_handler: wifi STA_CONNECTED
I (01:00:04.065) Network: trying preset ip 192.168.0.123 gw 192.168.0.1 mask 255.255.255.0
I (01:00:05.220) Network: Network connected, ip 192.168.0.140 SSID yyyyyy
I (01:00:05.221) App: FTP started
I (01:00:05.284) WebServer: ReadFile: read from /fs/acme/production/certkey.der, len 1192
I (01:00:05.347) WebServer: ReadFile: read from /fs/acme/production/certificate.der, len 5638
I (01:00:05.348) esp_https_server: Starting server
I (01:00:05.354) esp_https_server: Server listening on port 4435
I (01:00:05.358) WebServer: Start SSL webserver (4435)
I (01:00:05.365) WebServer: Start webserver(80)
I (01:00:05.369) WebServer: Start: registered /alarm HTTP_GET handler for HTTPS
I (01:00:05.377) JSONServer: registered /json HTTP_PUT for HTTPS server
I (01:00:05.384) JSONServer: registered /json HTTP_GET for HTTPS server
I (01:00:05.392) JSONServer: registered /json HTTP_PUT for HTTP server
I (01:00:05.399) JSONServer: registered /json HTTP_GET for HTTP server
I (01:00:05.406) Ota: registered /update HTTP_POST handler for HTTP
I (01:00:05.413) Ota: registered / HTTP_GET handler for HTTP
I (01:00:05.419) Ota: registered /serverIndex HTTP_GET handler for HTTP
I (01:00:05.427) TLS loop: No tls task started
I (01:00:05.432) Network: NetworkConnected: start SNTP
I (01:00:05.441) Network: mDNS set (lite2)
I (01:00:05.442) Network: Return from ip_event_handler
E (01:00:05.448) Network: event_handler: ip STA_GOT_IP
I (15:42:44.998) Acme: TimeSync
I (15:42:45.007) App: Controller lite2 boot at 2021-11-10 15:42:44
I (15:42:45.095) App: DynDNS update succeeded for xxxxxx.dns-cloud.net
I (15:42:45.174) App: DynDNS update succeeded for yyyyyy.cloudns.cl
I (15:42:45.215) Acme: Reading root certificate from /fs/root.pem
I (15:42:45.218) Acme: Querying directory at https://acme-v02.api.letsencrypt.org/directory
I (15:42:54.273) Acme: ReadCertificate: we have a certificate in /fs/acme/production/certificate.der
I (15:42:54.274) Acme: Valid from 2021-10-11 14:46:48 to 2022-01-09 14:46:47

```

### Actual Behavior
Logging with the static IP address enabled (remark 2 lines below showing the same IP address, but SNTP never updates the log timestamps).
```
I (01:00:01.228) Network: Waiting for wifi
I (01:00:01.233) Network: Wifi 0, ssid [HolidayHome59B]
E (01:00:03.400) Network: event_handler: wifi STA_DISCONNECTED
I (01:00:03.401) Network: discon_event_handler
E (01:00:03.402) Network: Failed to connect to SSID HolidayHome59B (reason 201 NO_AP_FOUND)
I (01:00:03.412) Network: Waiting for wifi
I (01:00:03.416) Network: Wifi 1, ssid [RoeselUBackx]
E (01:00:04.157) Network: event_handler: wifi STA_CONNECTED
I (01:00:04.158) Network: trying preset ip 192.168.0.123 gw 192.168.0.1 mask 255.255.255.0
I (01:00:04.660) Network: Network connected, ip 192.168.0.123 SSID RoeselUBackx
I (01:00:04.661) App: FTP started
I (01:00:04.720) WebServer: ReadFile: read from /fs/acme/production/certkey.der, len 1192
I (01:00:04.782) WebServer: ReadFile: read from /fs/acme/production/certificate.der, len 5638
I (01:00:04.782) esp_https_server: Starting server
I (01:00:04.788) esp_https_server: Server listening on port 4435
I (01:00:04.792) WebServer: Start SSL webserver (4435)
I (01:00:04.800) WebServer: Start webserver(80)
I (01:00:04.804) WebServer: Start: registered /alarm HTTP_GET handler for HTTPS
I (01:00:04.812) JSONServer: registered /json HTTP_PUT for HTTPS server
I (01:00:04.819) JSONServer: registered /json HTTP_GET for HTTPS server
I (01:00:04.826) JSONServer: registered /json HTTP_PUT for HTTP server
I (01:00:04.833) JSONServer: registered /json HTTP_GET for HTTP server
I (01:00:04.841) Ota: registered /update HTTP_POST handler for HTTP
I (01:00:04.847) Ota: registered / HTTP_GET handler for HTTP
I (01:00:04.854) Ota: registered /serverIndex HTTP_GET handler for HTTP
I (01:00:04.861) TLS loop: No tls task started
I (01:00:04.866) Network: NetworkConnected: start SNTP
I (01:00:04.875) Network: mDNS set (lite2)
I (01:00:04.877) Network: Return from ip_event_handler
E (01:00:04.883) Network: event_handler: ip STA_GOT_IP

```

### Code to reproduce this issue

This code is in https://sourceforge.net/p/emptyesp32/code/HEAD/tree/https/main/ .

david-cermak, 2021-11-12:
@dannybackx Have you tried to setup the static IP as it's done in this example https://github.com/espressif/esp-idf/tree/master/examples/protocols/static_ip ?

dannybackx, 2021-11-13:
I believe I based it on that example. You'll see that my con_event_handler and the example's example_set_static_ip are almost the same.
The example has one event handler for several wifi and ip events, I have a couple of handlers for separate cases. Shouldn't make a difference.
Trouble is if I use the con_event_handler then I do get a fixed IP address but only half of my app functionality starts.

Hence my question : I start all that functionality after the event caught in the example by event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP (in my code : ip_event_handler), but something appears to be off when using the static IP address calls.

david-cermak, 2021-11-15:
@dannybackx Have you setup the DNS servers as it's suggested in

https://github.com/espressif/esp-idf/blob/4a011f318377ab717b9dc459019c1055602c93ff/examples/protocols/static_ip/main/static_ip_example_main.c#L81-L82

Maybe that could be the reason of `something to appear slightly off` in your application?

dannybackx, 2021-11-15:
I just added that but it doesn't make a difference.

david-cermak, 2021-11-19:
@dannybackx Okay, let's start with one service first, say the standard SNTP client.
Could you please create a minimal project that exhibits this issue?

You can refer to a working demo application on `v4.3` using SNTP and assigning a static IP:
https://gist.github.com/david-cermak/c95c5415031888ca4be7dfa5c36a7e5e

```bash
I (929) esp_netif_handlers: sta ip: 192.168.10.222, mask: 255.255.255.0, gw: 192.168.10.1
I (929) sntp_static_ip: static ip:192.168.10.222
I (939) sntp_static_ip: connected to ap SSID:xxxxxxxxxx password:xxxxxxxxxxx
I (949) sntp_static_ip: Initializing SNTP
I (949) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (959) wifi:<ba-add>idx:0 (ifx:0, c8:e7:d8:93:a9:94), tid:0, ssn:3, winSize:64
I (959) sntp_static_ip: Waiting for system time to be set... (1/10)
I (2969) sntp_static_ip: Waiting for system time to be set... (2/10)
I (3649) sntp_static_ip: Notification of a time synchronization event
I (4969) sntp_static_ip: The current date/time in Shanghai is: Sat Nov 20 01:56:42 2021
```

dannybackx, 2021-11-19:
Thanks, I was already simplifying my source. I'll gradually bring your 
version and mine towards each other until I find something.

     Danny

dannybackx, 2021-11-20:
Thanks for pushing me to look further. It's not in the code.

Apparently my provider's DNS servers respond in an unfriendly way to queries outside their domain :
```
hp: {64} nslookup pool.ntp.org 195.130.136.242
Server:         195.130.136.242
Address:        195.130.136.242#53

** server can't find pool.ntp.org: REFUSED
```
but also the esp32 code in no way detects that.

I used this code to try and display the DNS servers the ESP gets from DHCP, but it will print only 0.0.0.0 . What am I doing wrong ?
Trying to figure out which DNS servers I can specify locally.

david-cermak, 2021-11-22:
> I used the code below to try and display the DNS servers the ESP gets from DHCP, but it will print only 0.0.0.0 . What am I doing wrong ?

This code looks good and works for me. Have you run it after getting the DHCP lease?

> Trying to figure out which DNS servers I can specify locally.

You can try to setup the GW address as your primary DNS and some global service, e.g. `8.8.8.8` as the fallback server, the same way it's done in the example:

https://github.com/espressif/esp-idf/blob/4a011f318377ab717b9dc459019c1055602c93ff/examples/protocols/static_ip/main/static_ip_example_main.c#L32

dannybackx, 2021-11-22:
Your comments are spot on again, thanks for making me feel silly ;-)

Setting DNS servers and querying them are different circumstances indeed.

     Danny

david-cermak, 2021-11-25:
Hi @dannybackx 

Do you have any other comments or concerns? Did your application started to work after fixing the DNS problem? If so, can you close the issue, please?

dannybackx, 2021-11-26:
You're right, thanks.

closed 2021-11-26 by dannybackx