espressif/esp-idf #7871: Rollback with Arduino as ESP-IDF Component

GKFernandes, 2021-11-10:
Hello everyone,

I ran into the same issue as stated in: https://github.com/espressif/arduino-esp32/issues/3318 and also in https://github.com/espressif/arduino-esp32/issues/3749

I think this issue about the weakly defined verify0ta() should be fixed and made in such a way that the user can perform the verification later in the code instead of inside the initArduino() function.

atanisoft, 2021-11-10:
This should be moved to https://github.com/espressif/arduino-esp32 since `verifyOta` and `initArduino` are implemented over there.

GKFernandes, 2021-11-10:
Thank you @atanisoft , I've moved the issue over there.

Alvin1Zhang, 2021-11-11:
@GKFernandes Thanks for reporting, see the issue is tracked in https://github.com/espressif/arduino-esp32/issues/5871, so will close this one, thanks.

closed 2021-11-11 by Alvin1Zhang