espressif/esp-idf #7876: How to modify IRAM memory size.

ShenRen, 2021-11-11:
My program doesn't use DRAM very much, but it has speed and performance requirements, so I need to put some data and functions into IRAM, and the result is IRAM overbound.

[image]

According to the official data manual, ESP32 DRAM and IRAM share one SRAM1, which is assigned to DRAM by default.

[image]

[ESP32 Programmers’ Memory Model](https://blog.espressif.com/esp32-programmers-memory-model-259444d89387)

So I want to find a way to give SRAM1 to IRAM. No matter how troublesome it is, please tell me how to do it.  I have tried to find several relevant blogs and articles and followed the steps provided. Although the program was compiled, it could not run normally.

Reference :
 1. [How to modify IRAM size](https://www.esp32.com/viewtopic.php?f=13&t=16188)
 2. [ESP32 optimization notes (4) IRAM optimization](https://www.programmersought.com/article/57566928755/)

ShenRen, 2021-11-11:
I've tried to modify the ld file and configuration file:

components/esp32/ld/esp32.ld File modification, IRAM increase 8K:
/* IRAM for PRO cpu. Not sure if happy with this, this is MMU area... */
    `iram0_0_seg (RX) :                 org = 0x40080000, len = 0x20000`
change into:
    `iram0_0_seg (RX) :                 org = 0x40080000, len = 0x22000`

components/soc/esp32/include/soc/soc.h File modification:
Increase IRAM by 8K:
    `#define SOC_IRAM_HIGH           0x400A0000`
change into:
    `#define SOC_IRAM_HIGH           0x400A2000`

DRAM reduced by 8K:
    `#define SOC_DIRAM_IRAM_LOW      0x400A0000`
change into:
    `#define SOC_DIRAM_IRAM_LOW      0x400A2000`

The program was compiled, but it could not run normally.

ShenRen, 2021-11-17:
@zikalino Hello, could you please reply to me when you are free?

ESP-Marius, 2022-08-31:
@ShenRen Sorry for the late reply!

Just from a quick test, the following changes seemed to work for me:

`iram0_0_seg (RX) :                 org = 0x40080000, len = 0x40000`
`#define SOC_IRAM_HIGH           0x400C0000`

Remove this area from being allocated as heap:
```
 // { 0x3FFE0000, 0x4000, 1, 0x400BC000}, //pool 9 blk 1
 // { 0x3FFE4000, 0x4000, 1, 0x400B8000}, //pool 9 blk 0
 // { 0x3FFE8000, 0x8000, 1, 0x400B0000}, //pool 8 <- can be remapped to ROM, used for MAC dump
 // { 0x3FFF0000, 0x8000, 1, 0x400A8000}, //pool 7 <- can be used for MAC dump
 // { 0x3FFF8000, 0x4000, 1, 0x400A4000}, //pool 6 blk 1 <- can be used as trace memory
 // { 0x3FFFC000, 0x4000, 1, 0x400A0000}, //pool 6 blk 0 <- can be used as trace memory
```
I was able to compile and run an app which would overflow the IRAM section without these changes. No guarantees that these changes are comprehensive though, there might be other places which needs to be changed that I overlooked.

ESP-Marius, 2022-09-08:
Closing this one due to lack of activity, but feel free to reopen if my answer didnt fully solve your issue.

closed 2022-09-08 by ESP-Marius

KaeLL, 2022-09-08:
@ESP-Marius Isn't this worth officially adding to IDF?

ESP-Marius, 2022-09-09:
> @ESP-Marius Isn't this worth officially adding to IDF?

I'll bring it up at our next team meeting and see if this is something we wanna officially support or not. 

In anycase,  I encourage people to add upvotes/comments here, as any decision we make might be changed if we see there is a big demand for.

oursland, 2022-09-10:
Please do.  Today I have been porting an application from v4.4.2 to the v5.0.0-beta to take advantage of the SDMMC improvements, TRIM support, and ESP32-S3 support and found that I can no longer build for our existing ESP32-based product without disabling vital IRAM optimizations for WiFi.  The penalty for disabling these IRAM optimizations is reduction in WiFi throughput by more than 50%.

At this time, my team would prefer to increase IRAM for real-time task optimizations and use our PSRAM for non-real-time tasks.

SlightlyStoopid, 2022-09-10:
This would be incredibly useful, I'm already riding 100% iram usage right now and only 58% dram, with still some options I'd love to enable still if only I could reliably increase it.

ESP-Marius, 2023-04-11:
Added a kconfig option for this in https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f.

Ribisl, 2023-08-23:
would the same be possible for the RTC_SLOW_MEMORY data segement?

filzek, 2023-12-13:
Why  do not add a SLIDER to choose from DRAM to IRAM and match the size needed ? as it is quite simple to adjust in the memory layout, is no the best to do?

ZengMetandao, 2024-03-17:
> Added a kconfig option for this in [5cbd311](https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f).

Hi @ESP-Marius 

I still cannot see this option in MenuConfig: CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM.

My ESP-IDF version is 6.4, and I'm using the PlatformIO environment.

[env]
platform = espressif32 @ 6.4.0
framework = arduino, espidf